adafruit/circuitpython #8690: Web workflow interrupts running code.py and crashes after uploading a file

RetiredWizard, 2023-12-04:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.5-31-g1f5e69243e on 2023-12-03; LILYGO T-DECK with ESP32S3
```

### Code/REPL

```python
import time
from sys import stdin
from supervisor import runtime
cmd = ""

while cmd != 'q':
    print(time.time())
    time.sleep(5)
    if runtime.serial_bytes_available:
        cmd = stdin.read(1)
```

### Behavior

The code.py file occasionally stops displaying time updates when the circuitpy drive is ejected in preparation for web workflow file transfers.

When the web workflow web page is accessed the code.py file stops displaying time updates. 

Refreshing the screen causes one time update and then stops again

Accessing the "File Browser" link causes a time update and usually starts the code.py running again but not always

Uploading a file causes the code.py to stop for an auto-reload soft reboot, after which the code.py starts and runs normally however the web workflow page no longer responds. Other Circuit Python devices still show the device, however the link doesn't work.and evenually times out with a "this site can't be reached" message.

### Additional information

By altering my workflow as follows I'm able to use the web workflow file transfer without issue.

Before accessing the web workflow page or ejecting the circuitpy drive, cancel the running code.py and enter the REPL >>>

After uploading files via web workflow I've been running a linux sync, but I suspect that's just me being superstitious.

I can restart the code.py file or run any other scripts as long as I return to the REPL before doing anything else with the web workflow webpage.

bill88t, 2023-12-05:
Oh this issue has always been there.
I can't upload (reliably, it fails around 70%) my 300 file root filesystem over it.
The code hang is just that the `run_backround_tasks` takes over and doesn't let go as far as my understanding goes.
I just didn't know how to report the upload hang and gave up. (and made ftp for that, but now that's broken too)

tannewt, 2023-12-06:
The cause of this was incorrect code in background task reset. This happens after code.py is run. Background tasks are a linked list of callback objects with function and data to call when running background tasks. The web workflow has a callback object it reuses to queue its work and sets the function to call once at the start. The background task reset code was completely zeroing the whole callback object including the function pointer and breaking it forever more. #8699 preserves callbacks that live off the python heap and fixes this issue.

RetiredWizard, 2024-01-23:
I loaded the latest artifact from #8699 up on a Feather ESP32-S3  and I'm still seeing web workflow issues. If the file browser screen is accessed the REPL appears to hang, refreshing the file browser screen releases the REPL for a few seconds and then it freezes again, after several minutes the REPL does become responsive again at least until the web workflow file browser screen is refreshed again.

Renameing or deleting a file from the web workflow file browser when the REPL is frozen will release it the same as waiting the 2-3 minutes for it to timeout. 

I've also noticed the hitting the back button from the file browser to the web workflow welcome screen will hang the REPL in the same was a screen refresh on the file browser page.

This is an improvement from Alpha.6 on which I can't even get the web workflow welcome screen to display but seems to be behaving the same as Alpha.5.

RetiredWizard, 2024-01-23:
Along with file delete/rename/uploads changing the current folder (going into the SD folder for example) through the web workflow file browser also releases a hung REPL.

> Uploading a file causes the code.py to stop for an auto-reload soft reboot, after which the code.py starts and runs normally however the web workflow page no longer responds. Other Circuit Python [...]

I've also noticed that file uploads no longer hang the web workflow.

tannewt, 2024-01-25:
I believe this is still an issue.

tannewt, 2024-02-13:
This is working for me running #8909 code. I have the serial over web socket pulled up and can browse the filesystem without any apparent hangs on a MatrixPortal S3. @RetiredWizard please retest.

RetiredWizard, 2024-02-13:
I tested using a Feather ESP32S3 4/2 and the artifact from #8909. I was not able to cause any web workflow page hangs, however the code.py file did stop running for 1-2 minutes when the initial web workflow page was displayed. Accessing the file browser screen would start the code.py program for a few seconds and it would then stop again. Performing file I/O (uploading, renaming, deleting files) would start the code.py program running until I returned (using the browser back button) to the web workflow welcome screen at which point the code.py program seemed to freeze (this time without the 1-2 minute timeout) until the web workflow file browser page was once again accessed.

RetiredWizard, 2024-02-13:
When the test code was actually placed in a code.py file accessing the web workflow page didn't interrupt it's execution until a file was uploaded via web workflow which caused an auto-reload. After the auto-reload, the code.py program started running again but then if the web workflow welcome page was accessed the code.py program would hang and the behavior described in my prior post was observed.

RetiredWizard, 2024-02-13:
With the test code running as code.py even when the code "hangs" it seems to time out and start up again in about a minute. 

Personally, since the web workflow interface no longer hangs and the code running on the microcontroller usually restarts in about 60 seconds (if it's interrupted at all), this feels resolved to me.

tannewt, 2024-02-14:
Ok, I'll close this. We'll get more testing on the web workflow and can open new issues when folks run into them.

PR #8699 (2024-01-25): Make SD cards available over web workflow
This changes storage.mount() to require that a mount point exist on the parent file system.

A bug in background tasks is also fixed where the function parameter is cleared on pending callbacks during "reset".

Disk usage is shown on the directory listing and changes based on the mounted file system. Writable is also loaded per-directory.

Fixes #8108. Fixes #8690. Fixes #8107.
commit 25e862d1 (2024-01-25): Make SD cards available over web workflow

This changes storage.mount() to require that a mount point exist
on the parent file system.

A bug in background tasks is also fixed where the function
parameter is cleared on pending callbacks during "reset".

Disk usage is shown on the directory listing and changes based on
the mounted file system. Writable is also loaded per-directory.

Fixes #8108. Fixes #8690. Fixes #8107.
closed 2024-02-14 by tannewt