adafruit/circuitpython #8692: Intermittant Delay while reading Capacitive Touch

makermelissa, 2023-12-04:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.5 on 2023-11-15; Adafruit-Qualia-S3-RGB666 with ESP32S3
```

### Code/REPL

```python
https://github.com/adafruit/Adafruit_CircuitPython_Qualia/blob/main/examples/qualia_paint.py
```

### Behavior

After about every 10 reads or so from the capacitive touch screen, there is a noticeable delay. You can really tell by running the paint program:
[image]

### Description

- Run Paint Program
- Draw

### Additional information

Switching to CP Alpha 2 fixes the issue. With CP Alpha 4, the problem is there, but you need to make sure the PYSTACK size and such is big enough to run it. My guess is it's related to the split heap stuff.

tannewt, 2023-12-04:
My guess is that we're doing a collect when the heap is full to see if we can allocate into it (and we can.) This will happen earlier with split heap than it did with the one big heap. The pauses should be shorter though.

tannewt, 2023-12-04:
Raising the starting heap size may give you enough buffer to capture a full finger drag before triggering the GC. You could also try running gc.collect() when a finger isn't active.

makermelissa, 2023-12-04:
Ok, I'll try those to see if it helps...

makermelissa, 2023-12-04:
Raising starting heap size did help. Garbage collection didn't help.

tannewt, 2024-01-24:
Do we want a larger default heap? This seems like a pretty specific case to change a default for.

tannewt, 2024-02-14:
> Raising starting heap size did help. Garbage collection didn't help.

What did you set it to? 128k? We could make it default on PSRAM builds.

makermelissa, 2024-02-21:
Tested this on latest CP 9 version and 512K seemed to work better than 128K or 256K.

tannewt, 2024-02-21:
Thanks for the update! I've moved this to 9.x.x. I'd like to see if any other uses like this crop up before we change the default heap size. Having it in stable should make it clearer.