adafruit/circuitpython #8697: espidf.IDFError: Invalid argument on 240x960 DotClock Display

makermelissa, 2023-12-05:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.5 on 2023-11-15; Adafruit-Qualia-S3-RGB666 with ESP32S3
```

### Code/REPL

```python
from displayio import release_displays
release_displays()

import time
import random
import displayio
import busio
import board
import dotclockframebuffer
from framebufferio import FramebufferDisplay

tft_pins = dict(board.TFT_PINS)
tft_timings = {
    "frequency": 16000000,
    "width": 240,
    "height": 960,
    "overscan_left": 120,

    "hsync_pulse_width": 8,
    "hsync_back_porch": 20,
    "hsync_front_porch": 20,
    "hsync_idle_low": False,

    "vsync_pulse_width": 8,
    "vsync_back_porch": 20,
    "vsync_front_porch": 20,
    "vsync_idle_low": False,

    "pclk_active_high": True,
    "pclk_idle_high": False,
    "de_idle_high": False,
}

init_sequence_hd371 = bytes((
    b'\xff\x05w\x01\x00\x00\x13'
    b'\xef\x01\x08'
    b'\xff\x05w\x01\x00\x00\x10'
    b'\xc0\x02w\x00'
    b'\xc1\x02\x11\x0c'
    b'\xc2\x02\x07\x02'
    b'\xcc\x010'
    b'\xb0\x10\x06\xcf\x14\x0c\x0f\x03\x00\n\x07\x1b\x03\x12\x10%6\x1e'
    b'\xb1\x10\x0c\xd4\x18\x0c\x0e\x06\x03\x06\x08#\x06\x12\x100/\x1f'
    b'\xff\x05w\x01\x00\x00\x11'
    b'\xb0\x01s'
    b'\xb1\x01|'
    b'\xb2\x01\x83'
    b'\xb3\x01\x80'
    b'\xb5\x01I'
    b'\xb7\x01\x87'
    b'\xb8\x013'
    b'\xb9\x02\x10\x1f'
    b'\xbb\x01\x03'
    b'\xc1\x01\x08'
    b'\xc2\x01\x08'
    b'\xd0\x01\x88'
    b'\xe0\x06\x00\x00\x02\x00\x00\x0c'
    b'\xe1\x0b\x05\x96\x07\x96\x06\x96\x08\x96\x00DD'
    b'\xe2\x0c\x00\x00\x03\x03\x00\x00\x02\x00\x00\x00\x02\x00'
    b'\xe3\x04\x00\x0033'
    b'\xe4\x02DD'
    b'\xe5\x10\r\xd4(\x8c\x0f\xd6(\x8c\t\xd0(\x8c\x0b\xd2(\x8c'
    b'\xe6\x04\x00\x0033'
    b'\xe7\x02DD'
    b'\xe8\x10\x0e\xd5(\x8c\x10\xd7(\x8c\n\xd1(\x8c\x0c\xd3(\x8c'
    b'\xeb\x06\x00\x01\xe4\xe4D\x00'
    b'\xed\x10\xf3\xc1\xba\x0ffwDUUDwf\xf0\xab\x1c?'
    b'\xef\x06\x10\r\x04\x08?\x1f'
    b'\xff\x05w\x01\x00\x00\x13'
    b'\xe8\x02\x00\x0e'
    b'\x11\x80x'
    b'\xe8\x82\x00\x0c\n'
    b'\xe8\x02@\x00'
    b'\xff\x05w\x01\x00\x00\x00'
    b'6\x01\x00'
    b':\x01f'
    b')\x80\x14'
    b'\xff\x05w\x01\x00\x00\x10'
    b'\xe5\x02\x00\x00'
))

board.I2C().deinit()
i2c = busio.I2C(board.SCL, board.SDA)
tft_io_expander = dict(board.TFT_IO_EXPANDER)
#tft_io_expander['i2c_address'] = 0x38 # uncomment for rev B
dotclockframebuffer.ioexpander_send_init_sequence(i2c, init_sequence_hd371, **tft_io_expander)
i2c.deinit()

BAR_WIDTH = 128
BAR_HEIGHT = 64

bitmap = displayio.Bitmap(BAR_WIDTH, 7*BAR_HEIGHT, 65535)
fb = dotclockframebuffer.DotClockFramebuffer(**tft_pins, **tft_timings)
display = FramebufferDisplay(fb, auto_refresh=False)

# Create a TileGrid to hold the bitmap
tile_grid = displayio.TileGrid(bitmap, pixel_shader=displayio.ColorConverter(input_colorspace=displayio.Colorspace.RGB565))

# Create a Group to hold the TileGrid
group = displayio.Group()

# Add the TileGrid to the Group
group.append(tile_grid)

# Add the Group to the Display
display.root_group = group

display.auto_refresh = True

for i in range(BAR_WIDTH):
    b = i >> 3
    g = (i >> 2) << 5
    r = b << 11
    for j in range(BAR_HEIGHT):
        bitmap[i, j] = b
        bitmap[i, j+BAR_HEIGHT*1] = b|g
        bitmap[i, j+BAR_HEIGHT*2] = g
        bitmap[i, j+BAR_HEIGHT*3] = g|r
        bitmap[i, j+BAR_HEIGHT*4] = r
        bitmap[i, j+BAR_HEIGHT*5] = r|b
        bitmap[i, j+BAR_HEIGHT*6] = r|g|b

# Loop forever so you can enjoy your image
while True:
    time.sleep(1)
    display.auto_refresh = False
    group.x = random.randint(0, display.width - BAR_WIDTH)
    group.y = random.randint(0, display.height - BAR_HEIGHT*7)
    display.auto_refresh = True
    pass
```

### Behavior

```
code.py output:
Traceback (most recent call last):
  File "code.py", line 93, in <module>
espidf.IDFError: Invalid argument
```

### Description

If you look at the product page for the display at https://www.adafruit.com/product/5799, under Technical Details there are some data sheets. Under Timing Diagram, the full width (overscan_left + width) should be 360 pixels, but any combination above 320 pixels results in the IDFError and setting them to a combination of 320 pixels leave 40 cutoff pixels. Also, the the values in the description are accurate and act as a confirmation of my findings.

jepler, 2023-12-05:
It looks like something internal to IDF is leading to a "multiple of 16 pixels width" requirement, though I didn't find the specific line of code that does it. This may be related to the DMA unit size.

jepler, 2023-12-05:
I'll look into that, yes.

jepler, 2023-12-05:
Here's the underlying error:
```
E (3410) cache: esp_cache_msync(35): size isn't aligned with the data cache line size (32)B
E (3410) lcd_panel.rgb: rgb_panel_draw_bitmap(742): flush cache buffer failed
```
This confirms there's a 32 bytes / 16 pixels requirement for line lengths.

I'll just round up the (width + overscan_left) to a multiple of 16. PR tomorrow.

commit 15ba61ef (2023-12-06): rgbframebuffer: round up internal size to multiple of 16 pixels

For a (width + left_overscan) of 360, e.g., debug info shows the following:
```
E (3410) cache: esp_cache_msync(35): size isn't aligned with the data cache line size (32)B
E (3410) lcd_panel.rgb: rgb_panel_draw_bitmap(742): flush cache buffer failed
```

Internally add extra pixels as needed at the end of a pixel row,
like an automatic overscan_right.

This may not be compatible with all panels, but there's not really
another alternative with this peripheral. So let's try this!

Closes: #8697
closed 2023-12-06 by jepler