adafruit/circuitpython #8706: Help needed with adding an ESP32 board

szsoftware, 2023-12-09:
Hi guys,
I try to "port" circuitpython to a cheap, bit older board:

Heltec WifiKit 32, featuring esp32 + Oled.
Pinout, Schematics and Specs here:  [https://resource.heltec.cn/download/WiFi_Kit_32/](url)

So far I got the circuitpython build framework incl. esp-idf up and running and succesfully built and flashed resulting firmware.

However, I don't get a Repl or human readable Console output after the bootloader init.

This is what I get in CuteCom:
[image]

However, I don't know how to debug something.

Currently I put some code from the esp-idf blinky example right into the board_init() function in board.c:

```
void board_init(void) {
    reset_board();
    /* Blinky Begin */

    /* Configure the peripheral according to the LED type */
    configure_led();

    for (int i = 5; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        
        ESP_LOGI(TAG, "Turning the LED %s!", s_led_state == true ? "ON" : "OFF");
        blink_led();
        /* Toggle the LED state */
        s_led_state = !s_led_state;
        vTaskDelay(BLINK_PERIOD / portTICK_PERIOD_MS);
    }
    
    for (int i = 10; i >= 0; i--) {
        printf("Flashing LED %d times...\n", i);
        blink_led();
        /* Toggle the LED state */
        s_led_state = !s_led_state;
        vTaskDelay(100 / portTICK_PERIOD_MS);
    }
    
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
    /* Blinky End */
}
```

At least this is working. However, I get the bootloader messages only when pressing the reset button on the device.

So, I don't know where and what to configure and would appreciate any hints and help.

Many thanks in advance!

dhalbert, 2023-12-09:
> `esp_restart();`

If you put this in `board_init()`, it's going to cause a reset each time the board starts up, so you'll end up with a boot loop.

We'd have to see the rest of your code to see what else is wrong.

dhalbert, 2023-12-09:
Your board seems close to this, which has already been added: https://circuitpython.org/board/heltec_esp32s3_wifi_lora_v3/

szsoftware, 2023-12-09:
Hi and thank you so much for answering that quickly.

The esp_restart() is only for testing purpose.

I expected at least seeing the printf() output, but instead I just get garbage.

Thus I suspect having some esp-idf uart stuff not configured right.

The mentioned Heltec Lora 32 V3 board is a ESP32-S3 and has some other relevant GPIO pinout.

Okay, I try to do the github stuff with forking and so on.

I hope that will help.

dhalbert, 2023-12-09:
I see - there are multiple "WiFi" boards, and the original ESP32 one (yours) seems to be discontinued

There seem to be several versions of the original WiFi, with different-sized flash chips. Which one do you have? You will need to set `CIRCUITPY_ESP_FLASH_SIZE` and the other values properly

szsoftware, 2023-12-09:
Hi Dan,
I got the github fork, and commit workflow managed.
Also tried the best I can to introduce the board and descriptions.

Highly appreciate if you took a look on it.

szsoftware, 2023-12-11:
I got some further success:
After compiling the esp-idf system/console  example and having similar behaviour (garbage on the console) I suspected indeed UART misconfiguration and found out, that the CP2102 of this board is connected to Custom UART 1, so using the default led to the issues described.

Now I have console output, yayyy!
Let's fight the next issues...

'''
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:3
load:0x3fff0030,len:7312
load:0x40078000,len:15584
load:0x40080400,len:4
load:0x40080404,len:4104
entry 0x40080670
<0x1b>[0;32mI (36) boot: ESP-IDF 60ff95f5 2nd stage bootloader<0x1b>[0m
<0x1b>[0;32mI (36) boot: compile time Dec 11 2023 17:31:14<0x1b>[0m
<0x1b>[0;32mI (36) boot: Multicore bootloader<0x1b>[0m
<0x1b>[0;32mI (39) boot: chip revision: v1.0<0x1b>[0m
<0x1b>[0;32mI (43) boot.esp32: SPI Speed      : 26.7MHz<0x1b>[0m
<0x1b>[0;32mI (48) boot.esp32: SPI Mode       : DIO<0x1b>[0m
<0x1b>[0;32mI (53) boot.esp32: SPI Flash Size : 8MB<0x1b>[0m
<0x1b>[0;32mI (57) boot: Enabling RNG early entropy source...<0x1b>[0m
<0x1b>[0;32mI (63) boot: Partition Table:<0x1b>[0m
<0x1b>[0;32mI (66) boot: ## Label            Usage          Type ST Offset   Length<0x1b>[0m
<0x1b>[0;32mI (73) boot:  0 nvs              WiFi data        01 02 00009000 00005000<0x1b>[0m
<0x1b>[0;32mI (81) boot:  1 otadata          OTA data         01 00 0000e000 00002000<0x1b>[0m
<0x1b>[0;32mI (88) boot:  2 ota_0            OTA app          00 10 00010000 00200000<0x1b>[0m
<0x1b>[0;32mI (96) boot:  3 ota_1            OTA app          00 11 00210000 00200000<0x1b>[0m
<0x1b>[0;32mI (103) boot:  4 user_fs          Unknown data     01 81 00410000 003f0000<0x1b>[0m
<0x1b>[0;32mI (111) boot: End of partition table<0x1b>[0m
<0x1b>[0;32mI (115) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=4f684h (325252) map<0x1b>[0m
<0x1b>[0;32mI (260) esp_image: segment 1: paddr=0005f6ac vaddr=3ff80063 size=00008h (     8) load<0x1b>[0m
<0x1b>[0;32mI (261) esp_image: segment 2: paddr=0005f6bc vaddr=3ffb0000 size=0095ch (  2396) load<0x1b>[0m
<0x1b>[0;32mI (267) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=111f04h (1122052) map<0x1b>[0m
<0x1b>[0;32mI (746) esp_image: segment 4: paddr=00171f2c vaddr=3ffb095c size=03814h ( 14356) load<0x1b>[0m
<0x1b>[0;32mI (753) esp_image: segment 5: paddr=00175748 vaddr=40080000 size=18358h ( 99160) load<0x1b>[0m
<0x1b>[0;32mI (800) esp_image: segment 6: paddr=0018daa8 vaddr=400c0000 size=00064h (   100) load<0x1b>[0m
<0x1b>[0;32mI (800) esp_image: segment 7: paddr=0018db14 vaddr=50000ff0 size=00c00h (  3072) load<0x1b>[0m
<0x1b>[0;32mI (821) boot: Loaded app from partition at offset 0x10000<0x1b>[0m
<0x1b>[0;32mI (821) boot: Disabling RNG early entropy source...<0x1b>[0m
<0x1b>[0;32mI (832) cpu_start: Multicore app<0x1b>[0m
<0x1b>[0;32mI (833) cpu_start: Pro cpu up.<0x1b>[0m
<0x1b>[0;32mI (833) cpu_start: Starting app cpu, entry point is 0x40083f20<0x1b>[0m
<0x1b>[0;32mI (0) cpu_start: App cpu up.<0x1b>[0m
<0x1b>[0;32mI (853) cpu_start: Pro cpu start user code<0x1b>[0m
<0x1b>[0;32mI (853) cpu_start: cpu freq: 240000000 Hz<0x1b>[0m
<0x1b>[0;32mI (853) cpu_start: Application information:<0x1b>[0m
<0x1b>[0;32mI (858) cpu_start: Project name:     circuitpython<0x1b>[0m
<0x1b>[0;32mI (863) cpu_start: App version:      9.0.0-alpha.5-291-gbf5eaa6038-d<0x1b>[0m
<0x1b>[0;32mI (870) cpu_start: Compile time:     Dec 11 2023 17:30:53<0x1b>[0m
<0x1b>[0;32mI (876) cpu_start: ELF file SHA256:  745e3ecbfa42be26...<0x1b>[0m
<0x1b>[0;32mI (882) cpu_start: ESP-IDF:          60ff95f5<0x1b>[0m
<0x1b>[0;32mI (887) cpu_start: Min chip rev:     v0.0<0x1b>[0m
<0x1b>[0;32mI (892) cpu_start: Max chip rev:     v3.99 <0x1b>[0m
<0x1b>[0;32mI (897) cpu_start: Chip rev:         v1.0<0x1b>[0m
<0x1b>[0;32mI (902) heap_init: Initializing. RAM available for dynamic allocation:<0x1b>[0m
<0x1b>[0;32mI (909) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM<0x1b>[0m
<0x1b>[0;32mI (915) heap_init: At 3FFBC620 len 000239E0 (142 KiB): DRAM<0x1b>[0m
<0x1b>[0;32mI (921) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM<0x1b>[0m
<0x1b>[0;32mI (927) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM<0x1b>[0m
<0x1b>[0;32mI (934) heap_init: At 40098358 len 00007CA8 (31 KiB): IRAM<0x1b>[0m
<0x1b>[0;32mI (942) spi_flash: detected chip: winbond<0x1b>[0m
<0x1b>[0;32mI (945) spi_flash: flash io: dio<0x1b>[0m
<0x1b>[0;33mW (949) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`<0x1b>[0m
<0x1b>[0;33mW (959) pcnt(legacy): legacy driver is deprecated, please migrate to `driver/pulse_cnt.h`<0x1b>[0m
<0x1b>[0;33mW (968) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`<0x1b>[0m
[... 102 lines cut ...]
<0x1b>[0;32mI (1132) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 <0x1b>[0m
<0x1b>[0;32mI (1142) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 <0x1b>[0m
<0x1b>[0;32mI (1152) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 <0x1b>[0m
<0x1b>[0;32mI (1162) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 <0x1b>[0m
<0x1b>[0;32mI (1172) gpio: GPIO[20]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 <0x1b>[0m
[... cut ...]
<0x1b>[0;32mI (1302) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 <0x1b>[0m
<0x1b>[0;31mE (1312) timer_group: timer_deinit(343): HW TIMER NEVER INIT ERROR<0x1b>[0m
<0x1b>[0;31mE (1322) timer_group: timer_deinit(343): HW TIMER NEVER INIT ERROR<0x1b>[0m
<0x1b>[0;31mE (1332) timer_group: timer_deinit(343): HW TIMER NEVER INIT ERROR<0x1b>[0m
<0x1b>[0;31mE (1342) timer_group: timer_deinit(343): HW TIMER NEVER INIT ERROR<0x1b>[0m
<0x1b>[0;32mI (1382) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 <0x1b>[0m
'''

szsoftware, 2023-12-11:
Next focus: After disabling DEBUG in mpconfigboard.mk the now relevant part seems to be:

Serial console setup (hanging)

szsoftware, 2023-12-11:
Got it more than basically working. Current status: REPL works, WebREPL works. Thus, wifi works.

Currently testing display init code.

szsoftware, 2023-12-12:
Got oled display initializing.

szsoftware, 2023-12-12:
I probably made something wrong in the PullRequest procedure.

CI checks fail. I accidentally merged main branch into my fork/feature branch after I commited the last changes.
Can that be the source of "Some checks were not successful" in Build CI?

Any help thankful appreciated!

dhalbert, 2024-01-24:
Closing to continue discussion in #8713.

closed 2024-01-24 by dhalbert