adafruit/circuitpython #8709: Option to disable safemode user trigger.

bill88t, 2023-12-11:
When testing stuff, running experimental builds or other weird stuff, safemode is great.
Now that `safemode.py` exists, it can do so much more.
But on production (for those few that want to ship CP), it may be desirable to not permit the user to enter safe mode through the boot button.

`CIRCUITPY_DISABLE_SAFEMODE_TRIGGER`
This would disable the ability to press the boot button to enter safe mode.
This would not stop the code from entering safe mode.

What do you about it?
I can probably do it myself.
I just want feedback before I do it.

With it, it would be possible to outright lock the user out and require a board wipe (which in this context is intended behaviour).
It's not something many users will or should use, but I do think it should exist.

dglaude, 2023-12-11:
`safemode.py` is new to me, but one option would be to put some self destruction code in safemode.py or just restart if someone tries to enter safemode... so I am not sure you need `CIRCUITPY_DISABLE_SAFEMODE_TRIGGER` if those work around are OK.

What would it be, a compilation option so that if you create custom CircuitPython firmware (that you rename another name) you can enable that flag?
Or is it something in `boot.py` or `settings.toml`.

There are a few things that people that want to do device with CircuitPython want to change, such as remove the blinka logo, accelerate the boot process to avoid waiting for safemode (or other) boot trick... not sure how those idea or PR have been treated.

bill88t, 2023-12-11:
`safemode.py` does not work when safe mode is triggered by the boot button.

Having a fork / downstream build is exactly what I'm trying to avoid.

Having a "fast boot" of sorts would require build parameters (since we can't read fs that early) and is not something I am willing to do.

dhalbert, 2023-12-11:
To be clear, you want this to be settable in `settings.toml`?

This is handled in `supervisor/shared/safe_mode.py`, `wait_for_safe_mode_reset()`.

> `safemode.py` does not work when safe mode is triggered by the boot button.

More precisely, `safemode.py` is deliberately not run in that circumstance.

bill88t, 2023-12-11:
> To be clear, you want this to be settable in settings.toml?

I mean, is there a better way to permenantly set a flag?
nvm would be nice here, as we could make an efuse of sorts.
A module could set a specific bit, and early boot could read it.
But since we don't have a region (to my knowledge) specifically for internal flags, it's not something I wanna do.

> This is handled in supervisor/shared/safe_mode.py, wait_for_safe_mode_reset().

I was thinking of just blocking off repl and the usb filesystem in case a safe mode was manually triggered.

This wouldn't need to be done during early boot.
It could be done when boot.py is normally run.

Something like: 
```
Running in safe mode! Not running saved code.

You are in safe mode because:
You pressed the BOOT button at start up
Press reset to exit safe mode.

Safe mode access has been restricted.
```
And then it would lock up, requiring a reset.

shubham0x13, 2024-11-11:
> When testing stuff, running experimental builds or other weird stuff, safemode is great. Now that `safemode.py` exists, it can do so much more. But on production (for those few that want to ship CP), it may be desirable to not permit the user to enter safe mode through the boot button.
> 
> `CIRCUITPY_DISABLE_SAFEMODE_TRIGGER` This would disable the ability to press the boot button to enter safe mode. This would not stop the code from entering safe mode.
> 
> What do you think about it? I can probably do it myself. I just want some feedback before I do it.
> 
> With it, it would be possible to outright lock the user out and require a board wipe (which in this context is intended behaviour). It's not something many users will or should use, but I do think it should exist.

Did you create a fork for it? I want the same thing.

dhalbert, 2024-11-11:
There is already `CIRCUITPY_SKIP_SAFE_MODE_WAIT`, which you can enable to prevent the one second interval during which you can press the reset button again. On many boards that's the only way to enter safe mode.

On boards where you can hold down a button to enter safe mode, you can disable entering safe mode by pressing a button by removing `board_requests_safe_mode()`  from `board.c` for the board, or for a few ports, from `port.c`.

Which board are you thinking of.

shubham0x13, 2024-11-11:
> Which board are you thinking of.

I'm thinking of using the Waveshare RP2040 Zero. I want to ensure that my device never enters safe mode.

dhalbert, 2024-11-11:
> I want to ensure that my device **never** enters safe mode.

There are software and hardware failures that will trigger safe mode. Without that there may be a loop of continuous errors. You could turn that off by modifying the safe mode code.

shubham0x13, 2024-11-12:
> There are software and hardware failures that will trigger safe mode.

Is there a way to trigger safe mode by connecting certain pins on the board if the board doesn't have a button to boot into safe mode?

bill88t, 2024-11-12:
> > There are software and hardware failures that will trigger safe mode.
> 
> Is there a way to trigger safe mode by connecting certain pins on the board if the board doesn't have a button to boot into safe mode?

On RP2, no.
On ESP32, ESP32-SX, yes by using pin IO0.
On ESP32-CX, ESP32-HX, yes by using pin IO9.
On nRF and SAMD, by using a double reset.