adafruit/circuitpython #8712: Adafruit ESP32-S3 Rev TFT displayio broken

DJDevon3, 2023-12-11:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; Adafruit Feather ESP32-S3 Reverse TFT with ESP32S3
Board ID:adafruit_feather_esp32s3_reverse_tft

adafruit-circuitpython-bundle-8.x-mpy-20231210
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2021 Tim C for Adafruit Industries
# SPDX-License-Identifier: MIT
"""
CircuitPython simple text display demo
"""
import board
import terminalio
from adafruit_display_text import bitmap_label
import displayio

displayio.release_displays()
display = board.DISPLAY
print(display)
# Update this to change the text displayed.
text = "Hello, World!"
# Update this to change the size of the text displayed. Must be a whole number.
scale = 3

text_area = bitmap_label.Label(terminalio.FONT, text=text, scale=scale)
text_area.x = 10
text_area.y = 10

text_group = displayio.Group()
main_group = displayio.Group()

main_group.append(text_group)
main_group.append(text_area)
display.show(main_group)

while True:
    pass
```

### Behavior

`AttributeError: can't set attribute 'root_group'` provides same error if attempting to use display.show()

### Description

Could not get the onboard display to display with either display.show() or root_group

### Additional information

Bug discovered by Lindahl Engineering.  I replicated the issue on an identical Adafruit ESP32-S3 Rev TFT board with 8.2.9 and latest 8.x bundle.

dhalbert, 2023-12-11:
This is deliberate. When you `displayio.release_displays()`, `board.DISPLAY` is set to `None`, because you gave up the display.
```py
Adafruit CircuitPython 8.2.9 on 2023-12-06; Adafruit Feather ESP32-S2 TFT with ESP32S2
>>> 
>>> import board
>>> board.DISPLAY
<Display>
>>> import displayio
>>> displayio.release_displays()
>>> board.DISPLAY
None
```
This is not new. The same thing happens on 7.3.3.

closed 2023-12-11 by dhalbert

DJDevon3, 2023-12-11:
Same error occurs without `displayio.release_displays()`

DJDevon3, 2023-12-11:
nvm you're right just needed a hard reset

```py
import board
import terminalio
from adafruit_display_text import bitmap_label
import displayio

display = board.DISPLAY
print(display)
# Update this to change the text displayed.
text = "Hello, World!"
# Update this to change the size of the text displayed. Must be a whole number.
scale = 1

text_area = bitmap_label.Label(terminalio.FONT, text=text, scale=scale)
text_area.x = 10
text_area.y = 10

text_group = displayio.Group()
main_group = displayio.Group()

my_display_group = displayio.Group()
my_display_group.append(text_group)
my_display_group.append(text_area)
display.show(my_display_group)

while True:
    pass
``` this works

dhalbert, 2023-12-11:
Yes, releasing the display is persistent.