adafruit/circuitpython #8730: MEMENTO does not have STEMMA_I2C() or microphone pin defined

BlitzCityDIY, 2023-12-19:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.6 on 2023-12-12; Adafruit Camera with ESP32S3
```

### Code/REPL

```python
>>> import board
>>> dir(board)
['__class__', '__name__', '__dict__', 'A0', 'A1', 'BATTERY_MONITOR', 'BUTTON', 'CAMERA_DATA', 'CAMERA_DATA2', 'CAMERA_DATA3', 'CAMERA_DATA4', 'CAMERA_DATA5', 'CAMERA_DATA6', 'CAMERA_DATA7', 'CAMERA_DATA8', 'CAMERA_DATA9', 'CAMERA_HREF', 'CAMERA_PCLK', 'CAMERA_PWDN', 'CAMERA_RESET', 'CAMERA_VSYNC', 'CAMERA_XCLK', 'CARD_CS', 'DISPLAY', 'I2C', 'IRQ', 'MISO', 'MOSI', 'NEOPIXEL', 'SCK', 'SCL', 'SDA', 'SPEAKER', 'SPI', 'TFT_BACKLIGHT', 'TFT_CS', 'TFT_DC', 'TFT_RESET', 'board_id']
>>>
```

### Behavior

i'm currently working on the MEMENTO board guide and found that CircuitPython is missing two expected pin attributes. The most important is the analog microphone attached to GPIO2. The second is STEMMA_I2C. The STEMMA port is on the same SCL/SDA pins as I2C so i wasn't sure if that was intentional or not. i'm happy to do a PR to add them but wanted to check first.

jepler, 2023-12-19:
STEMMA_I2C is "yes" and while I don't know if circuitpython can actually use the mic (we don't have audioin or pwmaudioin on espressif do we?) I don't see where it hurts to add.

deshipu, 2023-12-19:
There is `analogbufio` https://docs.circuitpython.org/en/latest/shared-bindings/analogbufio/index.html

closed 2023-12-19 by jepler