adafruit/circuitpython #8733: `RGBMatrix` display showing white vertical lines when re-initialized on ESP32-S2

arduino12, 2023-12-19:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; S2Mini with ESP32S2-S2FN4R2
```

### Code/REPL

```python
import board
import displayio
from framebufferio import FramebufferDisplay
from rgbmatrix import RGBMatrix
from terminalio import FONT

def get_display():
    matrix = RGBMatrix(
        width=128,
        bit_depth=1,
        rgb_pins=[board.IO1, board.IO40, board.IO2, board.IO4, board.IO38, board.IO6],
        addr_pins=[board.IO8, board.IO34, board.IO10, board.IO21, board.IO36],
        clock_pin=board.IO13,
        latch_pin=board.IO17,
        output_enable_pin=board.IO14)
    return FramebufferDisplay(matrix)

# deinit previous displays
displayio.release_displays()

# first instance works
display = get_display() 

# draw a white 4x4 square
g = displayio.Group()
p = displayio.Palette(1)
p[0] = 0xffffff
b = displayio.Bitmap(4, 4, 1)
tg = displayio.TileGrid(b, pixel_shader=p, x=10, y=10)
g.append(tg)
display.root_group = g

# I use Thonny, so pressing F5 will run the code again (after a soft-restart)
# and the bug will occur..

# I can also cause the bug by repeating the above lines like this:
input('Press ENTER to re-initialized the display to cause the BUG!')

# deinit first instance - display turns off,
# `matrix` and `display` become `None`
displayio.release_displays()

# BUG: the display shows white vertical lines!
display = get_display()
```

### Behavior

Display works on the first time,
But after pressing ENTER - it is re-initialized and the bug occurs.

Running the code again won't work either- the only way to fix it is with a hardware-reset to the board.

### Description

Bug occurs only on ESP32-S2 (it seems to draw the REPL text for a split second before the white vertical lines),
on my ESP32-S3 it works as expected.  

I tried using `DigitalInOut` on GPIOs, `gc.collect()`, `supervisor.reload()`,  
I tried adding to code.py: `status_bar.console = False`, `status_bar.display = False`..
Nothing seems to fix it - only a hardware-reset (or a `__import__('microcontroller').reset()`).

Same thing here:
https://github.com/adafruit/circuitpython/issues/8303#issuecomment-1836480620

It makes it hard to develop the code since a hardware-reset is needed before every re-run.

dhalbert, 2023-12-23:
Do you see this same behavior on 8.2.9?

arduino12, 2023-12-23:
Yes on both 8.2.9 and 9.0.0,
Made a video of running the code above:
(first time I run it- it work and draw a small rect, after I press ENTER and on every next run - showing white vertical lines):

https://github.com/adafruit/circuitpython/assets/11454280/a04b629f-8e08-4192-bcfc-32ca9bc83588

tannewt, 2024-01-17:
Please post a picture. The video file is corrupt.

arduino12, 2024-01-18:
Here is a screenshot from the video on this page:
[image]

Let me know if I can provide more help :)

mfinkle, 2024-03-15:
I ran into the same problem on the two Feather ESP32-S2 boards I have. I narrowed down the regression to something that happened between these two `UF2` files:
* `adafruit-circuitpython-adafruit_feather_esp32s2-en_US-8.2.0-beta.1.uf2`  [WORKS]
* `adafruit-circuitpython-adafruit_feather_esp32s2-en_US-8.2.0-rc.0.uf2` [FAILS]

I tested by installing the UF2, running the code, saving the same code back to the device (forcing a soft-reboot). As mentioned above, the initial load works, but forcing the soft-reboot causes the vertical lines to display.