adafruit/circuitpython #8734: some Indexed BMP Images look corrupted

carson-coder, 2023-12-20:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.6 on 2023-12-12; Adafruit Matrix Portal M4 with samd51j19
```

### Code/REPL

```python
import board
import displayio
import framebufferio
import rgbmatrix

displayio.release_displays() # Release current display, we'll create our own

# Create RGB matrix object for a chain of four 64x32 matrices tiled into
# a single 128x64 pixel display -- two matrices across, two down, with the
# second row being flipped. width and height args are the combined size of
# all the tiled sub-matrices. tile arg is the number of rows of matrices in
# the chain (horizontal tiling is implicit from the width argument, doesn't
# need to be specified, but vertical tiling must be explicitly stated).
# The serpentine argument indicates whether alternate rows are flipped --
# cabling is easier this way, downside is colors may be slightly different
# when viewed off-angle. bit_depth and pins are same as other examples.
MATRIX = rgbmatrix.RGBMatrix(
    width=128, height=64, bit_depth=6, tile=2, serpentine=True,
    rgb_pins=[board.MTX_R1,
              board.MTX_G1,
              board.MTX_B1,
              board.MTX_R2,
              board.MTX_G2,
              board.MTX_B2],
    addr_pins=[board.MTX_ADDRA,
               board.MTX_ADDRB,
               board.MTX_ADDRC,
               board.MTX_ADDRD],
    clock_pin=board.MTX_CLK, latch_pin=board.MTX_LAT,
    output_enable_pin=board.MTX_OE)

# Associate matrix with a Display to use displayio features
DISPLAY = framebufferio.FramebufferDisplay(MATRIX, auto_refresh=False,
                                           rotation=0)

# Load BMP image, create Group and TileGrid to hold it
FILENAME = "/code/team0_logos/NYJ.bmp"

# CircuitPython 6 & 7 compatible
BITMAP = displayio.OnDiskBitmap(open(FILENAME, "rb"))
TILEGRID = displayio.TileGrid(
    BITMAP,
    pixel_shader=getattr(BITMAP, 'pixel_shader', displayio.ColorConverter()),
    tile_width=BITMAP.width,
    tile_height=BITMAP.height
)

# # CircuitPython 7+ compatible
# BITMAP = displayio.OnDiskBitmap(FILENAME)
# TILEGRID = displayio.TileGrid(
#     BITMAP,
#     pixel_shader=BITMAP.pixel_shader,
#     tile_width=BITMAP.width,
#     tile_height=BITMAP.height
# )

GROUP = displayio.Group()
GROUP.append(TILEGRID)
DISPLAY.root_group = GROUP
DISPLAY.refresh()

# Nothing interactive, just hold the image there
while True:
    pass
```

### Behavior

Images look corrupted / weird (look at uploaded file)
[image]

### Description

Only happens with some bitmap files. Maybe because the bitmaps palette index 0 is while and 255 is black?

### Additional information

I was messing around with my own build of circuit python and discovered that it looks like `displayio_convert_color' treats the color as a rgb332 color because colorspace.depth is 8 right. Or would it be 24.

dhalbert, 2023-12-20:
Could you please include the `.bmp` in your report, not using `file.io`, which is only temporary? If you can't upload a `.bmp`, zip it up and upload it and add that as a file. See "Paste, drop, or click to add files" below the post editing box.

carson-coder, 2023-12-20:
Ok I will edit the post

dhalbert, 2023-12-20:
BTW, `identify -verbose NYJ.bmp` and similar gives you all kind of info about the file, including its indexed colormap. `identify` is from `ImageMagick`.

carson-coder, 2023-12-20:
I made a python script to take a indexed bmp image and output a picture as if that bmp was rgb332 and this is what it got
[image]
so its not just thinking the image is rgb332

carson-coder, 2023-12-20:
> BTW, `identify -verbose NYJ.bmp` and similar gives you all kind of info about the file, including its indexed colormap. `identify` is from `ImageMagick`.

When I run that I get

```Image:
  Filename: NYJ.bmp
  Permissions: rwxrwxrwx
  Format: BMP (Microsoft Windows bitmap image)
  Class: PseudoClass
  Geometry: 32x32+0+0
  Resolution: 37.79x37.79
  Print size: 0.846785x0.846785
  Units: PixelsPerCentimeter
  Colorspace: sRGB
  Type: Palette
  Base type: Undefined
  Endianness: Undefined
  Depth: 8-bit
  Channel depth:
    red: 8-bit
    green: 8-bit
    blue: 8-bit
  Channel statistics:
    Pixels: 1024
    Red:
      min: 0  (0)
      max: 255 (1)
      mean: 21.0635 (0.0826019)
      standard deviation: 51.9172 (0.203597)
      kurtosis: 8.30727
      skewness: 2.91146
      entropy: 0.364514
    Green:
      min: 0  (0)
      max: 255 (1)
      mean: 28.1787 (0.110505)
      standard deviation: 54.5796 (0.214038)
      kurtosis: 5.64251
      skewness: 2.41808
      entropy: 0.512323
    Blue:
      min: 0  (0)
      max: 255 (1)
      mean: 24.7979 (0.0972465)
      standard deviation: 54.4788 (0.213642)
      kurtosis: 6.74497
      skewness: 2.67332
      entropy: 0.477403
  Image statistics:
    Overall:
      min: 0  (0)
      max: 255 (1)
      mean: 24.68 (0.0967844)
      standard deviation: 53.6585 (0.210426)
      kurtosis: 6.79313
      skewness: 2.6553
      entropy: 0.451413
  Colors: 57
  Histogram:
           608: (0,0,0) #000000 black
             9: (0,4,0) #000400 srgb(0,4,0)
             1: (0,8,0) #000800 srgb(0,8,0)
            27: (0,12,0) #000C00 srgb(0,12,0)
            44: (0,12,8) #000C08 srgb(0,12,8)
             4: (0,16,0) #001000 srgb(0,16,0)
            44: (0,16,8) #001008 srgb(0,16,8)
             5: (0,20,6) #001406 srgb(0,20,6)
            12: (0,21,21) #001515 srgb(0,21,21)
             3: (0,24,7) #001807 srgb(0,24,7)
             9: (0,28,11) #001C0B srgb(0,28,11)
[... 313 lines cut ...]
  Matte color: grey74
  Transparent color: black
  Interlace: None
  Intensity: Undefined
  Compose: Over
  Page geometry: 32x32+0+0
  Dispose: Undefined
  Iterations: 0
  Compression: RLE
  Orientation: Undefined
  Properties:
    date:create: 2023-12-19T17:12:40+00:00
    date:modify: 2023-12-19T17:12:40+00:00
    date:timestamp: 2023-12-20T15:26:18+00:00
    signature: be9b4d95bbd6fc7ed10efda845f1a0a6ff9e06988f335b17197d4a7a6d3eaaa3
  Artifacts:
    filename: NYJ.bmp
    verbose: true
  Tainted: False
  Filesize: 2108B
  Number pixels: 1024
  Pixels per second: 6.09477MB
  User time: 0.000u
  Elapsed time: 0:01.000
  Version: ImageMagick 6.9.12-98 Q16 x86_64 18038 https://legacy.imagemagick.org```

dhalbert, 2023-12-20:
I reproduced your problem on a single panel with the Matrix Portal S3 (I can't find my Matrix Portal M4 at the moment).

I saw a similarly corrupted image. However, when I read the file into GIMP and then export it as .bmp again, with no changes, it displays properly:
[NYJ1.bmp.zip](https://github.com/adafruit/circuitpython/files/13729188/NYJ1.bmp.zip)
Try that.
The indexed palette as described by `identify -verbose` is exactly the same. However, there are these differences. The Compression difference might be the significant one. The new file is just slightly larger.

[image]

dhalbert, 2023-12-20:
I tested with 8.2.9 and 9.0.0-alpha.6: no difference. BTW, you can also use the `# # CircuitPython 7+ compatible` code in the example instead of the older code, but both work.

dhalbert, 2023-12-20:
This is also helpful: https://www.metadata2go.com/. Note that the size is 946 for NYJ.bmp, and it's RLE, and NYJ1.bmp is None, and the size is 1024. It seems like there's a misinterpretation of RLE for some reason for these files. Did you make them exactly according to the recipe in the scoreboard guide?

dhalbert, 2023-12-20:
OK, this looks like #7450, which reports the same thing. We should complain about compressed files but we're not checking carefully enough.

carson-coder, 2023-12-20:
Thank you. Is there a way to decompress the images in python/bash. I have a lot of image that need to be decompressed.

dhalbert, 2023-12-20:
On the command line:
`convert NYJ.bmp -compress None NYJ-uncompressed.bmp`