adafruit/circuitpython #8739: SPI bus management T-DECK Lilygo

Freyr86, 2023-12-21:
I am working on programming the Lilygo T-Deck. I have chosen CircuitPython because of its flexibility, which is welcome for conducting tests.

In this module, there are two systems that operate on the SPI bus: the screen (ST7789) and a LORA transceiver (SX1262).

During the boot of CircuitPython on this board, the REPL activates on the screen, so the SPI is initialized for the screen. If I want to send data to the LORA module, I need to have the name of the object managing the SPI bus.

For example, if I make the following code call to prepare the SPI for using the LORA module:

`
import busio
import board
testSPI = busio.SPI(clock = board.SCK, MOSI=board.MOSI, MISO=board.MISO)
`

The error that comes up is the following:

`ValueError: IO40 en cours d'utilisation`

That makes perfect sense since the bus is used for the screen.

I can, of course, use the command
`displayio.release_displays()`

But that would force me to redo its initialization. However, if I could access the busio.SPI object used for the REPL, I could simply use the CS outputs to operate both the screen and the LORA module.

Do you have any ideas?

deshipu, 2023-12-21:
You can either use `board.SPI()`, which will create a new SPI object if it doesn't exist yet, or give you the existing one if it's already used, or you can get it from the display itself with `board.DISPLAY.display_bus.spi_bus`.

Freyr86, 2023-12-21:
It seems that the board.SPI() function is apparently working. I will have to investigate further because I have other issues in the library I'm using. In any case, a big thank you. I've been stuck on this issue for a while.

Regarding the board.DISPLAY.display_bus.spi_bus, it is not compatible with the board. I mention it here for those who might have the same issue.

Thanks, @deshipu

closed 2023-12-21 by Freyr86