adafruit/circuitpython #8741: Crash into the HardFault_Handler, bricks rp2040

schaefer01, 2023-12-21:
### CircuitPython version

```python
8.2.4. on rp2040
```

### Code/REPL

```python
I have 2 versions of my program/app, one that leads to the hard fault and an earlier version that does not. I can send both files to you but I also did a delta and the difference is an import of adafruit_ntp that has a May 2023 date on it
```

### Behavior

Auto-reload is off.
Running in safe mode! Not running saved code.
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.

### Description

At this point, I cycled power several times with no improvement and still had a circuitpy drive after power up. I then tried a reflash using Feather_RP2040.uf2 but this did not work (i.e. no circuitypy drive following the reflash), I then tried a flash_nuke followed by flash boot but still no circuitpy drive. So I basically bricked my raspberry pi featherwing. I did this again on another featherwing because I couldn't believe it. I then swapped featherwings agin to a third featherwing but kept the earller version of the app, copied the code.py off of the drive and did a diff against the newer version that caused 2 brickings. The diff  indicated the  difference was the import and use of adafruit_ntp.

### Additional information

What is interesting is that the app with ntp worked without problems monday three days ago but today caused the boot to safe mode.

schaefer01, 2023-12-21:
Correction: the adafruit_ntp.mpy that crashed the code is dated August 19, 2023

I see a newer version, Dec 4, 2023. I'll try that and get back to you.

In either case I still need help unbricking 2 featherwings.

jepler, 2023-12-21:
https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython/circuitpython use "nuke.uf2" from here. It will erase circuitpthon and the circuitpy drive

schaefer01, 2023-12-21:
i tried flash_nuke.uf2, and it did not result in creating a CIRCUITPY drive on reboot or power cycle.

I followed flash_nuke.uf2 by Feather_RP2040.uf2, this too, did not result in creating a CIRCUITPY drive on reboot or power cycle.

schaefer01, 2023-12-21:
Updating adafruit_ntp to the latest (Dec. 4)  did not brick my rp2040. I'm closiing this problem report and opening a new problem report for my bricked rp2040s.

schaefer01, 2023-12-21:
Ooops, this worked until I cycled power. Then the same problem as before:

.]0;üêçDone | 8.2.4.\Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
soft reboot

schaefer01, 2023-12-21:
I'm down to my last featherwing.  I'm going to retry my code that  causes the entry into safe code but remove the calls to NTP.

schaefer01, 2023-12-21:
Crashed another, so it is not NTP. There has to be a different difference between the two files that causes the soft reboot. I'm making a copy of each and renaming the .py files to .txt and attaching. The first one is "safe" which doesn't reboot, the second is "remagen" because it is a version of code too far.
Also I can now recover due to help I received in problem #8743
[mqtt_ether-safe.txt](https://github.com/adafruit/circuitpython/files/13745217/mqtt_ether-safe.txt)
[mqtt_ether-remagen.txt](https://github.com/adafruit/circuitpython/files/13745218/mqtt_ether-remagen.txt)

schaefer01, 2023-12-21:
retried with v8.2.9 with no improvement, first boot runs code.py and repeated ^C followed by ^D runs code.py but cycling power leads to "CircuitPython core code crashed hard. Whoops!"

schaefer01, 2023-12-21:
I can replicate the code crashed problem. It is a combination of two issues:
 1. import and use of ntp
 2. looping on oled button press the adafruit button library .update() call
 If I comment out both kinds of references, the problem goes away. If I use either or both, the problem comes back. I'm attaching the file that works that is closes to remagen but with imports and button update 
[mqtt_ether-no-ntp.txt](https://github.com/adafruit/circuitpython/files/13745680/mqtt_ether-no-ntp.txt)
looping calls commented out.

schaefer01, 2023-12-21:
more information, I tailored mqtt to be able to poll because of a bug in mini mqtt held onto the call until a very lengthy timeout that had bad side-effects if you tried to shorten it (issue #175 and issue #189) else I would not be able to poll mqtt and also poll oled buttons.

schaefer01, 2023-12-22:
Since the buttons worked previously, and ntp and the buttons worked concurrently for a while until power cycling, and the error message points to the core, is it possible I've written a program that uses too much heap or stack?  Is this recoverable or are there hard limits to the size of the python program that can run on the featherwing?

tannewt, 2024-01-31:
Please re-test this with a 9.0 beta release. If it is still an issue, then file a new issue referencing this one.

closed 2024-01-31 by tannewt