adafruit/circuitpython #8743: bricked rp2040, tried the given steps to recover but no recovery

schaefer01, 2023-12-21:
### CircuitPython version

```python
adafruit 8.2.4 on rp2040
```

### Code/REPL

```python
See #8741 for steps to reproduce (summary: invoke an out-of-date adafruit_ntp library)
```

### Behavior

the rp2040 circuitpy goes away, whatever you put into RPI-RP2 does not return with a CIRCUITPY drive

### Description

 tried flash_nuke.uf2, and it did not result in creating a CIRCUITPY drive on reboot or power cycle.

I followed flash_nuke.uf2 by Feather_RP2040.uf2, this too, did not result in creating a CIRCUITPY drive on reboot or power cycle.

dhalbert, 2023-12-21:
What is `Feather_RP2040.uf2`? That is not the name of a circuitpython build. which board are you using?

schaefer01, 2023-12-21:
If you go to URL: https://learn.adafruit.com/adafruit-feather-rp2040-pico/downloads

then press on the button below the label: "The Feather_RP2040.uf2 file below is the code that ships on the Feather RP2040."

The file downloaded is Feather_RP2040.uf2.

By brick, I mean, you can't get a CIRCUITPY drive.

schaefer01, 2023-12-21:
The board says "feather" on it.

carson-coder, 2023-12-21:
If there was a circuitpython drive before then try to connect to the device via a serial monitor and run storage.erase_filesystem in the repl

dhalbert, 2023-12-21:
@schaefer01  OK, that is not CircuitPython, that is the factory test program. Please download CircuitPython from here:
https://circuitpython.org/board/adafruit_feather_rp2040/

schaefer01, 2023-12-21:
No need, I was using the wrong .uf2 file

closed 2023-12-21 by schaefer01