adafruit/circuitpython #8753: Feather M4 Express on 8.2.9 Wiped Itself

DJDevon3, 2023-12-24:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; Adafruit Feather M4 Express with samd51j19
Board ID:feather_m4_express
with bootloader v3.15.0.uf2

Airlift Featherwing
```

### Code/REPL

```python
Full code for the project I was working on is [here](https://github.com/DJDevon3/My_Circuit_Python_Projects/tree/main/Boards/atmel-samd/Adafruit%20Feather%20M4%20Express/Adafruit%20Airlift%20Featherwing/64x32%20Matrix%20Panel)
```

### Behavior

I did ctrl+s in Mu to save like I normally do.  It produced a memory allocation error which was a normal occurance working with the M4 Express and Featherwing to its ram limit.  This time though it just disconnected the device and CIRCUITPY device was completely new.  Nothing in /lib and code.py didn't have my 300 line script it just said "hello world".  The device was completely wiped back to a factory state without any warning.  This is the 2nd Feather M4 Express this has happened to.  The first one died and refused to be recognized by Windows as a USB device.

### Description

During normal use a Feather M4 Express set itself back to a factory state.  It completely wiped itself.

### Additional information

I think in both circumstances it happened shortly after I connected the Feather M4 Express using pin D4 to the Matrix Panel.  Since this is the 2nd time it's happened I'm reporting it.  I would be far more upset if I lost all my code again. I was actually in the process of documenting the project on my github and have a full backup of all the code, libraries, icons, and fonts.

dhalbert, 2023-12-24:
This happens when CircuitPython is unable to mount CIRCUITPY. It assumes it's corrupt or empty and reformats it. This generally happens unwantedly when the power supply sags enough for long enough that the external flash does not respond properly. Or it happens when something has corrupted CIRCUITPY so badly that it cannot be mounted, even read-only.

What host computer and OS are you using?

Are you powering the matrix through USB or separately?

DJDevon3, 2023-12-24:
Windows 10.  Powering the 5mm pitch 64x32 matrix panel separately with a [Meanwell 5V 18A PSU](https://www.amazon.com/gp/product/B018TEAPRQ).  Same one that I use to power 6 panels in my 12 panel matrix project.  I removed 1 panel and 1 PSU to make this 1 panel project.  This project started out with helping someone in the Adafruit Circuit Python Forum... and I ended up running into multiple issues with 2 different Feather M4's.  I don't use boot.py or have variables for web workflow.  Fairly standard install.

DJDevon3, 2023-12-24:
Power sag could be possible. While I was making the wiring schematic I noticed I hadn't plugged in pin D4.  So I plugged it in while the matrix panel was still powered instead of turning off the power to the matrix panel first via a rocker safety switch. 

This is actually the same process that I believe killed the first Feather M4.  I plugged in D4 while the matrix panel was on with the first feather m4 too. Because it killed that feather I unplugged D4 as I thought that pin might be a debug pin and somehow I put it into a hardware JTAG debug only mode.

After looking at the schematic I noticed D4 was one of the RGB pins so I wasn't getting correct colors without it and once again did the same thing with the 2nd feather.

Immediately after the 2nd feather wipe incident I drag/drop all the libraries, fonts, icons, code.py, and secrets.py back onto the Feather M4 from a recent backup.  It's been running fine since. Been running continuously for about 12 hours now fetching API data from OpenWeatherMap every 10 minutes and scrolling the new label data successfully.

While not a fan of the idea I can unplug D4 and attempt to replicate the issue.  Since I have a full backup I'm willing to do it if needed.  I'm just afraid that I'll kill the 2nd feather m4 by doing it again.

DJDevon3, 2023-12-24:
It just happened again this morning.  I was working as normally in Mu.  Did ctrl+s to save the file.  It gave a memory allocation error because I added a couple more labels and ran over the RAM limit.  Nothing unusual there as a ctrl+s usually works again, frees up some memory, and code.py works again.  

Except when this condition happens it presents two back to back dialogs. There is no serial log as it closes REPL completely as the device is detached.

[image]

[image]

CIRCUITPY then shows up as a generic USB device temporarily.
[image]

After a few minutes CIRCUITPY shows back up and has been completely restored to whatever the last UF2 is you installed. In my case that is currently 8.2.9

boot_out.txt
```py
Adafruit CircuitPython 8.2.9 on 2023-12-06; Adafruit Feather M4 Express with samd51j19
Board ID:feather_m4_express
```
/lib is empty and code.py is reset to "hello world".  My fonts, icons, and images folders are gone.
[image]

This is the 2nd time in 2 days this has happened.

dhalbert, 2023-12-24:
I'm not quite sure of the sequence of events after the ctrl-S. So did it save and restart? When it restarts, do the matrix panels start illuminating completely? Did you see any flash of illumination before it failed?

Could you recheck the wiring to the panels? Are you using a Matrix FeatherWing? Which one, 3036 or 4702? There are two versions. Are you connected through the barrel jack to the Meanwell power supply?

Is the pullup mentioned here, https://learn.adafruit.com/rgb-matrix-featherwing/troubleshooting#faq-3009756, 100k or 10k?

I am still wondering if this is some kind of power glitch. The external power supply may sag momentarily, though I'm not sure why the USB power would then sag if the power supply is separate. But I'm a bit taken aback that the 5v to the FeatherWing is powering the 3.3V at the same time that's being powered by the regulator on the Feather M4. 

I might try testing this setup with an oscilloscope or Saleae to see if there's a glitch on the 3.3V line when the panel starts up. For instance, if a disk write is in process, and then 3.3V sags, then that will maybe cause the external flash write to fail.

A  fix might be to insert a 1-second or so delay at the top of the program before you do anything with the matrix, to allow time for the write to complete and the power to stabilize.

DJDevon3, 2023-12-24:
The physical setup of my [Feather M4 Matrix Panel project](https://github.com/DJDevon3/My_Circuit_Python_Projects/tree/main/Boards/atmel-samd/Adafruit%20Feather%20M4%20Express/Adafruit%20Airlift%20Featherwing/64x32%20Matrix%20Panel) is also described at the bottom of the page with illustrations and pictures.  

There is no Matrix Featherwing because it conflicted with physical pinouts of the airlift... so I took it out of the equation.  Perhaps that is an issue if one of those breadboard wires requires an inline resistor. :/

DJDevon3, 2023-12-27:
Pretty good chance this is user error if I missed a required resistor.  Updated the wiring graphic to make it easier to replicate a test. I wanted the wiring in the project to be very clearly defined what I did in case it's another self-inflicted wound. Using breadboard headers on stacking header pins also doesn't provide the best connection.

DJDevon3, 2023-12-28:
I was using the bottom of the RGB Matrix Featherwing pinout in the learn guide as a reference, except then it becomes backwards and upside down. 

[image]

Seems as though I got the rows flip-flopped.  That shouldn't work at all but does actually work. After looking at the physical pinout on the matrix panel can see that there's something obviously incorrect with my wiring. The silkscreen on the physical panel doesn't correlate.  

[image]

Also where I have NC should actually be common ground. I've since learned that the NC's are the switch mode for jumper pads grounds for 64x32 vs 64x64 panels so 1 of those should also be GND. At least that's how it seems to be wired for the Matrix Portal S3 for a hub75 pinout.

I attempted to rewire it according to the silkscreen on the physical matrix panel, and it didn't work.  My guess is because I setup the address pins backwards. I'll give it another shot.  This is definitely seeming more like user error. With incorrect wiring all bets are off for something going haywire and causing a condition where the Feather M4 might reset itself.

DJDevon3, 2023-12-28:
It wiped itself again. I dismantled the entire project and put my Matrix Portal S3 on there instead which of course works splendidly because it was built exactly for that purpose.  

I think attempting to manually wire up a Feather M4 Express & Airlift to a matrix panel is just a bad idea.  It was an experiment. It doesn't work very well even when it does work. A lot of artifacts and glitching because it's using breadboard wires. It's a better idea and more fruitful use of time to use a Matrix Portal M4 or S3 instead.

Unless tracing wacky wiring and contradicting pinouts is your idea of a good time I would rather just skip this one.  Do whatever you like with it.  Follow it down a rabbit hole or chalk it up to user error.  I'm done with that maze.

justmobilize, 2023-12-28:
@DJDevon3 have you tried using a [Adafruit RGB Matrix Featherwing](https://www.adafruit.com/product/3036)? I use these without any issues...

dhalbert, 2023-12-28:
OK, I think we will close this because the actual cause is very unclear, could be power glitches.

closed 2023-12-28 by dhalbert

DJDevon3, 2023-12-28:
@justmobilize To my knowledge, you can't use an RGB Matrix Featherwing with both a Feather M4 Express and an Airlift featherwing. That's why I had to manually wire it in the first place. The RGB Matrix Featherwing requires some of the same pins used by the Airlift.  I could cut traces but I really don't feel like doing that if I'll need it again in the future.  Hence wiring around it manually which turned out to be a bad idea.