adafruit/circuitpython #8756: Stubs for board specific pins.

evilhamsterman, 2023-12-25:
The circtuitpython-stubs package is great, and along with the improved typing in the bundles you can get pretty good hinting in tools like VSCode. 

The one missing piece is the pins for specific boards. The circuitpython-stubs package for `board` only has hints for the `I2C`, `SPI`, and `UART` functions and the `board_id`. So you don't get autocomplete and possibly warnings for the board specific features.

@joedevivo who maintains a VSCode extension for Circuit Python does some good scripting to build some stubs https://github.com/joedevivo/vscode-circuitpython/blob/master/scripts/build_stubs.py that he uses in his extension. But I'm sure a native way could do it better perhaps individual packages for circuitpython-stubs-pipicow for example.

adasauce, 2024-01-16:
in case anyone is here from vim/neovim land trying to find board specific stubs.

you can also find your board in this project under ports. i.e. /ports/raspberrypi/boards/adafruit_kb2040

and find the pins definition to just add a few more type stubs to the `typings/board-stubs/__init__.pyi ` where needed.

```python
import microcontroller
D6: microcontroller.Pin
D9: microcontroller.Pin
def STEMMA_I2C(): -> busio.I2C:
    ...
```

kind of hacky but i've just been spending 3 minutes setting it up for the project/board I'm working on and it hasn't needed any maintenance and gets the job done.

evilhamsterman, 2024-02-06:
@adasauce that's basically what @joedevivo does in his extension, he has some scripts that parse the C code to combine it with the board.pyi stub. So it's certainly doable, just would be nice if it were provided from upstream.

FoamyGuy, 2024-02-06:
This functionality is in #8882

evilhamsterman, 2024-02-16:
@FoamyGuy I assume this'll have to wait till the 9.0.0 release to be available. Also I'd recommend that you can select the board more easily without having to change code before uploading by using a try in your import

```python
try:
    from board_definitions import adafruit_matrixportal_s3 as board
except ImportError:
   import board
```

justmobilize, 2024-02-16:
It should roll out with `9.0.0.beta.1` if you are willing to use a pre-release.

Also, you can use the board setter or the import method you have above. Either will work and just depends on your preferences.

evilhamsterman, 2024-02-26:
I just gave it a shot with beta 2 and it works perfect 👍 

I just gave the recommendation because I thought it would be something that would be good to document.

Thanks!

closed 2024-02-26 by evilhamsterman