adafruit/circuitpython #8757: try_lock() timeout

bludin, 2023-12-27:
I find the widely used 

`while not i2c.try_lock(): pass`

construct problematic (or even catastrophic in boot.py) because it will just get the code into an endless loop if try_lock() is unsuccessful.
 It would be nice to have an optional timeout argument so that an exception will be raised if the lock cannot be acquired within a certain timeframe. Currently, I would even opt for the timeout to have a default value of maybe 1 or 2 seconds.

deshipu, 2023-12-27:
Yes, you can easily do that. What is the issue?

```python
import time

timeout = time.monotonic() + 3.0
while not i2c.try_lock():
    if time.monotonic() > timeout:
        raise RuntimeException("i2c lock timeout")
```

bludin, 2023-12-27:
having it integrated would be a lot more effective/elegant both for the coder and the reader, i.e. more pythonic. And the proposed default timeout would work retrogradely.

deshipu, 2023-12-27:
Not sure what exactly you mean by efficient? Elegance is in the eye of the beholder.

What are you going to do when it times out anyways? The rest of the program assumes that you have acquired the lock.

bludin, 2023-12-27:
The rest of the program won't run if an unhandled exception is raised. But it gives the developer a clear indication of what the problem was (or the program a way to handle it).  One instance where this is particularly helpful is in boot.py. If try_lock goes into an endless loop, the cp module will simply not boot and remain completely inaccessible (i.e. with no USB drive mounted and no REPL connected).  The only way to get out of that is to manually boot into safe mode by pressing the reset button twice with the correct interval. And then you are left with no indication of what went wrong because no bootout.txt has been created.

deshipu, 2023-12-27:
You don't need `my_busio`, just use `busdevice`, it handles the locking correctly for you.

RetiredWizard, 2023-12-27:
I agree that the while loop used in the examples may not be the best way to use the try_lock function and for new users may lead to issues. 

Much of the core is still over my head so it's dangerous for me to make conclusions after looking at the C code, but when I look at the common_hal_busio_i2c_try_lock function (rp2040), it doesn't look to me like there is any advantage to calling the try_lock function a second time if the first call returns a False. Is it possible the while loop was used in the examples because in the limited context of the example there was no difference between infinite looping and simply exiting the code? 

i.e. in the context of the examples would the following code serve the same instructional function?
```py
if i2c.try_lock():
    # execute sample code
else:
    print("i2c bus not available")
```

deshipu, 2023-12-27:
> Much of the core is still over my head so it's dangerous for me to make conclusions after looking at the C code, but when I look at the common_hal_busio_i2c_try_lock function (rp2040), it doesn't [...]

The loop makes sense when there is a display on the same bus, that is update in the background, and may lock the bus temporarily while that happens. There was also some discussion about adding support for gpio expanders to libraries like keypad, which would result in a similar situation - the bus becoming locked temporarily in the background, independent from your own program.

bludin, 2023-12-27:
> You don't need `my_busio`, just use `busdevice`, it handles the locking correctly for you.

I dare to disagree. `try_lock` is a method of the bus object, which you have to specify to instantiate a bus device. The bus device will use the `try_lock` method of the bus object - and end up in an indefinite loop if it fails to acquire the lock

deshipu, 2023-12-27:
Maybe you should explain to us a little bit more about how you ended up with a permanently locked bus?

bludin, 2023-12-28:
There are situations that require manual locking. The scan() method of an I2C object, for instance. I accidentally deleted the unlock() line and then used then tried to access the I2C encoder breakout.

bradanlane, 2023-12-28:
While I do like the idea of a timeout on `try_lock()` and I already do this in most of my code, the cause-and-effect is one of _legacy examples_.

As @bludin notes, there are so many examples that show the `while not i2c.try_lock(): pass` code snippet that users see it as a best practice. If the `try_lock()` throws an exception and the examples showed a `try: except` code block, then much of the angst and head scratching would go away.

I recognize this is a case of _"water under the bridge"_ but I hope all of our future example code will be more representative of best practices ... especially give Copilot and other AI are taking our code as gospel and recommending it to others.

dhalbert, 2023-12-28:
`try_lock()` is exactly that: it returns a boolean, and throws no exception if there is a failure. Adding a timeout that might throw an exception changes its semantics. A separate `lock()` operation that has a default timeout and does throw an exception if the locking fails might be better.

bradanlane, 2023-12-28:
yes. I think the missing piece is for all of us to make better examples since they are used as "best practice".

dhalbert, 2023-12-28:
Following on @RetiredWizard's examples, the simplest might be:
```py
if not i2c.try_lock():
    raise RuntimeError()
```

deshipu, 2023-12-28:
@bludin how many users do you think are going to put an i2c scan in their boot.py and make the exact same mistake as you, and do you think preventing this is worth removing some built-in library from the smallest boards to make room for the extra code this will require? Do you already have an opinion about which built-in module would be best to remove?

deshipu, 2023-12-28:
@dhalbert that code will fail randomly when you have a display on that i2c bus.

bludin, 2023-12-28:
> @bludin how many users do you think are going to put an i2c scan in their boot.py and make the exact same mistake as you, and do you think preventing this is worth removing some built-in library [...]

@deshipu if that's indeed the only conceivable scenario where the real benefit (as opposed to the esthetic one) is significant, we might want to wait until the day when there is a web interface for custom builds that lets the user choose the desired modules :)

deshipu, 2023-12-28:
@bludin I'm afraid that the day when we drop the support for the smallest boards is closer.

anecdata, 2023-12-28:
> web interface for custom builds

#7594

tannewt, 2024-01-17:
I don't think we need to change anything internally. The wrapper libraries already make it easier to use.

If anyone has something concrete to do, please follow up and I'll reopen this.

closed 2024-01-17 by tannewt