adafruit/circuitpython #8758: super() does not work in an overloaded __setitem__ method

miguelgrinberg, 2023-12-27:
### CircuitPython version

```python
MicroPython 9.0.0-alpha.6-8-g9ac250aa3b on 2023-12-27; linux [GCC 11.4.0] version

(this is my custom build of the UNIX port of CircuitPython, the executable is also named `micropython`, it looks like you haven't renamed this port)
```

### Code/REPL

```python
>>> class D(dict):
...     def __setitem__(self, k, v):
...         super().__setitem__(k, v)
...
...
>>> d = D()
>>> d['foo'] = 'bar'
```

### Behavior

```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 3, in __setitem__
  File "<stdin>", line 3, in __setitem__
  File "<stdin>", line 3, in __setitem__
  File "<stdin>", line 3, in __setitem__
  File "<stdin>", line 3, in __setitem__
  File "<stdin>", line 3, in __setitem__
RuntimeError: maximum recursion depth exceeded
>>>
```

### Description

It appears as if `super()` does not work inside a `__setitem__()` method. The same code works fine under MicroPython, built in the same way:

```python
MicroPython v1.22.0 on 2023-12-27; linux [GCC 11.4.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> class D(dict):
...     def __setitem__(self, k, v):
...         super().__setitem__(k, v)
...
>>> d = D()
>>> d['foo'] = 'bar'
>>> d
{'foo': 'bar'}
>>>
```

Also works on CPython:

```python
Python 3.12.0 (main, Oct  5 2023, 10:46:39) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> class D(dict):
...     def __setitem__(self, k, v):
...         super().__setitem__(k, v)
...
>>> d = D()
>>> d['foo'] = 'bar'
>>> d
{'foo': 'bar'}
>>>
```

dhalbert, 2023-12-27:
Could you test this for a few previous versions of MicroPython? We have only merged from MicroPython v1.21.0, and perhaps this does not work in older versions of MicroPython.

miguelgrinberg, 2023-12-27:
For background, the reason I'm doing this is that I'm trying to get my [Microdot](https://github.com/miguelgrinberg/microdot) web framework to run under CircuitPython. My test suite runs on the UNIX port, and while I have been testing MicroPython releases for a long time this problem is currently preventing me from running my tests on CircuitPython.

Just in case and to be 100% sure, I tested this on a few older releases of MicroPython.

```python
MicroPython v1.19.1 on 2023-12-27; linux [GCC 11.4.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> class D(dict):
...     def __setitem__(self, k, v):
...         super().__setitem__(k, v)
...
>>> d = D()
>>> d['foo'] = 'bar'
>>> d
{'foo': 'bar'}
>>>
```

```python
MicroPython v1.20.0 on 2023-12-27; linux [GCC 11.4.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> class D(dict):
...     def __setitem__(self, k, v):
...         super().__setitem__(k, v)
...
>>> d = D()
>>> d['foo'] = 'bar'
>>> d
{'foo': 'bar'}
>>>
```

```python
MicroPython v1.21.0 on 2023-12-27; linux [GCC 11.4.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> class D(dict):
...     def __setitem__(self, k, v):
...         super().__setitem__(k, v)
...
>>> d = D()
>>> d['foo'] = 'bar'
>>> d
{'foo': 'bar'}
>>>
```

dhalbert, 2023-12-27:
Does it work on a non-Unix (microcontroller) version of CircuitPython? How about on a microcontroller version of MicroPython?

justmobilize, 2023-12-27:
Tried on `8.2.9` on a `ESP32-S3 TFT`:
[image]

justmobilize, 2023-12-27:
Also playing with it, 
```
class x:
    def y(self):
        print(1)

class xx(x):
    def y(self):
        super().y()

>>> xx().y()
1
```
Works as expected. 

Doing this:
```
class D(dict):
    pass

d = D()
d['foo'] = 'bar'

>>> d
{}
```
Shows it's specific to `dict` (or maybe other things)

miguelgrinberg, 2023-12-27:
To answer the remaining question, this works just fine on regular MicroPython builds for devices. I have personally tested it on various ESP8266, Pico and Pico W builds over the last 3 years or so.

dhalbert, 2024-01-22:
I tested back to MicroPython 1.11 and it works. In CircuitPython, this stopped working properly a long time ago:
```py
class D(dict):
    def __setitem__(self, k, v):
        super().__setitem__(k, v)

d = D()
d[2] = 3
print(d[2])
```
- starting with 5.00-beta.5: `RuntimeError: maximum recursion depth exceeded`
- starting with 5.00-beta.3: `Key Error: 2`
- 5.0.0-beta.1 and before: [no error]

The `Key Error` version gives surprising results:
```py
Adafruit CircuitPython 5.0.0-beta.3 on 2020-01-08; Adafruit Metro M0 Express with samd21g18
>>> from sup import D
>>> d = D()
>>> d[2] = 3
>>> d
{2: 3}
>>> d[2]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyError: 2
```

I'll look at the PR's in those ranges. Not sure if I can build such old versions to do a bisect, but I'll try if necessary.

tannewt, 2024-01-23:
It may be easiest to debug this using the UNIX CP build and writing a test for it.

We do slightly different native subclassing stuff and lookup so a diff in the py directory might be enough too. (IIRC this is for native subclassing and better property support.)

dhalbert, 2024-01-23:
There was a lot of work done for 5.0.0-beta.3 for `_pixelbuf` that involved handling subclasses of native classes. There were a few more changes in 5.0.0-beta.5. There may be such changes missing from `objdict.c`, or there is some fundamental issue in how it was done. For instance, I will try the same thing for a subclass of `list`, which does have the changes.

tannewt, 2024-01-23:
@dhalbert and I debugged this today. It is caused by CircuitPython passing subclass instances into native code. The setitem impl would then read the type and lookup what to call. The type from the instance is still the subclass which causes the recursion. PR incoming.

PR #8828 (2024-01-24): Fix subclassing dict
The get, set and del item methods didn't correctly lookup the value from the parent native instance because the functions took the type from the instance.

Fixes #8758
commit 283aac23 (2024-01-24): Fix subclassing dict

The get, set and del item methods didn't correctly lookup the value
from the parent native instance because the functions took the type
from the instance.

Fixes #8758
closed 2024-01-24 by dhalbert