adafruit/circuitpython #8759: Broadcom Port: CIRCUITPY Drive not available

rsbohn, 2023-12-27:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; Raspberry Pi Zero W with bcm2835
```

### Code/REPL

```python
None
```

### Behavior

Prints the version, Board ID, and UID on the HDMI screen, then stops.
The status line is visible:
 
    :snake: Done | 8.2.9

### Description

On the Raspberry Pi Zero W it doesn't seem to boot fully. The CIRCUITPY drive is not mounted. Windows complains of an unknown USB device.

### Additional information

Last known good version: Adafruit CircuitPython 8.0.5 on 2023-03-31; Raspberry Pi Zero W with bcm2835
Tested on Windows 11.

dhalbert, 2023-12-27:
Does #8044 seem like your problem? We thought we fixed it in 8.1.0.

rsbohn, 2023-12-27:
Deleted `code.py` and it boots, the CIRCUITPY drive is available. If I connect on the serial terminal I can get the '>>>' prompt, but if type anything more it crashes when I press ENTER.

skeyvin, 2023-12-28:
same problem with pi 4 b but deleting code.py doesnt fix it

RetiredWizard, 2023-12-28:
With 9.0.0 Alpha 6 on the Zero W, plugged into the inside USB connector  many "read sector" errors are reported, presumably while attempting to mount the CIRCUITPY drive.
```
...
read sectors result 8 when reading block 7772138 for 1
read sectors result 8 when reading block 7772140 for 1
read sectors result 8 when reading block 7772142 for 1
read sectors result 8 when reading block 524288 for 1
read sectors result 8 when reading block 524289 for 1
read sectors result 8 when reading block 524291 for 1
read sectors result 8 when reading block 524293 for 1
...
```
But the REPL appeared to behave normally after all the messages were displayed. Although an os.listdir() resulted in another read sector error.

If I connected via the UART rx/tx using a serial terminal and powered the PI Zero using the outside USB connector the read sector errors were not displayed and the os.listdir() worked fine.

skeyvin, 2023-12-29:
forced hdmi hotplug to get a repl
now i cant access the data due to
`SD card init failed 2                                                                                                                                                                                              
SD card init failed 2                                                                                                                                                                                              
SD card init failed 3 
0;🐍Done | 9.0.0-alpha.6                                                                                                                                                                                           
Auto-reload is off.                                                                                                                                                                                                
Running in safe mode! Not running saved code.                                                                                                                                                                      
`
on the serial