adafruit/circuitpython #8761: M5Stack M5Paper when accessing board.DISPLAY

richardanaya, 2023-12-28:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.6 on 2023-12-12; M5Stack M5Paper with ESP32
```

### Code/REPL

```python
import board
board.DISPLAY.refresh()
```

### Behavior

```
Serial console setup

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

### Description

Seems like some kind of error when doing any access to DISPLAY

### Additional information

None

carson-coder, 2024-01-01:
It looks like DISPLAY doesn't exist (https://github.com/adafruit/circuitpython/blob/main/ports/espressif/boards/m5stack_m5paper/board.c) so maybe there is a definition for DISPLAY but there isn't a display instants causing circuitpython to segfault causing safemode

carson-coder, 2024-01-01:
So we could remove the DISPLAY definition (https://github.com/adafruit/circuitpython/blob/main/ports/espressif/boards/m5stack_m5paper/pins.c#L76) or setup the display.

RetiredWizard, 2024-01-01:
There's some background info in #8293. Removing the DISPLAY definition even if it's only while the setup is being worked out sound right to me...

carson-coder, 2024-01-03:
I can make a pull request removing (or commenting out) the display definition until someone writes code to initialize the display

PR #8841 (2024-01-25): Comment out m5paper board.DISPLAY
It isn't init yet. Fixes #8761
commit 2310bafb (2024-01-25): Comment out m5paper board.DISPLAY

It isn't init yet. Fixes #8761
closed 2024-01-25 by dhalbert