adafruit/circuitpython #8762: no response for mdns circuitpython.local

davs2rt, 2023-12-29:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; Seeed Studio XIAO ESP32C3 with ESP32-C3FN4
Board ID:seeed_xiao_esp32c3
UID:43588170B08D
```

### Code/REPL

```python
http://circuitpython.local does not respond MDNS
```

### Behavior

I have 4 Seeed Studios Xiao ESP32C3 processors on the network, running 8.2.8 and 8.2.9. When I attempt to get to get to http://circuitpython.local or /code/ I get a timeout.  Wireshark shows no response to MDNS request packets.  The attached file is a pcapng, renamed to .txt.  [mdns_timeout.txt](https://github.com/adafruit/circuitpython/files/13790041/mdns_timeout.txt)

### Description

Possible clues:
- This originally worked on all 4 devices using 8.2.8, got harder to make a connection, then eventually failed everywhere. 
- It fails from Linux Mint, Windows 10, Firefox, Chrome. 
- I upgraded from 8.2.8 to 8.2.9 to see if it mattered: no.  Device failed to respond to mdns request on with a clean install.  
- Device works for outbound connections to access ntp servers, mqtt server/broker using asyncio. 
- Device only rarely responds to pings. 
- As the ESP32C3 doesn't support a file system over USB, my only other workflow is using http://10.1.99.2/fs/ to push files to the device, which works sometimes.
- USB serial is pretty reliable
- Running asyncio.  I considered that asyncio was interfering with the web response, but it also failed on a fresh install, no asyncio.

### Additional information

I'd be happy to develop additional debugging data, but I'm lost on what to look at next.

dhalbert, 2024-01-17:
This sounds at first hearing like obsolete DNs entries being cached on the host computers or the routers, but given the non-response to mDNS packets, Iam not so sure.

If you wipe the flash on the C3 and the reinstall CircuitPython and the test setup, does it work again?

davs2rt, 2024-01-19:
Solved: user error.
I had set the settings.toml keywords incorrectly.  e.g.
CIRCUITPY**THON**_WIFI_SSID = "redacted"
CIRCUITPY**THON**_WIFI_PASSWORD = "redacted"
which kills the web workflow at 
circuitpython/supervisor/shared/web_workflow/web_workflow.c    line 269

closed 2024-01-19 by davs2rt