adafruit/circuitpython #8763: ESP32-C3 auto boot into safemode?

llllllllqq, 2023-12-29:
### CircuitPython version

```python
adafruit-circuitpython-luatos_core_esp32c3-en_US-8.2.9
and also 9.0.0-alpha.6
```

### Code/REPL

```python
none
```

### Behavior

The led always blink 3 times on every reset (both on push rst button or re-plug usb cable), maybe in a safemode but repl doesn't say why.

### Description

It's a fresh install by the web installer. WIFI was configured by web installer too. I tried to use a external USB to TTL adapter to connect the esp32-c3 for more logs, but the serial gave me nothing else than repl. 
I tried other firmware (like tasmota) and it works ok, so I believe the hw is not broken. I'm new to circuitpython, if there are more I can do, please comment, thanks a lot.

carson-coder, 2024-01-03:
if safemode is entered then code.py doesn't run and it tells you when its in safe mode so its not in safemode

llllllllqq, 2024-01-19:
Today, I reflashed 9.0.0-alpha.6, it worked well. But when I switched to 8.2.9, bug occurred.
Idk why the 9.0.0-alpha.6 version works today, the hardware and flash methods are same as before.
If I find this bug again, I'll post here and reopen this issue. Thanks for your hard work.

closed 2024-01-19 by llllllllqq