adafruit/circuitpython #8770: espulp module broken in CircuitPython 9

Sola85, 2023-12-29:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.6 on 2023-12-29; ESP32-S3-DevKitM-1-N8
```

### Code/REPL

```python
>>> import espulp
>>> ulp = espulp.ULP()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError:
```

### Behavior

In CircuitPython 9 it is no longer possible to instantiate a ULP object. In CircuitPython 8 this still worked.

### Description

The reason for this seems to be the new preprocessor directives that were introduced between CircuitPython 8 and 9:
```c
#ifdef CONFIG_ULP_COPROC_TYPE_FSM
        case FSM:
            break;
#endif
```

In particular `CONFIG_ULP_COPROC_TYPE_FSM` does not seem to get defined anywhere. 

I had started to work on improving the espulp module for CircuitPython 8 and wanted to port my changes to CircuitPython 9, but i hit this issue that i was not yet able to resolve.

Sola85, 2023-12-30:
I have since found out that the location where `CONFIG_ULP_COPROC_TYPE_FSM` should be defined is in `ports/espressif/esp-idf-config/sdkconfig-*.defaults`.

The issue seems to be that `idf.py menuconfig` only allows either `CONFIG_ULP_COPROC_TYPE_FSM` or `CONFIG_ULP_COPROC_TYPE_RISCV` to be defined but not both at the same time. This is despite the ESP32S2 and S3 having both the FSM and RISCV ULPs...

It looks like this issue has to be fixed upstream in the espressif toolchain. In the meantime I would suggest to enable/disable FSM/RISCV support based on the `IDF_TARGET` and not based on `CONFIG_ULP_COPROC_TYPE_FSM`/`CONFIG_ULP_COPROC_TYPE_RISCV`

Sola85, 2023-12-30:
According to the [Reference Manual](https://cdn-learn.adafruit.com/assets/assets/000/110/710/original/esp32-s3_technical_reference_manual_en.pdf?1649790877) of the ESP32S3 (Page 29), the ULPs cannot be used simultaneously. This means that Espressif will likely not update `idf.py menuconfig` to allow `CONFIG_ULP_COPROC_TYPE_FSM` and `CONFIG_ULP_COPROC_TYPE_RISCV` to be enabled simultaneously.

But compiling Circuitpython with both ULPs worked for Circuitpython 8, so I suggest to revert the related changes from commit c38972b. 
@microdev1 @tannewt What do you think about this?

tannewt, 2024-01-16:
I suspect you'll need to copy over the IDF functions we use so they are both available at compile time until espressif fixes it. I'd suggest filing an issue with them about making the API work dynamically like we use it.

Sola85, 2024-02-28:
It doesn't look like we will get a response from espressif soon. In the meantime I would propose to enable the FSM ULP for all boards. Then the API is consistent across all ESPs (unlike now, where RISCV is enabled on some and FSM on others).

@tannewt What do you think?

tannewt, 2024-02-29:
> It doesn't look like we will get a response from espressif soon. In the meantime I would propose to enable the FSM ULP for all boards. Then the API is consistent across all ESPs (unlike now, where RISCV is enabled on some and FSM on others).
> 
> @tannewt What do you think?

If you want to get it working with FSM only that's fine with me. I'd be happy to review it. It isn't a priority for me at the moment.

dhalbert, 2024-06-03:
Fixed by #9277.

closed 2024-06-03 by dhalbert