adafruit/circuitpython #8771: RP2040: Deep sleep until alarm without reset

rgov, 2023-12-29:
Since the RP2040 preserves RAM when in the dormant (deep sleep) state, is it not possible to resume the program after a PinAlarm, without resetting the board? This would make it significantly easier to preserve state while dropping into lower power modes.

[@ghubcoder's blog post](https://ghubcoder.github.io/posts/awaking-the-pico/) describes how to recover from the dormant state by resetting clocks. This was implemented for MicroPython in [modpicosleep.c](https://github.com/ghubcoder/micropython-pico-deepsleep/blob/2b73bb0385644b8d3226da80c7f8b03566ed1cec/ports/rp2/modpicosleep.c) ([accompanying blog post](https://ghubcoder.github.io/posts/deep-sleeping-the-pico-micropython/)).

dhalbert, 2024-01-17:
I think we could do this for "light sleep", and make light sleep be a deeper sleep.

bablokb, 2025-02-04:
@rgov: you could test the artifact of PR #10023. This will use dormant mode without reset for lightsleep like you suggested. According to my measurements, we are now in the same range as the blog-post claims for MicroPython.