adafruit/circuitpython #8774: wakeup feature for usb_hid package

meesokim, 2023-12-30:
usb_hid packages support keyboard, mouse, consumer_contol features based on tinyusb. The examples of tinyusb can support wakeup in host sleep mode but Circuitpython usb_hid packages has no code to support it. I added some code from tinyusb example to Circuitpython usb implementation code. I tested this modification on esp32s2 board and works fine as I expected.

Here are the diff file.

 diff --git a/shared-module/usb_hid/Device.c b/shared-module/usb_hid/Device.c
index b28de9c1b1..7d0e6eff3a 100644
--- a/shared-module/usb_hid/Device.c
+++ b/shared-module/usb_hid/Device.c
@@ -229,12 +229,16 @@ void common_hal_usb_hid_device_send_report(usb_hid_device_obj_t *self, uint8_t *
         RUN_BACKGROUND_TASKS;
     }

-    if (!tud_hid_ready()) {
-        mp_raise_msg(&mp_type_OSError, MP_ERROR_TEXT("USB busy"));
-    }
+    if (!tud_suspended()) {
+        if (!tud_hid_ready()) {
+            mp_raise_msg(&mp_type_OSError, MP_ERROR_TEXT("USB busy"));
+        }

-    if (!tud_hid_report(report_id, report, len)) {
-        mp_raise_msg(&mp_type_OSError, MP_ERROR_TEXT("USB error"));
+        if (!tud_hid_report(report_id, report, len)) {
+            mp_raise_msg(&mp_type_OSError, MP_ERROR_TEXT("USB error"));
+        }
+    } else {
+        tud_remote_wakeup();
     }
 }

diff --git a/supervisor/shared/usb/usb_desc.c b/supervisor/shared/usb/usb_desc.c
index 811f61724f..9f932ddbfe 100644
--- a/supervisor/shared/usb/usb_desc.c
+++ b/supervisor/shared/usb/usb_desc.c
@@ -105,7 +105,7 @@ static const uint8_t configuration_descriptor_template[] = {
 #define CONFIG_NUM_INTERFACES_INDEX (4)
     0x01,        // 5 bConfigurationValue
     0x00,        // 6 iConfiguration (String Index)
-    0x80,        // 7 bmAttributes
+    0x80 | TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP,        // 7 bmAttributes
     0x32,        // 8 bMaxPower 100mA
 };

dhalbert, 2023-12-30:
Thanks! There is discussion of this in https://github.com/adafruit/circuitpython/issues/5380#issuecomment-925044500 and the whole issue. I think you did all that was noted there?

dhalbert, 2023-12-30:
Would you be interested in submitting this as a PR?

meesokim, 2023-12-31:
Yes of course.

dhalbert, 2024-01-17:
Still interested in a PR if you have time.

meesokim, 2024-01-24:
> Still interested in a PR if you have time.

Could you I submit a PR with this modification?

dhalbert, 2024-01-24:
It turns out this is a Linux issue. The MacroPad worked fine to wake up macOS and Windows. See https://github.com/adafruit/circuitpython/issues/5380#issuecomment-925044500 for how to determine which USB devices are enabled for wakeup. See also https://askubuntu.com/questions/848698/wake-up-from-suspend-using-wireless-usb-keyboard-or-mouse-for-any-linux-distro/874701 for details on how to change these settings. I would have thought there was an easier way, and maybe there is for some Linux distributions.

For example on my Linux box, only my primary keyboard is enabled for wakeup:
```py
$  grep . /sys/bus/usb/devices/*/product
/sys/bus/usb/devices/1-12/product:HD Pro Webcam C920
/sys/bus/usb/devices/1-1/product:Freestyle Edge RGB Keyboard
/sys/bus/usb/devices/1-2.5/product:USB Billboard Device   
/sys/bus/usb/devices/1-2/product:USB2.0 Hub             
/sys/bus/usb/devices/1-3/product:Microsoft Basic Optical Mouse 
/sys/bus/usb/devices/1-6/product:Macropad RP2040
/sys/bus/usb/devices/2-2.1/product:USB 10/100/1000 LAN
/sys/bus/usb/devices/2-2/product:USB3.0 Hub             
/sys/bus/usb/devices/usb1/product:xHCI Host Controller
/sys/bus/usb/devices/usb2/product:xHCI Host Controller

$  grep . /sys/bus/usb/devices/*/power/wakeup
/sys/bus/usb/devices/1-10/power/wakeup:disabled
/sys/bus/usb/devices/1-1/power/wakeup:enabled
/sys/bus/usb/devices/1-2.5/power/wakeup:disabled
/sys/bus/usb/devices/1-2/power/wakeup:disabled
/sys/bus/usb/devices/1-3/power/wakeup:disabled
/sys/bus/usb/devices/1-6/power/wakeup:disabled
/sys/bus/usb/devices/1-8/power/wakeup:disabled
/sys/bus/usb/devices/2-2.1/power/wakeup:disabled
/sys/bus/usb/devices/2-2/power/wakeup:disabled
/sys/bus/usb/devices/usb1/power/wakeup:disabled
/sys/bus/usb/devices/usb2/power/wakeup:disabled
```

PR #8830 (2024-01-25): shared-module/usb_hid: allow HID to wake sleeping host computer
- Fixes #8774.
- Fixes #5380.

Thanks very much to @meesokim for code to enable `usb_hid` to wake a sleeping host computer, and to @hathach for implementing this ability in TinyUSB in the past.

Tested successfully with MacroPad RP2040, Feather nRF52840, and Circuit Playground Express on Windows and macOS.

On Linux not all HID devices are automatically enabled to do wakeup. See @hathach's [comment](https://github.com/adafruit/circuitpython/issues/5380#issuecomment-925044500). See also https://askubuntu.com/questions/848698/wake-u [... cut]
closed 2024-01-25 by dhalbert