adafruit/circuitpython #8775: OSError Whenever running code

carson-coder, 2023-12-30:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9-18-gee8b6f4e9b on 2023-12-30; Adafruit MatrixPortal S3 with ESP32S3
```

### Code/REPL

```python
import board
import displayio
import framebufferio
import rgbmatrix

DISPLAY_WIDTH = 64
DISPLAY_HEIGHT = 32

matrix = rgbmatrix.RGBMatrix(
    width=DISPLAY_WIDTH, height=DISPLAY_HEIGHT, bit_depth=3,
    rgb_pins=[
        board.MTX_R1,
        board.MTX_G1,
        board.MTX_B1,
        board.MTX_R2,
        board.MTX_G2,
        board.MTX_B2
    ],
    addr_pins=[
        board.MTX_ADDRA,
        board.MTX_ADDRB,
        board.MTX_ADDRC,
        board.MTX_ADDRD
    ],
    clock_pin=board.MTX_CLK,
    latch_pin=board.MTX_LAT,
    output_enable_pin=board.MTX_OE,
    tile=1, serpentine=False,
    doublebuffer=False
)
display = framebufferio.FramebufferDisplay(matrix)

bitmap = displayio.OnDiskBitmap("/bitmap.bmp")
tile_grid = displayio.TileGrid(bitmap, pixel_shader=bitmap.pixel_shader)
group = displayio.Group()
group.append(tile_grid)
display.root_group = group

# Loop forever so you can enjoy your image
while True:
    pass
```

### Behavior

```
code.py output:
OSError: [Errno 5] Input/output error

```
This has been happening some times whenever I would run my code but I could just press ctrl+D to fix it but now circuitpython hardcrashed and now it is always throwing the oserror.

### Description

I do have one display connected and this started when I enabled serpentine. I copied this code from my matrixportal m4 that had 4 displays connected but just changed the tile amount to 1.

I am running my own compiled circuitpython from pull request #8752. I can download 9.x.x or 8.2.x if needed

jepler, 2023-12-30:
This may be a duplicate of #8449

carson-coder, 2023-12-30:
My code never runs it just always gives a os error

dhalbert, 2023-12-30:
Are you doing this on macOS Sonoma? I thought you were on Linux.

Try this with 8.2.9 or the 9.0.0-alpha.6 build. Before you do anything, save what you need on CIRCUITPY, and then do
```py
import storage
storage.erase_filesystem()
```
to make sure CIRCUITPY is not corrupted in some way. Then copy the files back and try again.

carson-coder, 2023-12-30:
I am on linux

carson-coder, 2023-12-30:
I erased the filesystem and now I am getting a too many display busses error
`
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 27, in <module>
RuntimeError: Too many display busses

Code done running.`

carson-coder, 2023-12-30:
I swiched my build to 8.2.x but still compiled on my computer. Im going to try downloading a compiled 8.2.x version

carson-coder, 2023-12-30:
With a build downloaded from https://circuitpython.org/board/adafruit_matrixportal_s3/ The code runs once then if I try ctrl-c ctrl-d then I get the too many display busses error

deshipu, 2023-12-30:
You forgot to call `displayio.release_displays()` before creating a new display.

carson-coder, 2023-12-30:
I have this new code that does the same thing ```import board
import displayio
import framebufferio
import rgbmatrix

displayio.release_displays()
matrix = rgbmatrix.RGBMatrix(
    width=64, bit_depth=4,
    rgb_pins=[
        board.MTX_R1,
        board.MTX_G1,
        board.MTX_B1,
        board.MTX_R2,
        board.MTX_G2,
        board.MTX_B2
    ],
    addr_pins=[
        board.MTX_ADDRA,
        board.MTX_ADDRB,
        board.MTX_ADDRC,
        board.MTX_ADDRD
    ],
    clock_pin=board.MTX_CLK,
    latch_pin=board.MTX_LAT,
    output_enable_pin=board.MTX_OE
)
display = framebufferio.FramebufferDisplay(matrix)

bitmap = displayio.OnDiskBitmap("/bitmap.bmp")
tile_grid = displayio.TileGrid(bitmap, pixel_shader=bitmap.pixel_shader)
group = displayio.Group()
group.append(tile_grid)
display.root_group = group

# Loop forever so you can enjoy your image
while True:
    pass

carson-coder, 2023-12-30:
Now I am just getting os errors but my code won;t run

carson-coder, 2023-12-30:
Doing another erase gets me the error MTX_R1 in use. Im going to close this error for now and move to the discord

closed 2023-12-30 by carson-coder