adafruit/circuitpython #8776: Enable Bluetooth LE support on ESP-WROOM-32 ( DOIT ESP32 Development Board )

ytyng, 2023-12-31:
[DOIT ESP32 Development Board](https://circuitpython.org/board/doit_esp32_devkit_v1/) implements ESP-WROOM-32.

However, the BLEIO module is not enabled.

## REPL

adafruit-circuitpython-doit_esp32_devkit_v1-en_US-8.2.9.bin

```
Adafruit CircuitPython 8.2.9 on 2023-12-06; ESP32 Devkit V1 with ESP32
>>> import _bleio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named '_bleio'
>>> 
```

## Web

dhalbert, 2023-12-31:
BLE has not been available on plain ESP32 ever in CircuitPython, for various reasons. See for instance <https://github.com/adafruit/circuitpython/pull/7220>.

tannewt, 2024-05-01:
Duplicate of #7170. Thanks @vladak!