adafruit/circuitpython #8778: hard fault on Matrix Portal S3

TonyLHansen, 2024-01-02:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; Adafruit MatrixPortal S3 with ESP32S3
Board ID:adafruit_matrixportal_s3
UID:84722ED31008
```

### Code/REPL

```python
import board
from digitalio import DigitalInOut, Direction
# from adafruit_debouncer import Debouncer

class InvertedDigitalInOut(DigitalInOut):
    """A class that provides an inverted version of DigitalInOut pins."""

    def __init__(self, pin):
        super().__init__(pin)

    def rose(self):
        return super().fell()

    def fell(self):
        return super().rose()

    @property
    def value(self) -> bool:
        """The value of the pin, either ``True`` for high/pulled-up or
        ``False`` for low.
        """
        return not super().value

btn = InvertedDigitalInOut(board.A1)
btn.direction = Direction.INPUT
btn.pull = Pull.UP

print("okay")
```

### Behavior

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

### Description

The above code snippet is the smallest version I could get that replicated the error.

I was trying to create a class that would act like Debouncer(DigitalInOut), but the value, rose and fall values were flipped. I mistakenly derived my class from DigitalInOut instead of Debouncer. But instead of getting a no such method error on super().fell() or super().rose(), I got the hard fail.

Thereafter, I started getting other errors on the drive. Reloading circuitpython did not clear those. I finally recovered by reflashing the boot loader, reloading circuitpython, and reloading my program.

carson-coder, 2024-01-03:
I got into safe mode by doing
```import board
from digitalio import DigitalInOut, Direction, Pull
# from adafruit_debouncer import Debouncer

class InvertedDigitalInOut(DigitalInOut):
    """A class that provides an inverted version of DigitalInOut pins."""
	pass

btn = InvertedDigitalInOut(board.A1)
btn.direction = Direction.INPUT
btn.pull = Pull.UP

print("okay")```

carson-coder, 2024-01-03:
It looks like a issue with deriving a class from DigitalInOut

dhalbert, 2024-01-03:
In general it does not work to subclass a native class, unless special arrangements are made in the C code. Duplicate of #7305.

closed 2024-01-03 by dhalbert

carson-coder, 2024-01-03:
I could add a error whenever someone tries to subclass a native class.

dhalbert, 2024-01-03:
Continue discussion in #7305. There _are_ native classes that can be subclassed if they take certain steps internally.