adafruit/circuitpython #8785: Add 4-Color Waveshare EPaper-Display to displayio/epaperdisplay

josch0, 2024-01-05:
I would like to use the following Waveshare EPaper display with CircuitPython.

https://www.waveshare.com/product/displays/e-paper/epaper-1/7.3inch-e-paper-g.htm

It is a 4-color display (Red/Yellow/Black/White).
4 pixels per byte are transmitted (= 2 bit color depth).

Black: 00b
White: 01b
Yellow: 10b
Red: 11b

I have seen in the current C source code that the 7-color display from Waveshare is already supported. Unfortunately, I am not able to contribute C code for the 4-color variant myself.

There is already a display driver for the 7-color display (https://github.com/adafruit/Adafruit_CircuitPython_ACeP7In)
I would provide a driver for the 4-color variant as soon as Circuit-Python supports 4 color mode.

@tannewt 
Could you add the 4 color mode? I guess the implementation is not very complex, based on the existing code of the 7-color variant.

Thanks!
Joachim

carson-coder, 2024-01-06:
I could try to add this but I don't have a epaper display

joshua-beck-0908, 2024-01-07:
Yeah, I'm trying to get four bits per pixels working for a greyscale eInk display, so it would be great to just have a parameters for the colour depth. Current implementation seems a little limited. I'll have a look.

tannewt, 2024-01-16:
I don't plan on adding two highlight color support anytime soon because Adafruit doesn't currently carry the display.