adafruit/circuitpython #8788: Samsung Magican causes Circuitpython on RP2040 to hard fault

drath42, 2024-01-06:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; Cytron Maker Pi RP2040 with rp2040

(issue also occurs with CircuitPython 9.0.0-alpha.6)
```

### Code/REPL

```python
n/a
```

### Behavior

When using Windows with Samsung Magican (SSD utility) installed (actually it's the SamsungMagicanSVC running) the board boots using the out-of-box demo code that comes with the Maker PI RP2040 but hard faults after a few seconds. The CIRCUITPY drive gets listed in Windows explorer but isn't accessible. After a few more seconds an unknown device is detected.
On the serial port, the following is printed (not reliably, as the serial port also dies along with the reset of the board):
`Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.`

### Description

This is actually documented here: https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#circuitpy-drive-does-not-appear-or-disappears-quickly-3004181
Since I like to keep my SSD's firmware updated and their health checked, uninstalling Samsung Magician isn't an option for me. I debugged this, and noticed that the Magician apparently causes accesses past the end of the device, e.g.
669	7.613582	host	2.2.2	USBMS	58	SCSI: Read(10) LUN: 0x00 (LBA: 0x00123420, Len: 1)
while the device reports only a capacity of 2048 blocks (1 MB).

### Additional information

Disabling the SamsungMagicianSVC in Windows resolves the problem, but is cumbersome.
It seems that there's no bounds checking in the code to make sure the read command is within the capacity of the emulated drive. I added a check of the LBA against the drive's capacity in tud_msc_read10_cb and return -1 in case it is out of bounds. With that change, the drive works even with Samsung Magician running.

Ve2mrx, 2024-01-14:
I'm affected by this bug as well.

dhalbert, 2024-01-14:
> I added a check of the LBA against the drive's capacity in tud_msc_read10_cb and return -1 in case it is out of bounds. With that change, the drive works even with Samsung Magician running.

Could you submit this as a PR? Thanks very much!

dhalbert, 2024-01-22:
@drath42 if not a PR could you just provide a diff? Thanks.

drath42, 2024-01-22:
> @drath42 if not a PR could you just provide a diff? Thanks.

Sorry for not replying earlier. I was hoping to find time for this last weekend. A diff is easier:

```diff
diff --git a/supervisor/shared/usb/usb_msc_flash.c b/supervisor/shared/usb/usb_msc_flash.c
index f212d55270..2ded6de2b2 100644
--- a/supervisor/shared/usb/usb_msc_flash.c
+++ b/supervisor/shared/usb/usb_msc_flash.c
@@ -171,10 +171,16 @@ bool tud_msc_is_writable_cb(uint8_t lun) {
 // Copy disk's data to buffer (up to bufsize) and return number of copied bytes.
 int32_t tud_msc_read10_cb(uint8_t lun, uint32_t lba, uint32_t offset, void *buffer, uint32_t bufsize) {
     (void)offset;
+    uint32_t disk_block_count;

     const uint32_t block_count = bufsize / MSC_FLASH_BLOCK_SIZE;

     fs_user_mount_t *vfs = get_vfs(lun);
+    disk_ioctl(vfs, GET_SECTOR_COUNT, &disk_block_count);
+
+    if (lba + block_count > disk_block_count)
+           return -1;
+
     disk_read(vfs, buffer, lba, block_count);

     return block_count * MSC_FLASH_BLOCK_SIZE;
@@ -195,7 +201,7 @@ int32_t tud_msc_write10_cb(uint8_t lun, uint32_t lba, uint32_t offset, uint8_t *
     // MicroPython let's update the cached FatFs sector if it's the one
     // we just wrote.
     #if FF_MAX_SS != FF_MIN_SS
-    if (vfs->fatfs.ssize == MSC_FLASH_BLOCK_SIZE) {
+    if (vfs->ssize == MSC_FLASH_BLOCK_SIZE) {
     #else
     // The compiler can optimize this away.
     if (FF_MAX_SS == FILESYSTEM_BLOCK_SIZE) {
```

Not sure how proper this fix is, especially since there's no way to return a differentiated error code, but this "works for me".

Regards,

Dominic

dhalbert, 2024-01-23:
I wonder if the validation should be pushed down into `disk_read()` or what it calls: ` mp_vfs_blockdev_read()`. That makes those routines safer to call from elsewhere.

@drath42 I don't immediately see the reason for this change:
```c
-    if (vfs->fatfs.ssize == MSC_FLASH_BLOCK_SIZE) {
+    if (vfs->ssize == MSC_FLASH_BLOCK_SIZE) {
```
I could probably figure it out, but I'll ask you instead :slightly_smiling_face:

drath42, 2024-01-23:
> I wonder if the validation should be pushed down into `disk_read()` or what it calls: ` mp_vfs_blockdev_read()`. That makes those routines safer to call from elsewhere.

I took some notes on the callstack up to the point where it crashes, but I didn't spend too much time figuring out how this code is used outside of my immediate problem. I'll look into it later (I hope).

> @drath42 I don't immediately see the reason for this change:
> 
...
> 
> I could probably figure it out, but I'll ask you instead 🙂

That is a very good point. It's probably the result of copying over the change from my old source tree to a fresh checkout, thus reverting some intermediate change (https://github.com/adafruit/circuitpython/commit/a5578cf0678625a246c3e2864aac27f75bf651a0)... aka "let me just try this real quickly" ;-)

dhalbert, 2024-02-16:
@hathach This problem and potential fix is probably of interest to you too, since it could affect other USB MSC uses.

closed 2024-02-16 by dhalbert

aaronse, 2024-09-20:
Thank you!  My RP2040 based Adafruit Macropad was also failing/crashing on Windows 11 when Samsung Magician is installed.  Temporarily uinstalled so I can update to newer CircuitPython with this fix.  Cheers!

PR #8933 (2024-02-16): Prevent out-of-range reads via msc
Some ports implement their own protection against this at a low level but it doesn't hurt to add it here instead.

Based on a patch from @drath42

Closes: #8788
commit 339d01f5 (2024-02-16): Prevent out-of-range reads via msc

Some ports implement their own protection against this at a low level
but it doesn't hurt to add it here instead.

Based on a patch from @drath42

Closes: #8788