adafruit/circuitpython #8790: wifi.Radio: add method to get obtain connected devices in access point mode

ultranoise, 2024-01-06:
The core module wifi.Radio does not provide any method to know if there is any device connected to the board when it is configured as access point. I would like to suggest the development of the following method:

- .connected_clients -> list of of connected devices
- connected_clients[].IP -> IP address of a particular connected device

For applications when we have to transmit sensor data to connected devices (e.g. musical controllers), this is radically important. We usually do not receive any communication from the connected clients where they would expose their IP.

anecdata, 2024-01-06:
What protocol is being used to communicate from the AP to the connected stations?

I think it would involve capturing the `IP_EVENT_AP_STAIPASSIGNED` event and handling this structure:
https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32s2/api-reference/network/esp_netif.html#_CPPv427ip_event_ap_staipassigned_t

ultranoise, 2024-01-06:
A typical use would be using OpenSoundControl (application level protocol), which is usually implemented with UDP. When I coded this in C or Arduino there were methods to obtain the clients' information. Ideally, it would be great to obtain it directly from wifi.Radio

anecdata, 2024-01-06:
In the mean time, I don't know if you have control over which order the stations connect to the AP, but the AP will be 192.168.4.1, and stations get assigned sequentially starting with 192.168.4.2. You can set a maximum number of stations allowed to connect in the `start_ap()` call (up to 10).

ultranoise, 2024-01-06:
Thanks for the tip. In fact, this is what I have engineered at this moment. It really works well but it will definitely would be great to deal with this automatically :) I hope this could be developed!

anecdata, 2024-01-07:
Can you describe the user case more? Are all connected stations equivalent, with the same packets going out to each, or is there some other way to know what each one can do?

I think the event handler process is more complex since stations connecting (and then obtaining an IP address) and later disconnecting means state for each station will have to be tracked in real time and persist until the next change.

The latter alternative above is a call to get a list of the IPs (and potentially MACs and `AID`s) as needed - a snapshot of the current state of connected stations.

ultranoise, 2024-01-07:
In my opinion it doesn't matter if different packets go to different clients if an enough informative event handler is created. In the past, I coded these applications in Arduino using similar code to this: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino

The device transmiting would implement the following event handlers:

``

anecdata, 2024-01-07:
I wouldn't change the title with the assumption of an implementation detail.

There is a similar [event handler](https://github.com/adafruit/circuitpython/blob/bbff8b56381ee8595b175be350ddbdf10a2afcdc/ports/espressif/common-hal/wifi/__init__.c#L72) already in the core for `espressif` port, but in general we don't print from the core. There would still need to be an API to query the core from CircuitPython user code to get the (current) IP(s), since there is no way to interrupt python code from the core.

Bullet point 1 above is addressed with `wifi.radio.ap_active` (and the presence of an AP IP address).

Bullet points 2 and 3 above for wifi station are addressed with `wifi.radio.connected` (and the presence of a station IP address).

A complication is that the API should be the same for `espressif` and `raspberrypi`. I don't know what is possible within the `raspberry` port to deliver IP addresses of connected stations, but that would need some research before implementing a feature for `espressif`.

romkey, 2024-01-21:
I've been reading the raspberrypi SDK documentation and I'm not 100% confident about this but it looks like there's no exposed API for querying the DHCP server that's started in AP mode, only to start and stop the server. That said, this is C and it may be possible to extract this from the internal data structures that the DHCP server exposes, which would work but obviously be a horrible coding practice and likely to break without warning with an SDK update.

romkey, 2024-01-22:
I posted [a PR](https://github.com/adafruit/circuitpython/pull/8820) which implements a new property returning a list of associated stations, and another new property that returns the maximum number of stations that can be associated with the device. It works with Espressif CPUs and (with some limitations) the Pi Pico W.

dhalbert, 2024-03-08:
Fixed by #8820.

closed 2024-03-08 by dhalbert