adafruit/circuitpython #8793: E-Ink displays not working with 9.0.0-alpha.6

rsbohn, 2024-01-09:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.6 on 2023-12-12; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
import board
import busio
import digitalio
import displayio
import os
from adafruit_ssd1680 import SSD1680
from terminalio import FONT
from adafruit_bitmap_font.bitmap_font import load_font
from adafruit_display_text.label import Label

# EQUIPMENT
# Raspberry Pi Pico https://www.adafruit.com/product/4864
# 2.13" Diagonal E-Ink DIsplay https://www.adafruit.com/product/4197
# SOFTWARE
# CircuitPython 9.0.0 alpha.6
# circup install --py adafruit_ssd1680
# circup install --py adafruit_display_text

displayio.release_displays()
spi = busio.SPI(clock=board.GP18, MOSI=board.GP19)

# display bus
bus = displayio.FourWire(
    spi, command=board.GP21, chip_select=board.GP20, reset=board.GP22, baudrate=1_000_000
)
# Create the display object
display = SSD1680(bus, width=250, height=122, busy_pin=board.GP17, rotation=270)

group = displayio.Group()

FONT = load_font("/fonts/Arial-12.pcf")
label = Label(FONT, text=os.uname().version, color=0x999999, x=12, y=display.height-22)
group.append(label)
display.root_group = group

display.refresh()
```

### Behavior

The E-Ink display does nothing, as if it were not even connected.

### Description

If you comment out the code that sets FONT to Arial-12 it does work. It clears the display and shows the correct Circuitpython version using the terminalio.FONT.
If you run with 8.2.9 it works with the Arial font and with the terminalio font.

### Additional information

Saw the same behavior with the 2.9" Grayscale eInk Display FeatherWing. https://www.adafruit.com/product/4777

rsbohn, 2024-01-13:
The font Arial-12 is from here: [MagTag Google Calendar/fonts](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/MagTag/MagTag_Google_Calendar/fonts/Arial-12.pcf)

FoamyGuy, 2024-01-16:
I also see this issue on a Magtag under certain circumstances. I tested a few different things with somewhat perplexing results.

This basic label test adapted from the code pasted above does seem to reliably reproduce the issue. When this runs the display does not refresh at all. It will continue showing whatever was on it previously. It does print "after refesh()" but nothing happens on display
```py
import time
import busio
import board
import displayio
import fourwire
import adafruit_il0373
from adafruit_bitmap_font.bitmap_font import load_font
from adafruit_display_text.label import Label
import os

displayio.release_displays()

# This pinout works on a Feather M4 and may need to be altered for other boards.
spi = busio.SPI(board.EPD_SCK, board.EPD_MOSI)  # Uses SCK and MOSI
epd_cs = board.EPD_CS
epd_dc = board.EPD_DC

display_bus = fourwire.FourWire(
    spi, command=epd_dc, chip_select=epd_cs, baudrate=1000000
)
time.sleep(1)

display = adafruit_il0373.IL0373(
    display_bus,
    width=296,
    height=128,
    rotation=270,
    black_bits_inverted=False,
    color_bits_inverted=False,
    grayscale=True,
    refresh_time=1,
)

group = displayio.Group()

FONT = load_font("/fonts/Arial-12.pcf")
label = Label(FONT, text=os.uname().version, color=0x999999, x=12, y=display.height-22)
group.append(label)
#display.root_group = group
display.root_group = displayio.CIRCUITPYTHON_TERMINAL
display.refresh()

print("after refresh()")
while True:
    pass
```

this simple image test with manual display setup seems to run and draw on the display successfully:
```py

import time
import busio
import board
import displayio
import fourwire
import adafruit_il0373
import os

displayio.release_displays()

# This pinout works on a Feather M4 and may need to be altered for other boards.
spi = busio.SPI(board.EPD_SCK, board.EPD_MOSI)  # Uses SCK and MOSI
epd_cs = board.EPD_CS
epd_dc = board.EPD_DC

display_bus = fourwire.FourWire(
    spi, command=epd_dc, chip_select=epd_cs, baudrate=1000000
)
time.sleep(1)

display = adafruit_il0373.IL0373(
    display_bus,
    width=296,
    height=128,
    rotation=270,
    black_bits_inverted=False,
    color_bits_inverted=False,
    grayscale=True,
    refresh_time=1,
)

g = displayio.Group()

pic = displayio.OnDiskBitmap("magtag_ruler.bmp")

# CircuitPython 7 compatible only
t = displayio.TileGrid(pic, pixel_shader=pic.pixel_shader)
t.x = 0
g.append(t)

display.root_group = g

display.refresh()
print("after refresh going to main loop")

while True:
    pass
```

Lastly I tried with automatic display using `board.DISPLAY` and the label does seem to show correctly in that case. 

It seems that whatever the issues is, it's tied to the display being manually setup rather than built-in and on board.DISPLAY

FoamyGuy, 2024-01-17:
After a little more tinkering and testing I think that this line is somehow relevant to the behavior. 
```
FONT = load_font("/fonts/Arial-12.pcf")
```

I found that if I comment this line out and use the built-in font the label will show properly.

I also tested commenting this out and showing a blank Group, and `CIRCUITPYTHON_TERMINAL` which both also showed successfully while this line was commented out. 

I took a guess that maybe somehow clobbering the name `FONT` could be problematic and changed the variable name to `custom_font` but it still seems to have the same behavior (commented out the script runs and display updates, un-commented the display does nothing)

I don't really have any guesses as to why loading the font could cause this kind of impact though, and it seems especially odd that it doesn't happen when using `board.DISPLAY` instead of initializing the display from python code.

rsbohn, 2024-01-17:
This code works on 8.2.9 but not on 9.0.0:

```python
use_arial = True
bg_rect = vectorio.Rectangle(pixel_shader=px, x=0, y=0, width=display.width, height=display.height)
group.append(bg_rect)
from terminalio import FONT
label = Label(FONT, text=os.uname().version, color=0, x=12, y=24)
group.append(label)
if use_arial:
    Arial12 = load_font("/fonts/Arial-12.pcf")
    label2 = Label(Arial14, text="Arial-12.pcf", color=0, x=12 ,y=48)
    group.append(label2)
display.root_group = group
```

If you change `use_arial` to False it will work on 9.0.0 (because you're not loading the Arial font.)

rsbohn, 2024-01-26:
> Adafruit CircuitPython 9.0.0-alpha.6-46-gd7716de968 on 2024-01-25; Raspberry Pi Pico with rp2040

Fix validated. Thanks!

PR #8839 (2024-01-25): Collect refresh sequence pointer
Otherwise it will be freed during a collect and potentially overwritten. This is a bug in 8.x but isn't seen as early as in 9.x because 9.x will collect before expanding the split heap further.

Fixes #8793
PR #8843 (2024-01-26): Collect refresh sequence pointer: backport to 8.2.x of #8839
Backport of #8839:
Otherwise it will be freed during a collect and potentially overwritten. This is a bug in 8.x but isn't seen as early as in 9.x because 9.x will collect before expanding the split heap further.

Fixes #8793
commit 461d1a9a (2024-01-26): Collect refresh sequence pointer

Otherwise it will be freed during a collect and potentially
overwritten. This is a bug in 8.x but isn't seen as early as in
9.x because 9.x will collect before expanding the split heap
further.

Fixes #8793