adafruit/circuitpython #8794: RTC does not save 'tm_isdst'

ilikecake, 2024-01-13:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.8 on 2023-11-16; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
```

### Code/REPL

```python
import time
import rtc

now = time.struct_time((2024, 1, 12, 21, 51, 40, 4, 12, 0))
print(now)
rtc.RTC().datetime = now
now = time.localtime()
print(now)
```

### Behavior

`struct_time(tm_year=2024, tm_mon=1, tm_mday=12, tm_hour=21, tm_min=51, tm_sec=40, tm_wday=4, tm_yday=12, tm_isdst=0)`
`struct_time(tm_year=2024, tm_mon=1, tm_mday=12, tm_hour=21, tm_min=51, tm_sec=40, tm_wday=4, tm_yday=12, tm_isdst=-1)`

### Description

It appears that the `tm_isdst` is not saved as part of the RTC. It does not thrown an error, but it always returns -1 when getting the time from the RTC. From the example above, I set the `tm_isdst` to 0, but when I get the time from the RTC, it returns -1.

I am not sure if this is something specific to the ESP32, or with the RTC library more generally.

dhalbert, 2024-01-16:
When you set the RTC, the number of seconds since some start date (the "epoch") is recorded. All the other info is discarded. One of `tm_isdst`'s main uses is to disambiguate times that fall near the boundaries of the date-time when DST starts or stops. See https://stackoverflow.com/a/47333019/142996.

Since the timezone and location info are note saved in the RTC, the DST state is not really interesting. You could save the timezone and location info if needed. Note that the timezone does not include all the info needed to determine DST state either. For instance, some states in the US observe daylight savings time, and others in the same timezone do no (e.g., Arizona).

dhalbert, 2024-01-22:
Closing for now.

closed 2026-11-13 by dhalbert