adafruit/circuitpython #8797: Hard Fault on CircuitPython 9.0.0 Alpha 6, it works OK on CircuitPython 8.2.9

davesmeghead, 2024-01-14:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.6 on 2023-12-12; Adafruit QT Py ESP32-S3 4MB Flash 2MB PSRAM with ESP32S3
Board ID:adafruit_qtpy_esp32s3_4mbflash_2mbpsram
UID:4F21AFA53806
```

### Code/REPL

```python
class ClassB:
    def __init__(self, bb=None, cc = None, packet_callback = None):
        None

class ClassA(ClassB):
    def __init__(self, *args, **kwargs) -> None:
        super().__init__(packet_callback=self.processReceivedPacket, *args, **kwargs)

    # This function handles a received message packet and processes it
    def processReceivedPacket(self, packet):
        None

print("Here Start")
v = ClassA(cc=42, bb=56)
print("Here End")
```

### Behavior

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Crikey!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.0.0-alpha.6 on 2023-12-12; Adafruit QT Py ESP32-S3 4MB Flash 2MB PSRAM with ESP32S3

### Description

I think that I have narrowed down a bug with CP9, the code works in CP8.

dhalbert, 2024-01-19:
I reduced this to:
```py
def f(a):
    pass

print("Here Start")
#f(a=2, *())     # TypeError: unexpected keyword argument ''
f(a=f, *())      # Safe mode: Hard fault
print("Here End")
```
So it's not `super()` or anything like that. It has to do with the `*` arg.
If the `*()` arg is removed, it works fine.

If the first call to `f()` is used it doesn't crash, but gets the indicated keyword arg error.
If the second call to `f()` is used, it crashes with a hard fault.
I suspect this is the same error.

It works fine on 8.2.9.

dhalbert, 2024-01-20:
Testing with MicroPython v1.21.0 and v1.22.1 on a Feather M0 Express: In the above program, I get `TypeError: unexpected keyword argument` (note no `''`)  on both calls to `f()`.

Testing with MicroPython v1.22.1 on a PyBoard: As with CircuitPython, uncommenting the second call to `f()` causes a hard crash. I'll open an issue for MicroPython.