adafruit/circuitpython #8824: hard fault on serial disconnect on MacroPad RP2040

tlyu, 2024-01-22:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.6-33-g9f016796e0-dirty on 2024-01-22; Adafruit Macropad RP2040 with rp2040
```

### Code/REPL

```python
# N/A; also happens with freshly reformatted CIRCUITPY drive
```

### Behavior

```text
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

Happens after some random number of serial disconnects, either at the REPL or at the "code done running" prompt. Only tested using `tio` on macOS 13.6.3. The CIRCUITPY volume also unmounts, with a warning about unsafe unmount, so I'm guessing it actually detached from USB in the process.

tlyu, 2024-01-22:
Also apparently happens with

```
Adafruit CircuitPython 8.2.9 on 2023-12-06; Adafruit Macropad RP2040 with rp2040
```

so possibly not a new bug?

tlyu, 2024-01-22:
Does not happen with nRF52840:
```
Adafruit CircuitPython 8.2.9 on 2023-12-06; Adafruit ItsyBitsy nRF52840 Express with nRF52840
```
(it was a board that I conveniently already soldered a debug harness to)

I'm not sure I have a an RP2040 board with easily accessible SWD traces to hook up a debugger to.

tannewt, 2024-01-22:
> Happens after some random number of serial disconnects, either at the REPL or at the "code done running" prompt.

What is causing these disconnects?

tlyu, 2024-01-23:
I'm causing the disconnects by exiting `tio` by using Ctrl-T Q. That shouldn't cause a hard fault in CircuitPython.

tlyu, 2024-01-23:
This also happens on:
```
Adafruit CircuitPython 8.2.9 on 2023-12-06; Adafruit KB2040 with rp2040
```
so I'm guessing it's generally true of the RP2040 port.

tlyu, 2024-02-03:
Replicated on macOS 10.5.7 on an Intel MacBook Pro with
```
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; Adafruit Macropad RP2040 with rp2040
```
and `tio` v2.4.

tlyu, 2024-02-03:
I'd be happy to get a stack trace or even do a bit of debugging once I get a RP2040 board with accessible SWD pins, which I don't currently have. It might be a while before I get around to obtaining one.

If you have a spare Pico or even better, a Feather RP2040 with SWD header that you can send my way, that would be great. Otherwise, I probably won't get one until the next time I have an Adafruit order to add on to.

tlyu, 2024-02-03:
Additional observation: running this loop on a M2 MacBook Air will succeed for a long time (several minutes before I stopped it), but will tend to cause the crash once interrupted with Ctrl-C:
```
while :; do stty -a < /dev/cu.usbmodem1101; done
```
In the past, when tracing USB CDC traffic on a different device with a Beagle, I noticed that macOS will sometimes abort a CDC-related control transfer partway through, and I wonder if that's related.

ure, 2024-02-04:
Same issue here ... ;(

dhalbert, 2024-02-04:
@tlyu what version of macOS are you using?
@ure What is your OS, and if macOS, what kind of Mac?

tlyu, 2024-02-04:
Mentioned above: 13.6.3 on the M2 MBA; 10.5.7 on the Intel MBP. If it does happen to be a timing issue, CPU speed and system load might be a factor.

ure, 2024-02-04:
Al running sonoma 14.3 one mac mini m1, one studio m2, macbook m1. I can confirm that this does not occur on linux, don't have access to a windows computer

tlyu, 2024-02-04:
It's likely to be a lower-level problem, possibly with the Pico SDK. On a hunch, I uploaded a Blink sketch from the Arduino IDE, Philhower core v3.6.2. connecting and then disconnecting with `tio` causes the program to freeze and the device to become unresponsive on USB CDC until physically reset.

ure, 2024-02-04:
@tlyu ok super, if fixed upstream this will follow in a future release, nice ;)

tlyu, 2024-02-04:
Updated to 3.7.0 of the Philhower core; bug still present.

tlyu, 2024-02-04:
With the Pico SDK, using a `blink` example modified to enable CDC for stdio, though not outputting anything to it, connecting and disconnecting from the CDC serial port a few times using `tio` freezes the program and makes CDC unresponsive.

dhalbert, 2024-02-05:
> With the Pico SDK, using a `blink` example modified to enable CDC for stdio, though not outputting anything to it, connecting and disconnecting from the CDC serial port a few times using `tio` freezes the program and makes CDC unresponsive.

This is very interesting. Pinging @hathach on this: this problem appears specific to macOS. This could be an issued opened either on https://github.com/hathach or https://github.com/raspberrypi/pico-sdk. @hathach would you prefer one repo or the other.

tlyu, 2024-02-05:
Update: the `hello_world/usb` Pico SDK example also fails in a similar way, though somewhat less obviously than a blinking LED ceasing to blink.

tannewt, 2024-02-21:
@hathach Mind looking at this?

eightycc, 2024-02-22:
Crash is readily reproducible using `tio` when disconnecting by `ctrl-T Q`. 

So far, I've found the crash is due to `panic()` being called under one or the other of these conditions:

- `rp2040_usb.c:_hw_endpoint_buffer_control_update32()` when `USB_BUF_CTRL_AVAIL` is already set in the endpoint's `buffer_control`,
- `rp2040_usb.c:hw_endpoint_xfer_continue()` when `!ep->active`.

tlyu, 2024-02-24:
This is starting to look like a possible hardware erratum where the device controller doesn't clear `USB_BUF_CTRL_AVAIL` on EP0 when a SETUP packet arrives before the OUT packet from a preceding control write. In the first case, there's a panic because TinyUSB finds the flag already set. In the second case, TinyUSB gets a completion interrupt from the OUT packet arriving before it configures EP0 to receive the data stage, because the device controller accepts the OUT DATA due to the flag already being set from the previous (aborted) transfer.

If this is an erratum, it can't be worked around in the general case, because there's no guarantee that the aborted control transfer has the same data length as the current one. Also, there's no guarantee that TinyUSB can clear the errant flag before the device controller acts on it.

I still don't have a RP2040 board with accessible SWD, so this is analysis based on the TinyUSB source code and the RP2040 datasheet.

tlyu, 2024-02-24:
@eightycc thanks for the debugging! Could you possibly get full stack traces for those?

eightycc, 2024-02-24:
Here is a backtrace from a breakpoint at `panic()`:

```
Thread 1 hit Breakpoint 1, panic (fmt=0x0) at sdk/src/rp2_common/pico_runtime/runtime.c:307
307	void __attribute__((noreturn)) __printflike(1, 0) panic(const char *fmt, ...) {
(gdb) bt
#0  panic (fmt=0x0) at sdk/src/rp2_common/pico_runtime/runtime.c:307
#1  0x20000ba2 in _hw_endpoint_buffer_control_update32 (ep=ep@entry=0x2000d0cc <hw_endpoints+32>, and_mask=and_mask@entry=0, 
    or_mask=or_mask@entry=54272) at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:115
#2  0x20000c06 in _hw_endpoint_buffer_control_set_value32 (value=54272, ep=0x2000d0cc <hw_endpoints+32>)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.h:122
#3  hw_endpoint_start_next_buffer (ep=ep@entry=0x2000d0cc <hw_endpoints+32>)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:212
#4  0x10032596 in hw_endpoint_xfer_start (ep=ep@entry=0x2000d0cc <hw_endpoints+32>, buffer=buffer@entry=0x0, total_len=total_len@entry=0)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:245
#5  0x100322ac in hw_endpoint_xfer (ep_addr=128 '\200', buffer=buffer@entry=0x0, total_bytes=total_bytes@entry=0)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c:188
#6  0x1003245a in dcd_edpt_xfer (rhport=0 '\000', ep_addr=<optimized out>, buffer=buffer@entry=0x0, total_bytes=total_bytes@entry=0)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c:531
#7  0x10022eb0 in usbd_edpt_xfer (rhport=<optimized out>, ep_addr=<optimized out>, buffer=buffer@entry=0x0, total_bytes=total_bytes@entry=0)
    at ../../lib/tinyusb/src/device/usbd.c:1249
#8  0x100235a4 in _status_stage_xact (rhport=rhport@entry=0 '\000', request=request@entry=0x20040f08)
    at ../../lib/tinyusb/src/device/usbd_control.c:70
#9  0x1002362c in tud_control_status (rhport=0 '\000', request=request@entry=0x20040f08) at ../../lib/tinyusb/src/device/usbd_control.c:81
#10 0x10022660 in cdcd_control_xfer_cb (rhport=<optimized out>, stage=<optimized out>, request=0x20040f08)
    at ../../lib/tinyusb/src/class/cdc/cdc_device.c:383
#11 0x100228e6 in invoke_class_control (rhport=rhport@entry=0 '\000', driver=0x100dbe40 <_usbd_driver>, request=request@entry=0x20040f08)
    at ../../lib/tinyusb/src/device/usbd.c:611
#12 0x100231e0 in process_control_request (rhport=<optimized out>, p_request=p_request@entry=0x20040f08)
    at ../../lib/tinyusb/src/device/usbd.c:774
#13 0x1002341a in tud_task_ext (timeout_ms=timeout_ms@entry=4294967295, in_isr=in_isr@entry=false) at ../../lib/tinyusb/src/device/usbd.c:522
#14 0x10023e10 in tud_task () at ../../lib/tinyusb/src/device/usbd.h:54
#15 usb_background () at ../../supervisor/shared/usb/usb.c:170
#16 0x10023e34 in usb_background_do (unused=<optimized out>) at ../../supervisor/shared/usb/usb.c:190
#17 0x2000704e in background_callback_run_all () at ../../supervisor/shared/background_callback.c:102
#18 0x10020d86 in mp_hal_stdin_rx_chr () at ../../supervisor/shared/micropython.c:49
#19 0x1009082a in readline (line=line@entry=0x20040f88, prompt=prompt@entry=0x100cfa0c ">>> ") at ../../shared/readline/readline.c:616
#20 0x10095c3c in pyexec_friendly_repl () at ../../shared/runtime/pyexec.c:674
#21 0x1001f106 in run_repl (safe_mode=SAFE_MODE_NONE) at ../../main.c:962
#22 0x1001f762 in main () at ../../main.c:1100
```

And some light tracing in TinyUSB:
```
Adafruit CircuitPython 9.0.0-beta.2-6-g89437cf031-dirty on 2024-02-22; Raspberry Pi Pico W with rp2040hw_endpoint_xfer 82 0x20009d9c 64
hw_endpoint_xfer_start ep 2 in 64
hw_endpoint_xfer 82 0x20009d9c 38
hw_endpoint_xfer_start ep 2 in 38

hw_endpoint_xfer 82 0x20009d9c 2
hw_endpoint_xfer_start ep 2 in 2
>>> hw_endpoint_xfer 82 0x20009d9c 4
hw_endpoint_xfer_start ep 2 in 4
hw_endpoint_xfer 00 0x2000ac94 7
hw_endpoint_xfer_start ep 0 out 7
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 00 0x2000ac94 7
hw_endpoint_xfer_start ep 0 out 7
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 00 0x2000ac94 7
hw_endpoint_xfer_start ep 0 out 7
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 00 0x2000ac94 7
hw_endpoint_xfer_start ep 0 out 7
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
WARN: starting new transfer on already active ep 0 in
```

eightycc, 2024-02-24:
@tlyu The reason that the trace is light is that full level 3 tracing in TinyUSB affects timing to the extent that MacOS cannot enumerate the device. If you'd like more tracing, let me know what you'd like and I'll add it.

tlyu, 2024-02-24:
@eightycc thanks! That is helpful, even if the trace is missing the control request details. It looks like in this instance, there are back-to-back control writes with no data, so they're answered with a ZLP IN transaction (control status stage). The one that panics is because the AVAILABLE flag set by the previous status stage wasn't cleared by the device controller hardware when the transfer was aborted with a new SETUP packet.

eightycc, 2024-02-25:
@tlyu Here are the traces of single failure event. The Control Center capture is the zip file at the end.

```
Thread 1 hit Breakpoint 1, panic (fmt=0x0) at sdk/src/rp2_common/pico_runtime/runtime.c:307
307	void __attribute__((noreturn)) __printflike(1, 0) panic(const char *fmt, ...) {
(gdb) bt
#0  panic (fmt=0x0) at sdk/src/rp2_common/pico_runtime/runtime.c:307
#1  0x20000ba2 in _hw_endpoint_buffer_control_update32 (ep=ep@entry=0x2000d0cc <hw_endpoints+32>, and_mask=and_mask@entry=0, 
    or_mask=or_mask@entry=54272) at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:115
#2  0x20000c06 in _hw_endpoint_buffer_control_set_value32 (value=54272, ep=0x2000d0cc <hw_endpoints+32>)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.h:122
#3  hw_endpoint_start_next_buffer (ep=ep@entry=0x2000d0cc <hw_endpoints+32>)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:212
#4  0x10032596 in hw_endpoint_xfer_start (ep=ep@entry=0x2000d0cc <hw_endpoints+32>, buffer=buffer@entry=0x0, total_len=total_len@entry=0)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:245
#5  0x100322ac in hw_endpoint_xfer (ep_addr=128 '\200', buffer=buffer@entry=0x0, total_bytes=total_bytes@entry=0)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c:188
#6  0x1003245a in dcd_edpt_xfer (rhport=0 '\000', ep_addr=<optimized out>, buffer=buffer@entry=0x0, total_bytes=total_bytes@entry=0)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c:531
#7  0x10022eb0 in usbd_edpt_xfer (rhport=<optimized out>, ep_addr=<optimized out>, buffer=buffer@entry=0x0, total_bytes=total_bytes@entry=0)
    at ../../lib/tinyusb/src/device/usbd.c:1249
#8  0x100235a4 in _status_stage_xact (rhport=rhport@entry=0 '\000', request=request@entry=0x20040f08)
    at ../../lib/tinyusb/src/device/usbd_control.c:70
#9  0x1002362c in tud_control_status (rhport=0 '\000', request=request@entry=0x20040f08) at ../../lib/tinyusb/src/device/usbd_control.c:81
#10 0x10022660 in cdcd_control_xfer_cb (rhport=<optimized out>, stage=<optimized out>, request=0x20040f08)
    at ../../lib/tinyusb/src/class/cdc/cdc_device.c:383
#11 0x100228e6 in invoke_class_control (rhport=rhport@entry=0 '\000', driver=0x100dbe40 <_usbd_driver>, request=request@entry=0x20040f08)
    at ../../lib/tinyusb/src/device/usbd.c:611
#12 0x100231e0 in process_control_request (rhport=<optimized out>, p_request=p_request@entry=0x20040f08)
    at ../../lib/tinyusb/src/device/usbd.c:774
#13 0x1002341a in tud_task_ext (timeout_ms=timeout_ms@entry=4294967295, in_isr=in_isr@entry=false) at ../../lib/tinyusb/src/device/usbd.c:522
#14 0x10023e10 in tud_task () at ../../lib/tinyusb/src/device/usbd.h:54
#15 usb_background () at ../../supervisor/shared/usb/usb.c:170
#16 0x10023e34 in usb_background_do (unused=<optimized out>) at ../../supervisor/shared/usb/usb.c:190
#17 0x2000704e in background_callback_run_all () at ../../supervisor/shared/background_callback.c:102
#18 0x10020d86 in mp_hal_stdin_rx_chr () at ../../supervisor/shared/micropython.c:49
#19 0x1009082a in readline (line=line@entry=0x20040f88, prompt=prompt@entry=0x100cfa0c ">>> ") at ../../shared/readline/readline.c:616
#20 0x10095c3c in pyexec_friendly_repl () at ../../shared/runtime/pyexec.c:674
#21 0x1001f106 in run_repl (safe_mode=SAFE_MODE_NONE) at ../../main.c:962
#22 0x1001f762 in main () at ../../main.c:1100
```

```
Adafruit CircuitPython 9.0.0-beta.2-6-g89437cf031-dirty on 2024-02-22; Raspberry Pi Pico W with rp2040hw_endpoint_xfer 82 0x20009d9c 64
hw_endpoint_xfer_start ep 2 in 64
hw_endpoint_xfer 82 0x20009d9c 38
hw_endpoint_xfer_start ep 2 in 38

hw_endpoint_xfer 82 0x20009d9c 2
hw_endpoint_xfer_start ep 2 in 2
>>> hw_endpoint_xfer 82 0x20009d9c 4
hw_endpoint_xfer_start ep 2 in 4
hw_endpoint_xfer 00 0x2000ac94 7
hw_endpoint_xfer_start ep 0 out 7
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 00 0x2000ac94 7
hw_endpoint_xfer_start ep 0 out 7
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 00 0x2000ac94 7
hw_endpoint_xfer_start ep 0 out 7
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 00 0x2000ac94 7
hw_endpoint_xfer_start ep 0 out 7
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
WARN: starting new transfer on already active ep 0 in
```

[issue8824.zip](https://github.com/adafruit/circuitpython/files/14394773/issue8824.zip)

tlyu, 2024-02-25:
@eightycc thanks so much for the coordinated traces! The Beagle trace confirms what I've seen before in a previous context: some CDC control requests sent by macOS are aborted. It also confirms that my suspicions are likely correct, that the RP2040 hardware doesn't clear some state for a SETUP packet that it probably should.

There are 4 OUT control transfers of length 7, which are CDC Set Line Coding, followed by 3 no-data control transfers, which are CDC Set Control Line State.

The first 2 Set Control Line State are aborted, and simply doesn't poll the status stage for the third, for some reason.

There are 3 `hw_endpoint_xfer_start ep 0 in 0` in the trace, corresponding to the status stage of the Set Line Coding, followed by the status stages of the 2 Set Control Line State. The second one panics due to an existing active transfer.

dhalbert, 2024-02-25:
@tlyu Have you discussed these possible issues with the RP2040 folks? I think https://github.com/raspberrypi/pico-feedback is the right place. There are some USB hw issues there already.

I'm guessing that they licensed IP for the peripherals and did not design them themselves. But I might be wrong. @hathach would probably know.

tlyu, 2024-02-25:
@dhalbert not yet; thanks for the pointer! It only occurred to me that it might be a hardware erratum after seeing the stack traces and doing a close read of the TinyUSB code.

tlyu, 2024-02-25:
@eightycc please try hathach/tinyusb#2492 to see if that helps?

eightycc, 2024-02-25:
@tlyu Your patch fixes the `WARN: starting new transfer on already active ep 0 in` failure mode, but the `ep 0 in was already available` remains. I'll follow up with traces shortly.

eightycc, 2024-02-25:
```
Thread 1 hit Breakpoint 1, panic (fmt=0x0) at sdk/src/rp2_common/pico_runtime/runtime.c:307
307	void __attribute__((noreturn)) __printflike(1, 0) panic(const char *fmt, ...) {
(gdb) bt
#0  panic (fmt=0x0) at sdk/src/rp2_common/pico_runtime/runtime.c:307
#1  0x20000bc6 in _hw_endpoint_buffer_control_update32 (ep=ep@entry=0x2000d0ec <hw_endpoints+32>, and_mask=and_mask@entry=0, 
    or_mask=or_mask@entry=54272) at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:115
#2  0x20000c2a in _hw_endpoint_buffer_control_set_value32 (value=54272, ep=0x2000d0ec <hw_endpoints+32>)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.h:122
#3  hw_endpoint_start_next_buffer (ep=ep@entry=0x2000d0ec <hw_endpoints+32>)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:212
#4  0x10032596 in hw_endpoint_xfer_start (ep=ep@entry=0x2000d0ec <hw_endpoints+32>, buffer=buffer@entry=0x0, total_len=total_len@entry=0)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:245
#5  0x100322ac in hw_endpoint_xfer (ep_addr=128 '\200', buffer=buffer@entry=0x0, total_bytes=total_bytes@entry=0)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c:188
#6  0x1003245a in dcd_edpt_xfer (rhport=0 '\000', ep_addr=<optimized out>, buffer=buffer@entry=0x0, total_bytes=total_bytes@entry=0)
    at ../../lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c:538
#7  0x10022eb0 in usbd_edpt_xfer (rhport=<optimized out>, ep_addr=<optimized out>, buffer=buffer@entry=0x0, total_bytes=total_bytes@entry=0)
    at ../../lib/tinyusb/src/device/usbd.c:1249
#8  0x100235a4 in _status_stage_xact (rhport=rhport@entry=0 '\000', request=request@entry=0x20040f08)
    at ../../lib/tinyusb/src/device/usbd_control.c:70
#9  0x1002362c in tud_control_status (rhport=0 '\000', request=request@entry=0x20040f08) at ../../lib/tinyusb/src/device/usbd_control.c:81
#10 0x10022660 in cdcd_control_xfer_cb (rhport=<optimized out>, stage=<optimized out>, request=0x20040f08)
    at ../../lib/tinyusb/src/class/cdc/cdc_device.c:383
#11 0x100228e6 in invoke_class_control (rhport=rhport@entry=0 '\000', driver=0x100dbe40 <_usbd_driver>, request=request@entry=0x20040f08)
    at ../../lib/tinyusb/src/device/usbd.c:611
#12 0x100231e0 in process_control_request (rhport=<optimized out>, p_request=p_request@entry=0x20040f08)
    at ../../lib/tinyusb/src/device/usbd.c:774
#13 0x1002341a in tud_task_ext (timeout_ms=timeout_ms@entry=4294967295, in_isr=in_isr@entry=false) at ../../lib/tinyusb/src/device/usbd.c:522
#14 0x10023e10 in tud_task () at ../../lib/tinyusb/src/device/usbd.h:54
#15 usb_background () at ../../supervisor/shared/usb/usb.c:170
#16 0x10023e34 in usb_background_do (unused=<optimized out>) at ../../supervisor/shared/usb/usb.c:190
#17 0x2000706e in background_callback_run_all () at ../../supervisor/shared/background_callback.c:102
#18 0x10020d86 in mp_hal_stdin_rx_chr () at ../../supervisor/shared/micropython.c:49
#19 0x1009082a in readline (line=line@entry=0x20040f88, prompt=prompt@entry=0x100cfa0c ">>> ") at ../../shared/readline/readline.c:616
#20 0x10095c3c in pyexec_friendly_repl () at ../../shared/runtime/pyexec.c:674
#21 0x1001f106 in run_repl (safe_mode=SAFE_MODE_NONE) at ../../main.c:962
#22 0x1001f762 in main () at ../../main.c:1100
```
```
Adafruit CircuitPython 9.0.0-beta.2-6-g89437cf031-dirty on 2024-02-25; Raspberry Pi Pico W with rp2040hw_endpoint_xfer 82 0x20009dbc 64
hw_endpoint_xfer_start ep 2 in 64
hw_endpoint_xfer 82 0x20009dbc 38
hw_endpoint_xfer_start ep 2 in 38

hw_endpoint_xfer 82 0x20009dbc 2
hw_endpoint_xfer_start ep 2 in 2
>>> hw_endpoint_xfer 82 0x20009dbc 4
hw_endpoint_xfer_start ep 2 in 4
hw_endpoint_xfer 00 0x2000acb4 7
hw_endpoint_xfer_start ep 0 out 7
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 00 0x2000acb4 7
hw_endpoint_xfer_start ep 0 out 7
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 00 0x2000acb4 7
hw_endpoint_xfer_start ep 0 out 7
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 00 0x2000acb4 7
hw_endpoint_xfer_start ep 0 out 7
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
hw_endpoint_xfer 80 0x0 0
hw_endpoint_xfer_start ep 0 in 0
WARN: starting new transfer on already active ep 0 in
```
[issue8824-2.tdc.zip](https://github.com/adafruit/circuitpython/files/14397335/issue8824-2.tdc.zip)

tlyu, 2024-02-25:
@eightycc thanks! Which RP2040 board are you using, and what is its CPU clock speed? I guess it could also be down to a host-side timing issue as to why I'm not longer seeing failures and you are.

There's some TinyUSB-side state that I needed to reset as well, which explains the "already active ep 0". There's also a delay missing from the buffer control setting, which might be necessary to ensure that the AVAILABLE flag is already clear before proceeding with the rest of the request handling.

tlyu, 2024-02-25:
@eightycc I updated the patch in the TinyUSB PR; please try it when you get the chance?

tlyu, 2024-02-25:
Thanks, please try again? I had a precedence oops previously with bitwise operators.

tlyu, 2024-02-25:
I might need to add an explicit clear of the ABORT_DONE register. I'm not sure when I'll get to it yet.

tlyu, 2024-02-25:
@eightycc what chip revisions do you have? Apparently revisions earlier than RP2040B2 have an erratum (RP2040-E2) that causes endpoint aborts to NAK the endpoint forever.

eightycc, 2024-02-25:
@tlyu The Pico W I've been testing with is B2, the Pico that fails to enumerate with your patch is a B1.

eightycc, 2024-02-26:
Note that `hw_endpoint_lock_update()` is null, so calls to `hw_endpoint_start_next_buffer()` are unprotected against concurrent calls from user and IRQ level routines.

tlyu, 2024-02-26:
I think that I might have to accept a larger race condition window on B1 hardware and earlier, by only doing the ABORT for the chip versions where it works.

tlyu, 2024-02-26:
@eightycc updated the PR to only use ABORT on hardware B2 and later. This means the workaround should work in some cases on B1 and earlier, if the firmware wins the race condition. Please give it a try when you have a chance.

(The failure to enumerate on B1 is consistent with the description of the erratum RP2040-E2.)

eightycc, 2024-02-26:
@tlyu I've done some testing with your PR and find that it works with B2 but B1 still has problems.

I believe that the root cause is the race condition in `hw_endpoint_start_next_buffer()` noted earlier. Saving the interrupt state and disabling interrupts at the start of `hw_endpoint_start_next_buffer()` and then restoring interrupt state at its exit resolves the problem for both B1 and B2. I am able to build with `-Og` and TinyUSB level-3 tracing with no difficulties.

eightycc, 2024-02-27:
@tlyu I've been testing with both patches in place on both B1 and B2 RP2040s and cannot get it to fail.

You can find my patch here: https://github.com/eightycc/tinyusb/tree/issue_8824. You can add it to your PR.

hathach, 2024-03-05:
thank you @tlyu and @eightycc for detailed explanation, I have pull out my good-old macbook with mac0s 12.6 and it seems to be able to reproduce the issue. I will do start to investigate and review your PR

hathach, 2024-03-13:
tinyusb pr is merged, cpy PR is on the way https://github.com/adafruit/circuitpython/pull/9042 . @tlyu @eightycc would you mind testing out the cpy PR to confirm if issue is fixed. Thank you.

dhalbert, 2024-03-13:
Moving this back to 9.0.0 assuming it's fixed.

PR #9042 (2024-03-13): update tinyusb to fix rp2040 multiple setup
- update tinyusb to include fix for multiple setup packets (reproduced by using macos) https://github.com/hathach/tinyusb/pull/2492
- fix #8824
closed 2024-03-13 by dhalbert