adafruit/circuitpython #8829: High duty cycle connectable directed advertising

hktklsh, 2024-01-24:
The minimum advertising interval is 20ms. I want to use a smaller interval time. Is there any way to support for high duty cycle connected direct advertising?

dhalbert, 2024-01-24:
The 20ms minimum is enforced by the underlying nRF software, and is part of the the BLE standard.

hktklsh, 2024-01-24:
> The 20ms minimum is enforced by the underlying nRF software, and is part of the the BLE standard.

In this file, it seems to be supported
[image]
[image]

dhalbert, 2024-01-27:
Hmm, I see what you mean. I'll take a look. Right now if you can do builds yourself you could just remove the range check and see if it works. If you could write a simple example for testing that would be great.

dhalbert, 2024-02-23:
@hktklsh Could you give me a small test program or a sample advertisement to try? If you have a piece of an example that is not in CircuitPython, that is fine too.

hktklsh, 2024-02-28:
> @hktklsh Could you give me a small test program or a sample advertisement to try? If you have a piece of an example that is not in CircuitPython, that is fine too.

I use one 52840 to send and another 52840 to receive, and found that some data will be lost.
send：
[image]

receive:
[image]

Some data is lost
[image]

dhalbert, 2024-02-28:
I did not realize you were using this for `adafruit_ble_radio`.

The lost data is not surprising. Advertising is not a reliable way to deliver data. It's like UDP. But it can be used as a best-effort way.

My understanding is that the motivation for high-speed advertising is mainly to speed up reconnections, not to deliver data quickly.

hktklsh, 2024-02-28:
In the file adafruit_ble_radio.py, set AD_DURATION = 0.12 and interval = 0.02, it's possible not to lose data, I think if the interval can be smaller, AD_DURATION can be set smaller too。
[image]