adafruit/circuitpython #8846: Unable to initialize larger display sizes with Matrix Portal S3

identifystation, 2024-01-26:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; Adafruit MatrixPortal S3 with ESP32S3
```

### Code/REPL

```python
import displayio
import board
import rgbmatrix
import framebufferio

displayio.release_displays()
DISPLAY_WIDTH = 320
DISPLAY_HEIGHT = 160
DISPLAY_ROTATION = 0
BIT_DEPTH = 3
AUTO_REFRESH = False

matrix = rgbmatrix.RGBMatrix(
    width=DISPLAY_WIDTH, height=DISPLAY_HEIGHT, bit_depth=BIT_DEPTH,
    rgb_pins=[
        board.MTX_R1,
        board.MTX_G1,
        board.MTX_B1,
        board.MTX_R2,
        board.MTX_G2,
        board.MTX_B2],
    addr_pins=[board.MTX_ADDRA, board.MTX_ADDRB, board.MTX_ADDRC, board.MTX_ADDRD],
    clock_pin=board.MTX_CLK,
    latch_pin=board.MTX_LAT,
    output_enable_pin=board.MTX_OE,
    tile=5,
    serpentine=True,
    doublebuffer=True)

display = framebufferio.FramebufferDisplay(matrix)
```

### Behavior

I'm attempting to use a MatrixPortal S3 to initialize a 320x160 display made up of 25 64x32 LED panels in a 5x5 configuration.  Doing so produces the error `ValueError: framebuffer length must be >= 102400`.

If I change the size to 320x96 with `tile=3`, things work as expected.

dhalbert, 2024-01-26:
I talked with @identifystation and @DJDevon3 on discord about this. The error is from the `FrameBufferDisplay` checking the `RGBMatrix` framebuffer size and finding that it doesn't match the size it's expecting. I don't know whether there's an unreported allocation failure in `RGBMatrix` or there's some size-computation issue.

DJDevon3, 2024-01-27:
Seeing as how it would require at least 20 panels to replicate this issue there will have to be a way to emulate the buffer data required to throw that error.  That is definitely dev territory and way over my head.

I have not encountered that issue with 12 panels and according to the dimensions listed @identifystation could successfully get up to 15 panels.  Even if it's a framebuffer size limitation or miscalculation I have my doubts that a single Matrix Portal S3 will have the signal strength to traverse the 25+ feet of copper unless a signal booster is used.

We've been in touch via DM's and both of us have no idea if each matrix panel has a signal booster out from each panel.  It would make sense if they do because they're designed to be chained together but we don't know for certain. If not then it might be impossible to drive that distance without needing to introduce some kind of custom signal booster PCB.  I have no idea if this might be related, my gut says no but it's worth bringing up as they could potentially run into that issue regardless. Might be a question for Phil B?

Apparently 12 panels is actually quite short of the maximum.  That means I can safely add another row of 3. Even with the failure we're all learning good info about the Matrix Portal S3's capability thanks to identifystation's project.  We're in uncharted territory here as far as I know. While it's very exciting to see someone go for the maximum the error puts a big wrench in their project plans in the short term. We've discussed getting around it by running 2 Matrix Portal S3's and using MQTT to sync each half.  If there is no signal booster it might be the only way to do it anyway.

dhalbert, 2024-01-27:
Since RGBMatrix is "write-only" we can try virtual large numbers of panels without actually connecting up any panels.

jepler, 2024-01-27:
Looks like *cough* someone *cough* assumed the size of the rgbmatrix framebuffer would fit in 16 bits: `uint16_t bufsize, width;`

dhalbert, 2024-01-27:
@identifystation Please try this build (unzip to get the UF2):
https://github.com/adafruit/circuitpython/actions/runs/7679441384/artifacts/1199728584

@jepler changed the buffer size variable to be 32 bits from 16 bits in #8847. I tested with the program in the first post and it does not give an error.

dhalbert, 2024-01-27:
Fixed by #8847.

PR #8847 (2024-01-27): Allow the framebuffer size of an rgbmatrix to be >65535 bytes
Closes #8846 if my analysis is correct
closed 2024-01-27 by dhalbert