adafruit/circuitpython #8850: dict.keys() is empty in subclassed dict

miguelgrinberg, 2024-01-27:
### CircuitPython version

```python
MicroPython 9.0.0-alpha.6-49-gb704452f70 on 2024-01-27; linux [GCC 11.4.0] version
```

### Code/REPL

```python
class MyDict(dict):
    def __init__(self, initial_dict=None):
        super().__init__(initial_dict)

d = dict({'foo': 'bar', 'a': 123})
print(d.keys())  # keys is correct
d = MyDict({'foo': 'bar', 'a': 123})
print(d.keys())  # keys is empty
```

### Behavior

When running the above example, the base dictionary is not initialized correctly.

Output using CircuitPython:

```
{'a': 123, 'foo': 'bar'} dict_keys(['a', 'foo'])
{'a': 123, 'foo': 'bar'} dict_keys([])
```

Output using MicroPython:

```
{'a': 123, 'foo': 'bar'} dict_keys(['a', 'foo'])
{'a': 123, 'foo': 'bar'} dict_keys(['a', 'foo'])
```

Output using CPython:

```
{'foo': 'bar', 'a': 123} dict_keys(['foo', 'a'])
{'foo': 'bar', 'a': 123} dict_keys(['foo', 'a'])
```

### Description

I think this might be in part related to #8758, but unfortunately that fix does not help.

dhalbert, 2024-01-27:
I made #8851, which I think fixes this.

miguelgrinberg, 2024-01-27:
Thanks! I can confirm that your fix addresses the problem, and I do not have any more failures in my [microdot](https://github.coim/miguelgrinberg/microdot) test suite related to my `dict` subclass. :tada:

dhalbert, 2024-01-27:
Great! Thanks for testing so thoroughly.

PR #8851 (2024-01-31): py/objdict.c: fix keys() for dict subclasses
- Fixes #8850.

I think that there are still other things about subclassing dict that will not always work, such as trying to override other operations. However, this fixes a clear bug.
closed 2024-01-31 by dhalbert