adafruit/circuitpython #8856: Pico ceases to function after adding circuit python binary

IlogicalKarma, 2024-01-28:
### CircuitPython version

```python
Using adafruit-circuitpython-raspberry_pi_pico-en_US-9.0.0-beta.0 and adafruit-circuitpython-raspberry_pi_pico-en_US-8.2.9:
```

### Code/REPL

```python
PS D:\> ls

    Directory: D:\

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----        01.01.2020     00:00                .fseventsd
d-----        01.01.2020     00:00                lib
-a----        01.01.2020     00:00              0 .metadata_never_index
-a----        01.01.2020     00:00              0 .Trashes
-a----        01.01.2020     00:00              0 settings.toml
-a----        01.01.2020     00:00             22 code.py
-a----        01.01.2020     00:00            125 boot_out.txt
```

### Behavior

The device is reported as corrupted by Windows and the files are inaccessible on both operating systems. I have attempted this on my Linux and Windows computers, with both versions(8.2.9 and 9.0.0) on both a pico and a pico W. After a couple of seconds the pico disconnects. Prior to this it produces a few sequences of flashes.

dhalbert, 2024-01-28:
If you have Samsung Disk Magician or some other utility programs installed on Windows, it can cause disconnect-behavior. Or your CIRCUITPY drive may be corrupted, though that usually does not cause disconnects. Try erasing all of flash with the "nuke" UF2, as described here: https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython/circuitpython#flash-resetting-uf2-3083182.

IlogicalKarma, 2024-01-28:
Nuking did not solve the issue.

dhalbert, 2024-01-28:
could you ask in discord? https://adafru.it/discord, channel #help-with-circuitpython. This may require some interactive debugging. There is something unusual about your setup or your board(s).

dhalbert, 2024-03-08:
A fix has been made to make CircuitPython 9 more compatible with Samsung Disk Magician. Closing this for now, but please reopen if you are still having problems.

closed 2024-03-08 by dhalbert