adafruit/circuitpython #8857: ESP32-S3 requests download causes watchdog hard reset

artBoffin, 2024-01-29:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; ESP32-S3-DevKitC-1-N8 with ESP32S3
```

### Code/REPL

```python
def wget(url, local_filename, chunk_size=1024):
    try:
        with requests.get(url, stream=True) as response:
            with open(local_filename, 'wb') as f:
                for chunk in response.iter_content(chunk_size=chunk_size):
                    f.write(chunk)
                    print(".", end="")
    except Exception as e:
        print(f"! error {str(e)}")
    print("\n> wget done")
```

### Behavior

You are in safe mode because: Internal watchdog timer expired.

### Description

Downloads for a while, resets, gives me the watchdog error.
So, seems to partly work, then hard-crash.
Sometimes, it will fully erase itself to the "Hello World" state

tannewt, 2024-01-29:
Please also try the latest 9.0 pre-release to see if the issue has already been fixed.

dhalbert, 2024-03-08:
This: https://www.seeedstudio.com/XIAO-ESP32S3-p-5627.html says that it's an N8R8 board. The instructions appear to use an N8 build. That shouldn't make a difference, but try the N8R8 version, and try 9.0.0-rc.0 or later.

dhalbert, 2025-07-29:
@artBoffin hi - did you ever resolve this?