adafruit/circuitpython #8858: CircuitPython core code crashed hard. Whoops!

rpcoe, 2024-01-29:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.9 on 2023-12-06; Raspberry Pi Pico W with rp2040
Board ID:raspberry_pi_pico_w
UID:E6616407E353582C
MAC:28:CD:C1:07:71:81
```

### Code/REPL

```python
import time
import wifi

wifi.Radio.enabled: True
radio = wifi.radio
radio.stop_station()
#setting up ssid, password, channel and authmode
ssid = 'RPPW'
password = 'RPPW1234'
channel = 1
authmode = (wifi.AuthMode.WPA2, wifi.AuthMode.PSK)

print('activating access point')
radio.start_ap(ssid, password, channel=channel, authmode=authmode)

time.sleep(1000)
```

### Behavior

Runs ok and Access Point starts.  I can log in via wifi.  When program ends, I get the message:
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Heap allocation when VM not running.

anecdata, 2024-01-29:
A number of changes have been made to wifi AP code since 8.2.9. Can you test with `9.0.0-beta.0`?

rpcoe, 2024-01-30:
I installed 9.0.0-beta.0
It is better, BUT:  On a hard reboot - unplug 5 volt - seems to run OK.  If
I restart the code, I get runtimeError: AP could not be started line 24.
If I comment out this line, it runs OK on subsequent runs.  I tried
stopping the AP at the end of the program, but this didn't change anything.

Here is the current code:
# I had a lot of trouble coding the access point.
# The code runs fine, but when it stops, it crashes the PicoW
#     AND SOMETIMES WIPES OUT THE PROGRAM !!!
# I found this code which has simiiar problems to mine.
#   CircuitPython core code crashed hard. Whoops!
#   Heap allocation when VM not running.
#   I filed a bug report #8858
#  anecdata responded very quickly and advised installing the latest beta
version.  Some inprovemeent

import time
import wifi

wifi.Radio.enabled: True
radio = wifi.radio
radio.stop_station()
#setting up ssid, password, channel and authmode
ssid = 'MARINE1'
password = '2150Wins'
channel = 1
authmode = (wifi.AuthMode.WPA2, wifi.AuthMode.PSK)

print('activating access point')
print("AP mode active = ", radio.ap_active)
radio.start_ap(ssid, password, channel=channel, authmode=authmode)

i = 0
while i < 15:
    time.sleep(1)
    print("AP running", i, "seconds", "AP mode active = ", radio.ap_active)
    i += 1
radio.stop_ap()

= > ÷
Rob Coe
***@***.***

anecdata, 2024-01-30:
There is still an issue ( #8718 ) with stopping and re-starting an AP. For now, `RuntimeError: AP could not be started` will arise if start_ap() is called twice in a row, even across reloads. A reset is needed to fully stop the AP.

I haven't seen the crashes in testing `9.0.0-beta.0`. It might not hurt to nuke the flash and re-install UF2 and code from scratch.

eightycc, 2024-01-30:
Build artifacts for `9.0.0-beta.0` with fix for #8718 are here: https://github.com/adafruit/circuitpython/actions/runs/7712683561/artifacts/1205824119.

rpcoe, 2024-01-30:
Thanks for following up.  I tried  9.0.0-beta.0-5-ged345f4ea2  , but the
same result.  If I add a microcontroller.reset() to the end of the program,
everything restarts and works just like a power reboot.

Without the reset, the code only runs correctly once.
= > ÷
Rob Coe
***@***.***

eightycc, 2024-01-30:
@rpcoe There's a submodule versioning problem with `cyw43-driver` that I'm looking into.

eightycc, 2024-01-30:
@rpcoe This build artifact picks up the correct `cyw43-driver` submodule branch: https://github.com/adafruit/circuitpython/actions/runs/7715586486/artifacts/1206479517.

rpcoe, 2024-01-30:
This looks like it works!!
Thank you eightycc and anecdata.
= > ÷
Rob Coe
***@***.***

anecdata, 2024-01-31:
Closing in light of testing after driver update. If similar behavior arises, feel free to re-open or create a new issue.

closed 2024-01-31 by anecdata