adafruit/circuitpython #8872: Circuitpython 9 RuntimeError: Mount point directory missing on RP2040

SalamCytron, 2024-02-02:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import board
import busio
import sdcardio
import storage

spi = busio.SPI(clock=board.GP18, MOSI=board.GP19, MISO=board.GP16)
cs = board.GP17

sd = sdcardio.SDCard(spi, cs)

vfs = storage.VfsFat(sd)
storage.mount(vfs, '/sd')

with open("/sd/pico.txt", "w") as file:
    file.write("1. Hello, world!\r\n")
```

### Behavior

```
Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
RuntimeError: Mount point directory missing
```

### Description

The error only occurs on CircuitPython 9.0.0-beta.0 , on CircuitPython 8.2.9 works perfectly fine.

RetiredWizard, 2024-02-02:
Starting with CP 9.x, a folder has to be created to be used as a mount point before the SD card can be mounted. The quick fix is to add
```py
import os

if not '/sd' in os.listdir('/'):
    os.mkdir('/sd')
```
sometime before the storage.mount command.

Although I believe PR #8860 will resolve this by creating the mount point file for most boards (they need to have the sdcardio or sdioio modules enabled on the build, which I believe is the default).

tannewt, 2024-02-02:
> Although I believe PR #8860 will resolve this by creating the mount point file for most boards (they need to have the sdcardio or sdioio modules enabled on the build, which I believe is the default).

This will only apply to new filesystems. It likely won't apply to most folks.

closed 2024-02-02 by tannewt

rsbohn, 2024-02-02:
You need write access to CIRCUITPY '/' to create the '/sd' folder. Maybe print instructions for doing that?

RetiredWizard, 2024-02-02:
> You need write access to CIRCUITPY '/' to create the '/sd' folder. Maybe print instructions for doing that?

I suspect most folks would create the folder through the CIRCUITPY drive, so it's probably not worth adding instruction text on changing the storage.remount flag in the CircuitPython core.

I personally liked the idea of adding the os.listdir check from the snippet above because it doesn't hide any error messages like "read-only file system" as a Try/Except block would. Getting the read-only error message should be enough of a lead to track down the solution especially since write locking the CIRCUITPY drive has been a long standing feature of CirucitPython.

SalamCytron, 2024-02-05:
By adding the 'sd' directory to the CIRCUITPY drive manually or through the code, solve the problem.
Thank you.

phantomsixthplayer, 2024-04-27:
Thanks! I was using the SD card on the grand central.

Adafruit CircuitPython 9.0.4 on 2024-04-16; Adafruit Grand Central M4 Express with samd51p20

RuntimeError: Mount point directory missing
```
spi = busio.SPI(board.SD_SCK, board.SD_MOSI, board.SD_MISO)
sdcard = sdcardio.SDCard(spi, board.SD_CS)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
```