adafruit/circuitpython #8874: Deque is not iterable

berkut0, 2024-02-02:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.6 on 2023-12-12; Waveshare RP2040-Zero with rp2040
```

### Code/REPL

```python
from collections import deque

sa = deque((), 4)

list(sa)
```

### Behavior

It's pretty self-explanatory:
[image]

tannewt, 2024-02-02:
Does this work in MicroPython?

berkut0, 2024-02-02:
MicroPython v1.22.1 on 2024-01-05
doesn't work either.

tannewt, 2024-02-05:
I'd filing an issue with MicroPython if one hasn't been filed already. We're trying to get our copy of the VM to match MP.

SAK917, 2024-02-06:
I have been working on an iterable version of the MP/CP deque for my own application and in researching this, it looks like the lack of iteration has come up numerous times over the years.  I believe this Issue in the MP repo covers it the best:  https://github.com/micropython/micropython/issues/4776.  

Long story short(er), the MP folks seem to have punted on a performant native C implementation of deque that is iterable and are suggesting folks that need/want deque iteration can use an interim Python implementation of deque that is more feature rich but less performant (and must be manually imported) until a long-term solution is implemented, possibly as a native module?  There seems to be some concerns about a C version of deque that is closer in feature parity to the Python version, but requires more space in the build.

@tannewt, if I am successful in creating an iterable C implementation of deque (based on the existing MP deque code), I was originally planning on offering it for incorporation into CP.  In light of your statement above that you are trying to get the CP copy of the VM to match MP, would your preference be for me to try getting it added to MP instead?

tannewt, 2024-02-06:
> @tannewt, if I am successful in creating an iterable C implementation of deque (based on the existing MP deque code), I was originally planning on offering it for incorporation into CP. In light of [...]

Please make a PR to MP with it first. If they don't respond quickly, then we can do it to CP directly. We can also merge it into CP after MP merges it but before it is released. That way the code will still be the same.

dhalbert, 2024-03-18:
Thank you. We'll pick this up on the next merge from upstream MicroPython.

PR #9065 (2024-03-26): Expand objdeque implementation
Fixes #8874 
This PR expands CP collections.deque functionality to be more congruent with the CPython implementation by adding iteration, subscription and initialization as well as pop(), appendleft() and extend().  Tests for the new functionality are also provided.

This code is a copy of [MP PR #10724](https://github.com/micropython/micropython/pull/10724) that was coded/submitted by LordFlashmeow and recently merged into MP.  I have built and tested versions of CP with this code for RP2040, ESP32S2 and atmel-samd M4 boards and everything compiles and works as  [... cut]
closed 2024-03-26 by dhalbert