adafruit/circuitpython #8875: Unable to get into web workflow with MatrixPortal S3

BlitzCityDIY, 2024-02-02:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; Adafruit MatrixPortal S3 with ESP32S3
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2023 Liz Clark for Adafruit Industries
#
# SPDX-License-Identifier: MIT

import os
import gc
import ssl
import time
import wifi
import socketpool
import adafruit_requests
import adafruit_display_text.label
import board
import terminalio
import displayio
import framebufferio
import rgbmatrix
import adafruit_json_stream as json_stream
import microcontroller
from adafruit_ticks import ticks_ms, ticks_add, ticks_diff
from adafruit_datetime import datetime, timedelta
import neopixel

displayio.release_displays()

# font color for text on matrix
font_color = 0xFFFFFF
# your timezone UTC offset and timezone name
timezone_info = [-4, "EDT"]
# the name of the sports you want to follow
sport_name = ["football", "baseball", "soccer", "hockey", "basketball"]
# the name of the corresponding leages you want to follow
sport_league = ["nfl", "mlb", "usa.1", "nhl", "nba"]
# the team names you want to follow
# must match the order of sport/league arrays
# include full name and then abbreviation (usually city/region)
team0 = ["New England Patriots", "NE"]
team1 = ["Boston Red Sox", "BOS"]
team2 = ["New England Revolution", "NE"]
team3 = ["Boston Bruins", "BOS"]
team4 = ["Boston Celtics", "BOS"]
# how often the API should be fetched
fetch_timer = 300 # seconds
# how often the display should update
display_timer = 30 # seconds

pixel = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness = 0.3, auto_write=True)

# matrix setup
base_width = 64
base_height = 32
chain_across = 2
tile_down = 2
DISPLAY_WIDTH = base_width * chain_across
DISPLAY_HEIGHT = base_height * tile_down
matrix = rgbmatrix.RGBMatrix(
    width=DISPLAY_WIDTH, height=DISPLAY_HEIGHT, bit_depth=3,
    rgb_pins=[
        board.MTX_R1,
        board.MTX_G1,
        board.MTX_B1,
[... 345 lines cut ...]
        microcontroller.reset()
```

### Behavior

I have not been able to get into web workflow on my MatrixPortal S3 with the 9.0 beta0. I've been able to consistently recreate the issue following these steps:

- Close Chrome completely out
- Power cycle or reset the MatrixPortal
- Wait to see that the MatrixPortal is connected to my SSID
- Open Chrome and go to code.circuitpython.org
- Prompted to log into web workflow
- The Chrome tab shows my board's code editor address, Blinka spins, dev window shows this:

```
Load different workflow
index.js:37 Initializing File Transfer Client...
```

- The MatrixPortal seems to be stuck in fetching the JSON requests (NeoPixel is blue) and the display is showing the 2nd out of 5 logos.
- The chrome tab finally shows this dialog box:

`Unable to connect. The device cpy-2048dc.local was not found. Be sure it is plugged in and set up properly.`

When I close out of the tab, the board seems to resume code.py and runs fine. I closed out Chrome and tried to connect thinking that maybe I needed to wait for the JSON requests to stop but the same events occurred.

RetiredWizard, 2024-02-02:
This sounds like it could be related to #8690

VPTechOps, 2024-02-03:
I have the same issue, but some additional info:
1) If I enter the IP address only into the browser, the main page for workflow loads correctly
2) If I click on the link for "file browser", the file browser page loads, but the following message appears near the top of the page:
ℹ️ USB is using the storage. Only allowing reads. Try ejecting the CIRCUITPY drive. See [Getting Started with Web Workflow](https://learn.adafruit.com/3654/device-setup#disabling-usb-mass-storage-3125964) for details.
But, the MatrixPortal S3 is not connected to the PC. USB port is connected to power adapter only.
3) If I click on the link for "full code editor" on the main page, I have the same results as previously described for code.circuitpython.org, except the connect already knows that WiFi is the type and IP and doesn't present a connection menu.

tannewt, 2024-02-13:
I believe this is due to API changes that code.circuitpython.org doesn't know how to deal with. Specifically, the directory listing response was changed to include mount status and free space. Assigning to @makermelissa for when she has time.

tannewt, 2024-02-13:
Version info here: https://docs.circuitpython.org/en/latest/docs/workflows.html#versions

makermelissa, 2024-02-20:
Yeah, it looks like the directory listing response was the main issue. In order to trigger the password prompt, a listing is requested on connection, which is why it wouldn't connect.

closed 2024-02-21 by makermelissa

VPTechOps, 2024-02-23:
Thanks, Melissa. The /code/ interface seems to be working, but the /fs/ entry still shows "ℹ️ USB is using the storage. Only allowing reads. Try ejecting the CIRCUITPY drive. See [Getting Started with Web Workflow](https://learn.adafruit.com/3654/device-setup#disabling-usb-mass-storage-3125964) for details." even though the device is not connected through USB. I assume this is a related issue.

makermelissa, 2024-02-23:
@VPTechOps the message may a bit misleading. Essentially, you can either have a writeable file system via USB or with web workflow, but not both at the same time. You'll want to disable Mass Storage to make it writeable. See https://learn.adafruit.com/getting-started-with-web-workflow-using-the-code-editor/device-setup#disabling-usb-mass-storage-3125964 for more info.

VPTechOps, 2024-02-24:
The message just started to appear on CP9. I have been using web workflow for quite some time using the same procedures. This is new, and as I pointed out, the USB on the device is not connected to anything other than a power supply. The circuitpy drive does not appear because there is no USB connection. Am I missing something?

tannewt, 2024-02-27:
> @makermelissa, the message just started to appear on CP9. I have been using web workflow for quite some time using the same procedures on CP8. This is new, and as I pointed out, the USB on the device is not connected to anything other than a power supply. The circuitpy drive does not appear because there is no USB connection. Am I missing something?

Please file a new issue and let us know what board you are using.