adafruit/circuitpython #8876: MatrixPortal S3 demo fails under CP9.0 beta.0 -- works under alpha.6

jerryneedell, 2024-02-02:
### CircuitPython version

```python
CP 9.0 beta.0

The code runs OK under CP 9.0 alpha.6
```

### Code/REPL

```python
I am trying to run the MartixPortal_S3 SMS Messge Board code https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/main/Matrix_Portal_S3_SMS_Scroller
```

### Behavior

```
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; Adafruit MatrixPortal S3 with ESP32S3
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Updating data from Adafruit IO
Traceback (most recent call last):
  File "code.py", line 106, in <module>
  File "code.py", line 56, in update_data
  File "/lib/messageboard/message.py", line 106, in add_text
  File "/lib/messageboard/message.py", line 57, in _add_bitmap
  File "/lib/messageboard/message.py", line 46, in _enlarge_buffer
ValueError: x1 must be 0--1

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

DJDevon3, 2024-02-03:
Might have something to do with the helper library having a [65535 buffer limit hardcoded](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/3a28acd2c690eb2277e9c85e0cc7ce2abebdd0cb/Matrix_Portal_S3_SMS_Scroller/lib/messageboard/message.py#L41C13-L41C72) which was recently raised to 32bits?

jepler, 2024-02-03:
This has to do with the changes to validation of arguments to bitmaptools.blit.

The bitmap is created with a size of 0x0 pixels initially:
```
        self._buffer = displayio.Bitmap(0, 0, 65535)
```
this is rather dubious,but in the past it worked.

Now, it appears the changed ("improved") argument validation rejects this, because the pixel (0,0) is not actually in the image. There are no valid X coordinates in the image, because the valid coordinates are from 0 to (width - 1), or from 0 to negative one.

Suggested change:
 * creation of a 0x0 bitmap is odd and maybe should be rejected (possible change to core)
 * messageboard library should initially create a 1x1 bitmap, or otherwise change how it handles the problem of initial sizing of the bitmap

This is all off the top of my head without testing, so take it with a grain of salt.

jerryneedell, 2024-02-03:
I tried simply changing
```
        self._buffer = displayio.Bitmap(0, 0, 65535)
```
to
```
        self._buffer = displayio.Bitmap(1, 1, 65535)
```  in the two places I found  it in message.py . It still works with alpha.6 but now fails in the display_text library under beta.0
It does  get further along and displays a message briefly before failing so this was  progress

```
Traceback (most recent call last):
  File "code.py", line 142, in <module>
  File "/lib/messageboard/message.py", line 91, in add_text
  File "/lib/adafruit_display_text/bitmap_label.py", line 118, in __init__
  File "/lib/adafruit_display_text/bitmap_label.py", line 214, in _reset_text
  File "/lib/adafruit_display_text/bitmap_label.py", line 455, in _place_text
  File "/lib/adafruit_display_text/bitmap_label.py", line 492, in _blit
ValueError: x1 must be 0--1

```

FoamyGuy, 2024-02-03:
From the conversation here it looks like we may have landed on deciding that 0 width or height blit should be allowed: https://github.com/adafruit/circuitpython/pull/7867

although it doesn't state specifically 0 height _and_ width it would follow that both should work if either is allowed IMO. Nor does it say that a Bitmap with 0 width or height is allowable, just the blit action. But perhaps we should aim for consistency. If blit allows 0 sized actions because it makes certain image processing algorithms easier perhaps the same is true of the 0 sized Bitmaps themselves. 

That PR was finished and merged, I'm not sure if its change got reverted in the meantime or if there was some other instance of the 0 check that wasn't updated at that time. 

I'm a little unsure why this issue is popping up between alpha.6 and beta.0 though, I had a look back through recent PRs and didn't see any related changes, but perhaps I overlooked something.

jepler, 2024-02-03:
I'm gone for a week, so I recommend that if someone decides how to fix this in the meantime that they add tests. Creation of bitmaps and use of bitmaptools (&bitmapfilter) functions can be tested at build time by adding files under `tests/circuitpython`: a `.py` file and a `.py.exp` file containing the expected output.

I do somewhat understand the purpose of allowing the creation of a 0x0 bitmap, and I can see that it's probably then also necessary to allow to *blit* from a 0x0 bitmap, which is no longer possible after I factored out blit coordinate range checking into the function `bitmaptools_validate_coord_range`.

The good news is that with this code centralized and shared among several functions we can make sure they're at least consistent with each other.

Otherwise if nobody else wants to grab it I'll look at this the week of feb 12.

jerryneedell, 2024-02-05:
This appears to be addressed by PR #8880.
Leaving open pending review and acceptance of the PR.

jerryneedell, 2024-02-06:
fixed by #8880

closed 2024-02-06 by jerryneedell