adafruit/circuitpython #8877: ValueError: incompatible .mpy file message on Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM. Works on 8.2.9

VPTechOps, 2024-02-03:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
Board ID:adafruit_feather_esp32s3_4mbflash_2mbpsram
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2023 Frederick M Meyer
#
# SPDX-License-Identifier: MIT

import ipaddress
import ssl
import wifi
import socketpool
import adafruit_requests
import random
import os
import time
import rtc
import adafruit_ntp
import adafruit_datetime
import displayio
import framebufferio
import rgbmatrix
import board
import digitalio
import terminalio
import adafruit_display_text as adt
import adafruit_display_text.label as adtl
```

### Behavior

8.2.9 works correctly using ,mpy files. 9.0.0-beta.0 gives this error:
[2K[0GAuto-reload is on. Simply save files over USB to run them or enter REPL to disable.
[2K[0Gcode.py output:
Traceback (most recent call last):
  File "code.py", line 9, in <module>
ValueError: incompatible .mpy file

Code done running.

### Description

The *.mpy files work correctly on MatrixPortal S3, but not on any Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM

### Additional information

This happens on all 4 imports for adafruit_* files. Deleting the files and installing the *.py versions then works correctly.

anecdata, 2024-02-03:
Are you using the 8.x `.mpy` bundle for running with 8.2.9, and the 9.x `.mpy` bundle for running with 9.0.0-beta.0?

VPTechOps, 2024-02-03:
I used circup to check the libs. I saw it download both 8. and 9. bundles. I'll do it manually and see if that makes a difference.

C:\Users\fred>circup update --all
Found device at D:\, running CircuitPython 9.0.0-beta.0.
None of the modules found on the device need an update.

closed 2024-02-03 by VPTechOps