adafruit/circuitpython #8883: espcamera IDFError with CP 9.0beta.0  -- works OK with CP 8.2.9

jerryneedell, 2024-02-05:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; ESP32-S3-EYE with ESP32S3
```

### Code/REPL

```python
import board
import espcamera

cam = espcamera.Camera(
    data_pins=board.CAMERA_DATA,
    external_clock_pin=board.CAMERA_XCLK,
    pixel_clock_pin=board.CAMERA_PCLK,
    vsync_pin=board.CAMERA_VSYNC,
    href_pin=board.CAMERA_HREF,
    pixel_format=espcamera.PixelFormat.JPEG,
    frame_size=espcamera.FrameSize.SVGA,
    i2c=board.I2C(),
    external_clock_frequency=20_000_000,
    grab_mode=espcamera.GrabMode.WHEN_EMPTY)
cam.vflip = True
```

### Behavior

Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; ESP32-S3-EYE with ESP32S3

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 15, in <module>
espidf.IDFError: Operation or feature not supported

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

I am trying to run the AIO webcam example from https://learn.adafruit.com/capturing-camera-images-with-circuitpython/example-webcam-with-adafruit-io  but I have stripped open the code to produce this error when trying to initialized the espcamera.

The code runs OK on CP 8.2.9 (both the full code and the stripped down test)

bill88t, 2024-02-07:
I have never used `WHEN_EMPTY`, but for me on `m5stack_timer_camera_x` everything works fine.
I will try that parameter when I can.

jerryneedell, 2024-02-08:
I also tried switching to GrabMode=LATEST  but it fails with the same IDFError....

dhalbert, 2024-02-09:
I am testing this on a MEMENTO board, and getting a slightly different error: `espidf.IDFError: Requested resource not found` which may be from a rework of error messages. Oddly, when I use the `adafruit_pycamera` library, it works, but the set-up appears similar. The library does a power-cycle and reset of the camera, but adding that to the test code above does not fix the problem. Still testing.

bill88t, 2024-02-09:
`espidf.IDFError: Requested resource not found` is incorrect configuration. I think it's the i2c check that raises that.
Prolly an incorrect pin or something.

For me:
```py
espcamera.Camera(
    data_pins=board.D,
    pixel_clock_pin=board.PCLK,
    vsync_pin=board.VSYNC,
    href_pin=board.HREF,
    i2c=board.SSCB_I2C(),
    external_clock_pin=board.XCLK,
    external_clock_frequency=20_000_000,
    powerdown_pin=None,
    reset_pin=board.RESET,
    pixel_format=vr("px"),
    frame_size=vr("fr"),
    jpeg_quality=vr("qual"),
    framebuffer_count=1,
    grab_mode=espcamera.GrabMode.LATEST,
```
works with `m5stack_timer_camera_x` and varying parameters in the `vr()` fields. It's `ov3660`.
Also last I checked my `firebeetle2_esp32s3`'s camera also worked.

dhalbert, 2024-02-09:
@jerryneedell If I power-cycle and reset the camera (I had a mistake in my testing script at first), first, then your example works on the MEMENTO:
```py
import board
from digitalio import DigitalInOut
import time
import espcamera

cam_reset = DigitalInOut(board.CAMERA_RESET)
cam_pwdn = DigitalInOut(board.CAMERA_PWDN)
cam_reset.switch_to_output(False)
cam_pwdn.switch_to_output(True)
time.sleep(0.01)
cam_pwdn.switch_to_output(False)
time.sleep(0.01)
cam_reset.switch_to_output(True)
time.sleep(0.01)

cam = espcamera.Camera(
    data_pins=board.CAMERA_DATA,
    external_clock_pin=board.CAMERA_XCLK,
    pixel_clock_pin=board.CAMERA_PCLK,
    vsync_pin=board.CAMERA_VSYNC,
    href_pin=board.CAMERA_HREF,
    pixel_format=espcamera.PixelFormat.JPEG,
    frame_size=espcamera.FrameSize.SVGA,
    i2c=board.I2C(),
    external_clock_frequency=20_000_000,
    grab_mode=espcamera.GrabMode.WHEN_EMPTY
)
```

jerryneedell, 2024-02-09:
I’ll retry on the Eye tomorrow but I have not had any issues with CP9 on my MEMENTO

dhalbert, 2024-02-09:
I do not have an Eye to try but I do have a camera breakout.

jerryneedell, 2024-02-10:
No change on the eye 
```

Adafruit CircuitPython 9.0.0-beta.0-22-g58b1ce46f1 on 2024-02-09; ESP32-S3-EYE with ESP32S3
>>> import code_test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "code_test.py", line 15, in <module>
espidf.IDFError: Operation or feature not supported
>>> 
```

jerryneedell, 2024-02-10:
> I do not have an Eye to try but I do have a camera breakout.

I also have some breakouts, Let me know what MCU and breakout you use if you can get it to work and I will try to reproduce.

jerryneedell, 2024-02-10:
hmmm -- I tried your code on the EYE
```
10.0.0.16 | REPL | 9.0.0-beta.0-22-g58b1ce46f1 reload.
Adafruit CircuitPython 9.0.0-beta.0-22-g58b1ce46f1 on 2024-02-09; ESP32-S3-EYE with ESP32S3
>>> 
>>> import code_test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "code_test.py", line 6, in <module>
AttributeError: 'module' object has no attribute 'CAMERA_RESET'
>>> 
```

jerryneedell, 2024-02-11:
Just a brief update.  I created a custom build of the adafruit_feather_esp32s3_4mbflash_2mbpsram  to add esp camera. I used the same mpconfigboard.mk settings as the memento 
```
USB_VID = 0x239A
USB_PID = 0x811C
USB_PRODUCT = "Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM"
USB_MANUFACTURER = "Adafruit"

IDF_TARGET = esp32s3

CIRCUITPY_ESP_FLASH_SIZE = 4MB
CIRCUITPY_ESP_FLASH_MODE = qio
CIRCUITPY_ESP_FLASH_FREQ = 80m

CIRCUITPY_ESP_PSRAM_SIZE = 2MB
CIRCUITPY_ESP_PSRAM_MODE = qio
CIRCUITPY_ESP_PSRAM_FREQ = 80m

CIRCUITPY_AUDIOBUSIO = 0
CIRCUITPY_CANIO = 0
CIRCUITPY_ESPCAMERA = 1
CIRCUITPY_FRAMEBUFFERIO = 0
CIRCUITPY_KEYPAD = 0
CIRCUITPY_ONEWIREIO = 0
CIRCUITPY_PARALLELDISPLAYBUS = 0
CIRCUITPY_RGBMATRIX = 0
CIRCUITPY_ROTARYIO = 0

OPTIMIZATION_FLAGS = -Os
```
With this the sample code from @dhalbert works with all the pins adjusted ...
I still can't actually get code to really use the camper working, but that may be due to many other issues and this MCU is not really very well suited for use with the camera. Not a lot of pins to spare... The main point is that the IDFError does not occur wit this build as it does not for a MEMENTO with CP 9 so it appears that this issue is limited to the ESP32S3-EYE.

jerryneedell, 2024-02-11:
hmmm -- I just realized that the above test was don with an OV5640, but the EYE uses an OV2640 -- I should try swapping breakouts to an OV2640...

jerryneedell, 2024-02-11:
Bingo!!  The same code that works with the OV5640 now fails with an OV2640... the plot thickens ;-)

bill88t, 2024-02-11:
Uhh on `Firebeetle 2 ESP32-S3` I have an OV2640.

```py
>>> import board
>>> from AXP313A import *
>>> i2c = board.I2C()
>>> a = AXP313A(i2c)
>>> enable_camera(a, camera_voltages.OV2640)
True
>>> import espcamera
>>> b = espcamera.Camera(data_pins=board.CAM_DATA, pixel_clock_pin=board.CAM_PCLK, vsync_pin=board.CAM_VSYNC, href_pin=board.CAM_HREF, external_clock_pin=board.CAM_XCLK, external_clock_frequency=20_000_000,
... powerdown_pin=None, reset_pin=None, i2c=i2c, pixel_format=espcamera.PixelFormat.JPEG, frame_size=espcamera.FrameSize.VGA, jpeg_quality=11, framebuffer_count=1, grab_mode=espcamera.GrabMode.LATEST)
>>> b.sensor_name
'OV2640'
>>> b.take()
<memoryview>
>>> b.take()
<memoryview>
```

Please do keep in mind that different cameras require different voltages.

jerryneedell, 2024-02-11:
@bill88t is that with CP 9?

bill88t, 2024-02-11:
@jerryneedell Yea I am always running the latest commit.

Also yes this board has a pmic and handles voltages with that library.
This way this board supports quite a few sensors.

I do not have any camera breakouts, so idk either.

jerryneedell, 2024-02-11:
AHA!! -- OV2640 support is not enabled for the esp32s3_eye!
adding it
```
#
# Espressif IoT Development Framework Configuration
#
#
# Component config
#
#
# LWIP
#
CONFIG_LWIP_LOCAL_HOSTNAME="espressif-esp32s3-eye"
CONFIG_OV2640_SUPPORT=y
# end of LWIP

# end of Component config

# end of Espressif IoT Development Framework Configuration
```
and I can now use the camera!!! 
Now to figure out the AIO issues  adafruit_requests is running out of retries but the camera is happy!

dhalbert, 2024-02-11:
@jerryneedell would you like to submit a PR for this?

jerryneedell, 2024-02-11:
> @jerryneedell would you like to submit a PR for this? If the other camera is turned on by default, it could be turned off
What do yo mean by "other camera"?
It's been along tine since did a "core" PR, but it will be a good exercise! First I should to  try some other code to make sure the camera is fully functional. I'm still not sure why my example using AIO is not working on this board so I also want to delve into that.

jerryneedell, 2024-02-11:
Yay -- The LCD Viewfinder example https://learn.adafruit.com/capturing-camera-images-with-circuitpython/example-lcd-viewfinder  now works.
I try to get a PR in later today and work on the AIO issues separately.

dhalbert, 2024-02-11:
> What do yo mean by "other camera"?

I mean that if `CONFIG_OV2640_SUPPORT=y`, to do `CONFIG_OV5640_SUPPORT= `, since we don't need support for that camera.

ladyada, 2024-02-11:
the cameras are easily swappable, if we support camera modules, we should support 'all' the types

jerryneedell, 2024-02-11:
> > What do yo mean by "other camera"?
> 
> I mean that if `CONFIG_OV2640_SUPPORT=y`, to do `CONFIG_OV5640_SUPPORT= `, since we don't need support for that camera.

There was no support enabled at all in the current  sdkconfig file.

jerryneedell, 2024-02-11:
> the cameras are easily swappable, if we support camera modules, we should support 'all' the types

Do they need to be individually listed in sdkconfig?

dhalbert, 2024-02-11:
> the cameras are easily swappable, if we support camera modules, we should support 'all' the types

There are quite a few models supported: see https://github.com/adafruit/esp32-camera/blob/8f3f2cc8cfb2e4371870b0c5a974d40468114a9e/Kconfig. Each one has its own driver .c file which is of noticeable size. So maybe all the OVxxxx ones?

jerryneedell, 2024-02-11:
> There are quite a few models supported: see https://github.com/adafruit/esp32-camera/blob/8f3f2cc8cfb2e4371870b0c5a974d40468114a9e/Kconfig. Each one has its own driver .c file which is of noticeable [...]

OK -- I'll see how it does with flash if I enable more.  The only board this applies to is the esp32s3_eye, correct.  Earlier I tried setting up a camera on the feather_esp32s3_4mbflash_2mpsram  board and had to take out a lot of modules to get it to fit.  I think the esp32s3_eye is the only board the has camera support except for the memento, but that only support the OV5640, correct?

jerryneedell, 2024-02-11:
but... I am confused....
the sdkconfig for the memento is
```
#
# Espressif IoT Development Framework Configuration
#
#
# Component config
#
#
# LWIP
#
CONFIG_LWIP_LOCAL_HOSTNAME="espressif-esp32s3"
# end of LWIP

#
# Camera configuration
#
# CONFIG_OV7725_SUPPORT is not set
# CONFIG_OV3660_SUPPORT is not set
# end of Camera configuration

# end of Component config

# end of Espressif IoT Development Framework Configuration
```  
so no specific support is enabled

Does it default to OV5640?

jerryneedell, 2024-02-11:
> the cameras are easily swappable, if we support camera modules, we should support 'all' the types

Are there other modules available for the esp32s3_eye  board? the docs only refer to the OV2640 "sub" board https://github.com/espressif/esp-who/blob/master/docs/en/get-started/ESP32-S3-EYE_Getting_Started_Guide.md

dhalbert, 2024-02-11:
This is really confusing. In https://github.com/adafruit/esp32-camera/blob/8f3f2cc8cfb2e4371870b0c5a974d40468114a9e/Kconfig, most of the cameras have `default y` entries. If I build `espressif_esp32_eye` without your change, in `build-espressif_esp32_eye/esp-idf/sdkconfig` (the computed sdkconfig made during the build) many are turned on:
```#
# Camera configuration
#
CONFIG_OV7670_SUPPORT=y
CONFIG_OV7725_SUPPORT=y
CONFIG_NT99141_SUPPORT=y
CONFIG_OV2640_SUPPORT=y
CONFIG_OV3660_SUPPORT=y
CONFIG_OV5640_SUPPORT=y
CONFIG_GC2145_SUPPORT=y
CONFIG_GC032A_SUPPORT=y
CONFIG_GC0308_SUPPORT=y
CONFIG_BF3005_SUPPORT=y
CONFIG_BF20A6_SUPPORT=y
# CONFIG_SC101IOT_SUPPORT is not set
CONFIG_SC030IOT_SUPPORT=y
# CONFIG_SC031GS_SUPPORT is not set
# CONFIG_SCCB_HARDWARE_I2C_PORT0 is not set
CONFIG_SCCB_HARDWARE_I2C_PORT1=y
CONFIG_SCCB_CLK_FREQ=100000
# CONFIG_GC_SENSOR_WINDOWING_MODE is not set
CONFIG_GC_SENSOR_SUBSAMPLE_MODE=y
CONFIG_CAMERA_TASK_STACK_SIZE=2048
CONFIG_CAMERA_CORE0=y
# CONFIG_CAMERA_CORE1 is not set
# CONFIG_CAMERA_NO_AFFINITY is not set
CONFIG_CAMERA_DMA_BUFFER_SIZE_MAX=32768
# end of Camera configuration
# end of Component config
```
But in `build-adafruit_esp32s3_camera/esp-idf/sdkconfig` we have:
```
#
# Camera configuration
#
# CONFIG_OV7670_SUPPORT is not set
# CONFIG_OV7725_SUPPORT is not set
# CONFIG_NT99141_SUPPORT is not set
# CONFIG_OV2640_SUPPORT is not set
# CONFIG_OV3660_SUPPORT is not set
CONFIG_OV5640_SUPPORT=y
# CONFIG_GC2145_SUPPORT is not set
# CONFIG_GC032A_SUPPORT is not set
# CONFIG_GC0308_SUPPORT is not set
# CONFIG_BF3005_SUPPORT is not set
# CONFIG_BF20A6_SUPPORT is not set
# CONFIG_SC101IOT_SUPPORT is not set
# CONFIG_SC030IOT_SUPPORT is not set
# CONFIG_SC031GS_SUPPORT is not set
# CONFIG_SCCB_HARDWARE_I2C_PORT0 is not set
CONFIG_SCCB_HARDWARE_I2C_PORT1=y
CONFIG_SCCB_CLK_FREQ=100000
CONFIG_CAMERA_TASK_STACK_SIZE=2048
CONFIG_CAMERA_CORE0=y
# CONFIG_CAMERA_CORE1 is not set
# CONFIG_CAMERA_NO_AFFINITY is not set
CONFIG_CAMERA_DMA_BUFFER_SIZE_MAX=32768
# CONFIG_CAMERA_CONVERTER_ENABLED is not set
# end of Camera configuration
# end of Component config
```
@jerryneedell but CONFIG_OV2640_SUPPORT is turned on in https://github.com/adafruit/circuitpython/blob/8af780c7fa0200b0077c8ebb0c275638affe7a9c/ports/espressif/boards/espressif_esp32_eye/sdkconfig#L12. Is your fork up to date? (But this has not been changed recently.)

jerryneedell, 2024-02-11:
You built esp32_eye  not esp32s3_eye

dhalbert, 2024-02-11:
ack :forehead_slap:

jerryneedell, 2024-02-11:
oops -- never mind.

dhalbert, 2024-02-11:
All right, I see. So if the camera is not replaceable in ESP32-S3 Eye, then I would say just enable OV2640 and be done with it, and there'll be more space.

jerryneedell, 2024-02-11:
and after enabling OV2640 , I see 
```
CONFIG_OV2640_SUPPORT=y
CONFIG_OV3660_SUPPORT=y
CONFIG_OV5640_SUPPORT=y
```
in the generated sdkconfig so I guess I'll just leave it that way unless you want ne to try to explicitly remove the others

PR #8901 (2024-02-11): add OV2640 support to sdkconfig for esp32s3_eye build
fixes #8883
closed 2024-02-11 by dhalbert