adafruit/circuitpython #8884: ESP32-S3 write() crashes when busio.UART  called with tx pin as IO15 (and others)

rowsail, 2024-02-05:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; ESP32-S3-DevKitC-1-N8R8 with ESP32S3
```

### Code/REPL

```python
# Crashes:
import board, busio
serial = busio.UART(board.IO15, board.IO18, baudrate=115200)
serial.write(b'1234567890')

# Works:
import board, busio
serial = busio.UART(board.IO15, board.IO18, baudrate=115200)
serial.write(b'1234567890')
```

### Behavior

```
>>> import board, busio
>>> serial = busio.UART(board.IO15, board.IO18, baudrate=115200)
>>> serial.write(b'1234567890')
**(System crashes)**
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
```
```
>>> import board, busio
>>> serial = busio.UART(board.IO16, board.IO18, baudrate=115200)
>>> serial.write(b'1234567890')
10
>>>
```

### Description

I need to use IO15 as the TX line for a UART but after instantiating the uart with the call to busio.UART(...) the call to write(...) causes a reboot.

Entering the following via the REPL (as well as within a program) demonstrates the issue.

>>> import board, busio
>>> serial = busio.UART(board.IO15, board.IO18, baudrate=115200)
>>> serial.write(b'1234567890')
(System crashes)
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

If the tx pin is IO16, it works fine.

>>> import board, busio
>>> serial = busio.UART(board.IO16, board.IO18, baudrate=115200)
>>> serial.write(b'1234567890')
10
>>>

anecdata, 2024-02-06:
I'm not able to reproduce this:
```py
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; ESP32-S3-DevKitC-1-N8R8 with ESP32S3
>>> import board, busio
>>> serial = busio.UART(board.IO15, board.IO18, baudrate=115200)
>>> serial.write(b'1234567890')
10
```

rowsail, 2024-02-06:
It's actually an S3 N16R8 - but I have the exact same issue with the build for the YD-ESP32-S3 N16R8.    This is what took me down the path that it might be a bug in busio common to all ESP32-S3 build variants.

dhalbert, 2024-02-06:
@rowsail what is the exact board you are using?

rowsail, 2024-02-06:
I have an ESP32-S3-WROOM-1 N16R8 mounted on a PCB with a direct connection to USB, and IO15 and IO18 connected to an RS485 transceiver (along with a couple of GPIO to control DE and RE).

rowsail, 2024-02-06:
(Please ignore the value field for the ESP32-S3, a N16R8 was placed.)

[image]

Along with an additional 100uF on the 3v3 rail close to the ESP32 power in.

anecdata, 2024-02-06:
Just for completeness, using GPIO15 does seem to operate fine too:

<details>
<summary>code</summary>

```py
import time, board, busio

serial1 = busio.UART(board.IO15, board.IO18, baudrate=921600)
serial2 = busio.UART(board.IO16, board.IO17, baudrate=921600)
while True:
    serial1.write(b'1234567890')
    print(f'serial 2 read: {serial2.read(10)}')
    serial2.write(b'1234567890')
    print(f'serial 1 read: {serial1.read(10)}')
    time.sleep(1)
```
</details>

<details>
<summary>output</summary>

```
serial 2 read: b'1234567890'
serial 1 read: b'1234567890'
serial 2 read: b'1234567890'
serial 1 read: b'1234567890'
# ...
```
</details>

rowsail, 2024-02-06:
Then I am stumped!  Thank you for your help on this.  Just to confirm, you are using an N16R8 device?

anecdata, 2024-02-06:
no, I only have N8R8 and N32R8... I could try the N32R8 flashed with N8R8, although that's a Wroom-2 and could be different in other ways

anecdata, 2024-02-06:
Same behavior (REPL and code, as above) on N32R8 flashed with N32R8 and with N8R8.

rowsail, 2024-02-06:
OK - Thank you again.  I'm going to close out this issue - there must be something else at play.  I can't believe there is something wrong with the N16 versions of the chip, surely?  Is there another way I can use a UART without busio?

dhalbert, 2024-02-06:
I guess I would check whether maybe IO15 is being pulled out of 0-3.3V range. If you just measure voltage there, what do you see? If you reverse TX and RX, does it crash? If you cut the IO15 connection to the transceiver, does it work?

You can try `bitbangio.UART`. But the fact that IO16 works but IO15 doesn't is really weird.

dhalbert, 2024-02-06:
Do you have anything in a `boot.py`?

rowsail, 2024-02-06:
I will check all those things. This is a fresh install of CP, so there shouldn't be anything untoward in boot.py - but I'll still check. I'll also load some earlier versions of Circuit Python to see if behavior is the same or not. Failing that, I'll look for a real Devkit with the S3 N16R8 on it and try that.

rowsail, 2024-02-07:
Does not crash on an N8R2 board.

```Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; ESP32-S3-DevKitC-1-N8R2 with ESP32S3
>>> import board, busio
>>> serial = busio.UART(board.IO15, board.IO18, baudrate=115200)
>>> serial.write(b'1234567890')
10
```

rowsail, 2024-02-07:
Everything works as it should using a different board (completely different design), but still using a S3 N16R8. 

Works OK:
Adafruit CircuitPython 8.2.9 on 2023-12-06; VCC-GND YD-ESP32-S3 (N16R8) with ESP32S3

Works OK:
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; VCC-GND YD-ESP32-S3 (N16R8) with ESP32S3

Removing the RS485 transceiver from the original board - so there is nothing externally connected to IO15, it still does not work - the only thing I can think of is that there is an internal fault with particular ESP32-S3 and when IO15 (and it happened on some others too) was driven low during the attempt at a transmit, it was pulling the supply rail down and causing a soft reboot.

Thank you for your help.

closed 2024-02-07 by rowsail

rowsail, 2024-02-07:
Confirmed - ESP32-S3 on board was damaged.