adafruit/circuitpython #8885: USB Host Basic Keyboard Test Configuration Error

FoamyGuy, 2024-02-05:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; Adafruit Feather RP2040 USB Host with rp2040
```

### Code/REPL

```python
https://github.com/adafruit/circuitpython/blob/main/tests/circuitpython-manual/usb/basic_keyboard.py

import array
import usb.core
import sys

# This is a WASD Code Keyboard with a generic controller in it.
USB_VID = 0x258a
USB_PID = 0x1
# This is ordered by bit position.
MODIFIERS = []

device = usb.core.find(idVendor=USB_VID, idProduct=USB_PID)

print(device.manufacturer, device.product)

# Test to see if the kernel is using the device and detach it.
if device.is_kernel_driver_active(0):
    device.detach_kernel_driver(0)

# Boot keyboards have 8 byte reports
buf = array.array("B", [0] * 8)
report_count = 0
while True:
    try:
        count = device.read(0x81, buf)
    except usb.core.USBTimeoutError:
        continue
    if report_count % 15 == 0:
        print("modifiers keys")
    print(buf[0], end=" ")
    for i in range(2, 8):
        if buf[i] > 0:
            print(buf[i], end=" ")
    print()
    report_count += 1
```
```

### Behavior

When running the basic_keyboard script linked above modified with the VID and PID of my device it results in this error:
```
Traceback (most recent call last):
  File "code.py", line 24, in <module>
usb.core.USBError: No configuration set
```

There is a `set_configuration()` method in the docs here: https://docs.circuitpython.org/en/latest/shared-bindings/usb/core/index.html#usb.core.Device.set_configuration which mentions: 

> The configuration parameter is the bConfigurationValue field of the configuration you want to set as active. If you call this method without parameter, it will use the first configuration found. As a device hardly ever has more than one configuration, calling the method without arguments is enough to get the device ready.

I tried calling set_configuration() with no argument and that does not raise an error itself. But the `count = device.read(0x81, buf)` line in the main loop does still raise an error albeit a different one now:
```
Traceback (most recent call last):
  File "code.py", line 25, in <module>
usb.core.USBError: 
```

tannewt, 2024-02-23:
@hathach The second read issue is due to the endpoint being claimed already by an internal HID read. Should I add a blanket abort transfer call or have the HID library manage the abort?

hathach, 2024-02-24:
@tannewt let me try it out first on Monday since I don't follow how usb host is managed in cpy.

hathach, 2024-02-29:
I have dived into this and able to find the root cause, there are 2 issue
1. `usb.core.find()` return currently attached device with vid/pid, but the returnned device has self->configuration_descriptor set to NULL. In order to use read/write endpoinnt API, you will need to get connfiguration descriptor with
```
config_descriptor = adafruit_usb_host_descriptors.get_configuration_descriptor(device, 0)
```
or set_configuration() which also try to read configuration as well.

2. Seconnd issue is due to the keyboard endpoint is currently busy with previous queue transfer as poinnted out by Scott. We should abort transfer when dettaching from kernel driver. Note: set_configuration() should also (per USB specs) reset all endpoint to its initial state including aborting all pending transfer. I will update tinyusb host with this as well.

tannewt, 2024-02-29:
> I have dived into this and able to find the root cause, there are 2 issue
>
>     1. `usb.core.find()` return currently attached device with vid/pid, but the returnned device has [...]

I added the set_configuration() call in #8976.

> 2. Seconnd issue is due to the keyboard endpoint is currently busy with previous queue transfer as poinnted out by Scott. We should abort transfer when dettaching from kernel driver. Note: [...]

I don't think adding the set_configuration() call does clear the endpoints currently. Let me know when you have a new TinyUSB version and I'll integrate it. Thanks for looking into this!

hathach, 2024-03-04:
https://github.com/adafruit/circuitpython/pull/8999 should fix this issue, resovled by either
- detach driver from kernnel or
- SET_CONFIGURATION which also abort all pending trannsfer for all non-control endpoinnts

closed 2024-03-04 by tannewt