adafruit/circuitpython #8887: UART swapping bytes?

tannewt, 2024-02-06:
A user on the forum reports bytes being swapped in UART read on ESP32-S3. https://forums.adafruit.com/viewtopic.php?t=208042

anecdata, 2024-02-06:
Tested:
`Adafruit CircuitPython 8.2.8 on 2023-11-16; Adafruit Feather RP2040 with rp2040`
with
```py
serial = busio.UART(board.TX, board.RX, baudrate=921600)
```

connected to:

`Adafruit CircuitPython 9.0.0-beta.0 on 2024-01-27; ESP32-S3-DevKitC-1-N8R8 with ESP32S3`
with 
```py
serial = busio.UART(board.IO15, board.IO18, baudrate=921600)
```

Each transmitting then receiving. All bytes arrive in order on both sides.

anecdata, 2024-02-07:
Switching the ESP32-S3 to `Adafruit CircuitPython 8.2.9 on 2023-12-06; ESP32-S3-DevKitC-1-N8R8 with ESP32S3` gives the same result. All bytes arrive in order on both sides.

tannewt, 2024-02-07:
Thanks for testing @anecdata! Would you mind sharing your full code? I'd like to use it myself :-)

closed 2024-02-07 by tannewt

anecdata, 2024-02-07:
Sure,...

<details>
<summary>ESP32-S3</summary>

```py
import time, board, busio

serial = busio.UART(board.IO15, board.IO18, baudrate=9600)
while True:
    print(f'serial wrote: {serial.write(b'1234567890')} bytes')
    time.sleep(1)
    print(f'serial read:  {serial.read(10)}')
    time.sleep(1)
```
</details>

<details>
<summary>RP2040</summary>

```py
import time, board, busio

serial = busio.UART(board.TX, board.RX, baudrate=9600)
while True:
    print(f'serial wrote: {serial.write(b'1234567890')} bytes')
    time.sleep(1)
    print(f'serial read:  {serial.read(10)}')
    time.sleep(1)
```
</details>

vladak, 2024-02-07:
Could someone elaborate on what the problem might be ? Did I perhaps solder the wires incorectly ?

dhalbert, 2024-02-07:
I replied in the thread: https://forums.adafruit.com/viewtopic.php?t=208042 that perhaps it is not out of order, but out of sync. Two bytes read in incorrect order will appear to be swapped bytes. We can continue discussing in the forum thread. It might be a US100 driver problem.

dhalbert, 2024-02-07:
Rediagnosed in #8890.