adafruit/circuitpython #8897: Silicon Labs devkit_xg24_brd2601b won't build

MHVUT, 2024-02-09:
Silicon Labs - Issues rebuilding the .bin file for the dev_kit_xg25_brd2601b. Followed the Adafruit guide and the SiLabs READ.md. Trying to build with Ubuntu Linux 22.04.2 (x86_64 emulation) on a Mac with the Parallels VM version 19.2.1. This is the entire build output from the ‘make BOARD=devkit_xg24_brd2601b’ cli command.

The guides have been followed exactly with a new clean VM.

The Adafruit guide has the 'make BOARD=circuitplayground_express' at the end of their documentation which works fine, no issues. The build problem looks to be specific to the SiLabs port. The 'make fetch-port-submodules' has been done for the SiLabs port.

parallels@ubuntu-linux-22-04-02-desktop:~/circuitpython/ports/silabs$ make BOARD=devkit_xg24_brd2601b
- Verbosity options: any combination of "steps commands rules", as `make V=...` or env var BUILD_VERBOSE
- Verbosity options: any combination of "steps commands rules", as `make V=...` or env var BUILD_VERBOSE
SLC generates project
sdk was set to /home/parallels/circuitpython/ports/silabs/gecko_sdk
The extension /home/parallels/circuitpython/ports/silabs/cp_efr32_extension/cp_efr32.slce is already trusted at the requested level. Nothing to do.
The sdk /home/parallels/circuitpython/ports/silabs/gecko_sdk/ is already trusted at the requested level. Nothing to do.
Found valid toolchain component toolchain_gcc for gcc
Generation location: /home/parallels/circuitpython/ports/silabs/build-devkit_xg24_brd2601b
Generation: Starting AC generation for ucProject in project circuitpython_efr32: Fri Feb 09 19:54:24 UTC 2024
Project is already viable -- no choices need be made.
Problems generating template files from component: bluetooth_stack - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: bluetooth_stack - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: board_control - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: cmsis_core - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: component_catalog - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: device_init_clocks - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: event_handler - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: event_handler - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: i2cspm_core - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: i2cspm_core - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: mbedtls_config - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: mbedtls_config - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: psa_crypto_config - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: toolchain_gcc - Potential jar issue: python script did not properly register a template generator
Generation issue: com.silabs.ss.framework.uc.core.api.exception.MultiException: Multi-Exceptions available:
  Multi-Exceptions available:
  Problems generating template files from component: bluetooth_stack - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: bluetooth_stack - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: board_control - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: cmsis_core - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: component_catalog - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: device_init_clocks - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: event_handler - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: event_handler - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: i2cspm_core - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: i2cspm_core - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: mbedtls_config - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: mbedtls_config - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: psa_crypto_config - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: toolchain_gcc - Potential jar issue: python script did not properly register a template generator
Generation to /home/parallels/circuitpython/ports/silabs/build-devkit_xg24_brd2601b for circuitpython_efr32 did not succeed.
Slc-cli:
JVM terminated. Exit code=255
/home/parallels/.sdkman/candidates/java/current/bin/java
-Dosgi.requiredJavaVersion=11
-Dosgi.console=none
-Xms40m
-Declipse.p2.unsignedPolicy=allow
-Djava.net.preferIPv4Stack=true
-Declipse.exitdata=
-Duc.workingdirectory=/home/parallels/circuitpython/ports/silabs
-Dslc_consolecolour=false
-Dslc_consolewidth=185
-jar /home/parallels/circuitpython/ports/silabs/tools/slc_cli_linux/bin/slc-cli//plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar
-os linux
-ws gtk
-arch x86_64
-launcher /home/parallels/circuitpython/ports/silabs/tools/slc_cli_linux/bin/slc-cli/slc-cli
-name Slc-cli
--launcher.library /home/parallels/circuitpython/ports/silabs/tools/slc_cli_linux/bin/slc-cli//plugins/org.eclipse.equinox.launcher.gtk.linux.x86_64_1.2.400.v20211117-0650/eclipse_11600.so
-startup /home/parallels/circuitpython/ports/silabs/tools/slc_cli_linux/bin/slc-cli//plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar
--launcher.appendVmargs
-exitdata 10002 generate
-name=circuitpython_efr32 circuitpython_efr32.slcp
--sdk gecko_sdk
--with brd2601b
-tlcn gcc
-d=build-devkit_xg24_brd2601b
-vm /home/parallels/.sdkman/candidates/java/current/bin/java
-vmargs
-Dosgi.requiredJavaVersion=11
-Dosgi.console=none
-Xms40m
-Declipse.p2.unsignedPolicy=allow
-Djava.net.preferIPv4Stack=true
-Declipse.exitdata=
-Duc.workingdirectory=/home/parallels/circuitpython/ports/silabs
-Dslc_consolecolour=false
-Dslc_consolewidth=185
-jar /home/parallels/circuitpython/ports/silabs/tools/slc_cli_linux/bin/slc-cli//plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar 
make[1]: *** [Makefile:177: slc-generate] Error 255
make: *** [Makefile:142: build-devkit_xg24_brd2601b/circuitpython_efr32.Makefile] Error 2

dhalbert, 2024-02-09:
@silabs-ChatNguyen I don't know what the issue might be in the build above. We asked @MHVUT to post here so we could ping you. My own build environment is working, but a clean build in a new environment should too. Do you know what the issue might be? 

Are there Silabs forums that might also be a place to ask this question, or is this the best route? Thanks.

MHVUT, 2024-02-20:
Hi https://github.com/silabs-ChatNguyen Would it be possible to get a response for this issue and help work out a solution? Thanks so much.

dhalbert, 2024-02-20:
@MHVUT I'm wondering, which version of Java are you using? I have
```sh
$ java --version
openjdk 11.0.21 2023-10-17
OpenJDK Runtime Environment (build 11.0.21+9-post-Ubuntu-0ubuntu122.04)
OpenJDK 64-Bit Server VM (build 11.0.21+9-post-Ubuntu-0ubuntu122.04, mixed mode, sharing)
```

silabs-ChatNguyen, 2024-02-20:
> @MHVUT I'm wondering, which version of Java are you using? I have
> 
> ```shell
> $ java --version
> openjdk 11.0.21 2023-10-17
> OpenJDK Runtime Environment (build 11.0.21+9-post-Ubuntu-0ubuntu122.04)
> OpenJDK 64-Bit Server VM (build 11.0.21+9-post-Ubuntu-0ubuntu122.04, mixed mode, sharing)
> ```

Sorry for the late reply. I was just wrapping up the Lunar New Year holiday.
I think this issue due to wrong java version. Code generation tool are using java 11.

MHVUT, 2024-02-20:
Thanks guys for the support.

The SiLabs UG520 app note states in section 2.1 for requirements…

￼
I actually had installed this version based on the UG520 app note of requiring version 17 or higher...

openjdk 21.0.2 2024-01-16 LTS
OpenJDK Runtime Environment Temurin-21.0.2+13 (build 21.0.2+13-LTS)
OpenJDK 64-Bit Server VM Temurin-21.0.2+13 (build 21.0.2+13-LTS, mixed mode, sharing)

Per the comments I used sdkman to install and set the following version as the default…

￼

Now confirmed as the version in use…

openjdk 11.0.21 2023-10-17 LTS
OpenJDK Runtime Environment Corretto-11.0.21.9.1 (build 11.0.21+9-LTS)
OpenJDK 64-Bit Server VM Corretto-11.0.21.9.1 (build 11.0.21+9-LTS, mixed mode)

But I still get the same problem.

So my question is exactly which version of Java should be installed?

Thanks,

JT

silabs-ChatNguyen, 2024-02-20:
I am using default-jdk of ubuntu 22.04
openjdk version "11.0.21" 2023-10-17
OpenJDK Runtime Environment (build 11.0.21+9-post-Ubuntu-0ubuntu122.04)
OpenJDK 64-Bit Server VM (build 11.0.21+9-post-Ubuntu-0ubuntu122.04, mixed mode, sharing)

Could you help to run below cmd in your side:
cd /port/silabs/
make BOARD=devkit_xg24_brd2601b clean
make BOARD=devkit_xg24_brd2601b slc-generate

MHVUT, 2024-02-20:
See output log for ‘clean’ and ’slc-generate’ commands as requested. I have posted the full output log here in case it provides more information that may be useful to understand the problem.

***@***.***:~/circuitpython/ports/silabs$ make BOARD=devkit_xg24_brd2601b clean
- Verbosity options: any combination of "steps commands rules", as `make V=...` or env var BUILD_VERBOSE
rm -rf build-devkit_xg24_brd2601b 
***@***.***:~/circuitpython/ports/silabs$ make BOARD=devkit_xg24_brd2601b slc-generate
- Verbosity options: any combination of "steps commands rules", as `make V=...` or env var BUILD_VERBOSE
SLC generates project
sdk was set to /home/parallels/circuitpython/ports/silabs/gecko_sdk
The extension /home/parallels/circuitpython/ports/silabs/cp_efr32_extension/cp_efr32.slce is already trusted at the requested level. Nothing to do.
The sdk /home/parallels/circuitpython/ports/silabs/gecko_sdk/ is already trusted at the requested level. Nothing to do.
Found valid toolchain component toolchain_gcc for gcc
Generation location: /home/parallels/circuitpython/ports/silabs/build-devkit_xg24_brd2601b
Generation: Starting AC generation for ucProject in project circuitpython_efr32: Tue Feb 20 09:07:14 PST 2024
Project is already viable -- no choices need be made.
Problems generating template files from component: bluetooth_stack - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: bluetooth_stack - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: board_control - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: cmsis_core - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: component_catalog - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: device_init_clocks - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: event_handler - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: event_handler - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: i2cspm_core - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: i2cspm_core - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: mbedtls_config - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: mbedtls_config - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: psa_crypto_config - Potential jar issue: python script did not properly register a template generator
Problems generating template files from component: toolchain_gcc - Potential jar issue: python script did not properly register a template generator
Generation issue: com.silabs.ss.framework.uc.core.api.exception.MultiException: Multi-Exceptions available:
  Multi-Exceptions available:
  Problems generating template files from component: bluetooth_stack - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: bluetooth_stack - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: board_control - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: cmsis_core - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: component_catalog - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: device_init_clocks - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: event_handler - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: event_handler - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: i2cspm_core - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: i2cspm_core - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: mbedtls_config - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: mbedtls_config - Potential jar issue: python script did not properly register a template generator
  Problems generating template files from component: psa_crypto_config - Potential jar issue: python script did not properly register a template generator
[... cut ...]
-vm /home/parallels/.sdkman/candidates/java/current/bin/java
-vmargs
-Dosgi.requiredJavaVersion=11
-Dosgi.console=none
-Xms40m
-Declipse.p2.unsignedPolicy=allow
-Djava.net.preferIPv4Stack=true
-Declipse.exitdata=
-Duc.workingdirectory=/home/parallels/circuitpython/ports/silabs
-Dslc_consolecolour=false
-Dslc_consolewidth=172
-jar /home/parallels/circuitpython/ports/silabs/tools/slc_cli_linux/bin/slc-cli//plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar 
make: *** [Makefile:177: slc-generate] Error 255

silabs-ChatNguyen, 2024-02-21:
I am not sure about your build environment.
Can you setup and build with docker ubuntu:22.04?
I will try build a docker environment and provide to you.

MHVUT, 2024-02-21:
OK I think I have docker build setup. I am way out of my comfort zone here 8-)). I am new to Linux just because that appears to be the only way to build Circuit Python. So not really sure what I am going at this stage.

Please send the docker environment and I’ll try to use it.

Thanks,

MHVUT, 2024-02-21:
So I decided to try something completely different rather than trying to fix the current Ubuntu 22.04.2 (x86_64 emulation) with Parallels on my Mac. I started a completely new VM but this time using Ubuntu 22.04.2 ARM64 to see if that makes any difference.

I followed the Adafruit notes as defined here…

https://learn.adafruit.com/building-circuitpython/linux￼
Building CircuitPython
learn.adafruit.com

and here…

https://docs.circuitpython.org/en/latest/ports/silabs/README.html#prerequisites￼

Here is the status…

***@***.***:~/circuitpython/ports/silabs$ java --version
openjdk 11.0.21 2023-10-17
OpenJDK Runtime Environment (build 11.0.21+9-post-Ubuntu-0ubuntu122.04)
OpenJDK 64-Bit Server VM (build 11.0.21+9-post-Ubuntu-0ubuntu122.04, mixed mode)

***@***.***:~/circuitpython/ports/silabs$ python3 --version
Python 3.10.12

***@***.***:~/circuitpython/ports/silabs$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/parallels/bin/gcc-arm-none-eabi-10.3-2021.10/bin:/home/parallels/circuitpython/ports/silabs/tools/slc_cli_linux
***@***.***:~/circuitpython/ports/silabs$ 

***@***.***:~/circuitpython/ports/silabs$ make BOARD=devkit_xg24_brd2601b clean
- Verbosity options: any combination of "steps commands rules", as `make V=...` or env var BUILD_VERBOSE
rm -rf build-devkit_xg24_brd2601b 

***@***.***:~/circuitpython/ports/silabs$ make BOARD=devkit_xg24_brd2601b slc-generate
- Verbosity options: any combination of "steps commands rules", as `make V=...` or env var BUILD_VERBOSE
SLC generates project
Traceback (most recent call last):
  File "/home/parallels/circuitpython/ports/silabs/tools/slc_cli_linux/slc", line 23, in <module>
    sys.exit(_entryPoint(sys.argv[1:]))
  File "/home/parallels/circuitpython/ports/silabs/tools/slc_cli_linux/lib/slc_run.py", line 1300, in _entryPoint
    ret = _simpleSlc(
  File "/home/parallels/circuitpython/ports/silabs/tools/slc_cli_linux/lib/slc_run.py", line 1158, in _simpleSlc
    return _launchCliNoDaemon(args, launch_uc, java_location, debug, do_timing)
  File "/home/parallels/circuitpython/ports/silabs/tools/slc_cli_linux/lib/slc_run.py", line 962, in _launchCliNoDaemon
    cli_process = subprocess.Popen(
  File "/usr/lib/python3.10/subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1863, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/home/parallels/circuitpython/ports/silabs/tools/slc_cli_linux/lib/../bin/slc-cli/slc-cli'
make: *** [Makefile:173: slc-generate] Error 1

So different problems than before but it still does not work. This is with a completely clean Ubuntu install. At least the java environment appears to match your setups now. 

I have been trying to get this to work for nearly a month now. There appears to be issues and/or confusion on exactly how the SiLabs CP build setup should work. I have done this so many times now. 

To go back to why I am trying to do this. I have a custom board with the same EFR32MG24 part that is on the xG24 Dev Kit. I want to make sure I can successfully rebuild CP for the Dev Kit before going down the path of making changes for my specific HW.

Maybe you could follow the instructions yourself with a clean Ubuntu VM and see what’s different to get it to work, clearly there has to be something that is missing or unclear.

Thanks again for support.

JT

silabs-ChatNguyen, 2024-02-22:
You can try with docker environment with below guide:
## Extract
```
dev-machine:~$  unzip build_cp.zip
dev-machine:~$  cd build_cp
```
## Build docker image
```
dev-machine:~/build_cp$  docker build -t docker_cp .
```

## Run docker image & build
````
dev-machine:~/build_cp$  docker run -it -v $PWD:/build docker_cp

root@docker:/#  cd ./build/
root@docker:/build/#  git clone https://github.com/adafruit/circuitpython.git
root@docker:/build/#  cd ./circuitpython
root@docker:/build/circuitpython#  git checkout 8.2.10
root@docker:/build/circuitpython#  cd ./ports/silabs
root@docker:/build/circuitpython/ports/silabs#  make fetch-port-submodules
root@docker:/build/circuitpython/ports/silabs#  make BOARD=devkit_xg24_brd2601b V=2 clean
root@docker:/build/circuitpython/ports/silabs#  make BOARD=devkit_xg24_brd2601b V=2

```
[build_cp.zip](https://github.com/adafruit/circuitpython/files/14377761/build_cp.zip)

MHVUT, 2024-02-22:
Thanks for the docker file and instructions. Looks to be exactly the same problem that I posted about yesterday when trying a clean new Ubuntu VM.

***@***.***:/build/circuitpython/ports/silabs# make BOARD=devkit_xg24_brd2601b V=2 clean
rm -rf build-devkit_xg24_brd2601b 

***@***.***:/build/circuitpython/ports/silabs# make BOARD=devkit_xg24_brd2601b V=2
SLC generates project
Traceback (most recent call last):
  File "/build/circuitpython/ports/silabs/tools/slc_cli_linux/slc", line 23, in <module>
    sys.exit(_entryPoint(sys.argv[1:]))
  File "/build/circuitpython/ports/silabs/tools/slc_cli_linux/lib/slc_run.py", line 1300, in _entryPoint
    ret = _simpleSlc(
  File "/build/circuitpython/ports/silabs/tools/slc_cli_linux/lib/slc_run.py", line 1158, in _simpleSlc
    return _launchCliNoDaemon(args, launch_uc, java_location, debug, do_timing)
  File "/build/circuitpython/ports/silabs/tools/slc_cli_linux/lib/slc_run.py", line 962, in _launchCliNoDaemon
    cli_process = subprocess.Popen(
  File "/usr/lib/python3.10/subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1863, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/build/circuitpython/ports/silabs/tools/slc_cli_linux/lib/../bin/slc-cli/slc-cli'
make[1]: *** [Makefile:174: slc-generate] Error 1
make: *** [Makefile:149: build-devkit_xg24_brd2601b/circuitpython_efr32.Makefile] Error 2
***@***.***:/build/circuitpython/ports/silabs#

MHVUT, 2024-02-25:
Hi Guys,

Just want to give an update on progress rebuilding the CircuitPython .bin file for the SiLabs EFR32MG24 Dev Kit. I finally got it to work. Here’s how…

I have an old Intel i5 based Mac on which I re-installed the mac OS, specifically Big Sur 11.7.10, so it was completely clean. Then I installed the Parallels VM tools, then used the Parallels tools to install Ubuntu 22.04. Following that I installed the Docker build tools, and then followed the instructions for building and running the docker VM. And it worked. So I can finally rebuild the CP .bin file for the SiLabs boards.

I made no changes to any of the code or setup instructions when using the older Intel i5 based Mac with macOS Big Sur. Up to that point I was trying to use an M2 Pro based Mac mini with macOS Sonoma, specifically 14.3.1. So there does appear to be some differences between either an Apple silicon or Intel silicon based Mac, and/or using Big Sur versus Sonoma macOS with Parallels that impacts the CP development tool chain for SiLabs parts.

I was able to take the new .bin file and program it on one of my EFR32MG24 Dev Kit boards, successfully connect with Thonny and run the simple ‘Hello World’ .py file. I know it’s the file I built because it has today’s date in the Adafruit CP boot message.

Now I am trying to bring up CP on my custom board. I have the Mini Simplicity Connector on the custom board, and I have the WSTK. I am able to use SS5 to build and download .s37 files to the custom board. I have been using the bootloader and the bare metal ‘blinky’ which all works as expected. I can flash the RGB LED on my custom board.

However, when I use Simplicity Commander to program CP on to the custom board, how do I then connect Thonny to get to the REPL? Thonny sees the J-Link interface for the WSTK, but Thonny says the ‘Device is busy or does not respond’. Is there additional config required on the WSTK, or elsewhere, to get Thonny to connect to the custom board using the Mini Simplicity interface? I believe the Mini Simplicity interface is working given that I can use SS5 to program the custom board. It is just the CP/REPL that doesn’t appear to be working.

Thanks so much Nguyen for the support, I wouldn’t have been able to get the build working w/o your docker build file.

JT

silabs-ChatNguyen, 2024-02-26:
Maybe the bring-up code on your custom board does not work, so Thonny can't connect to your custom board.
The board communicate with Thonny via EUSART0. You can try build & flash firmware to print something to VCOM first.
In addition, you should use Ozone-Segger tool to debug your bring-up code.

MHVUT, 2024-02-26:
Thanks. I have not used Ozone Segger and I will need to learn how to use those tools as well. Before going down that route I would like your comments on what I have tried. I will also check what options are available to build a simple Tx/Rx interface with EUSART0.

• With the EFR32MG24 Dev Kit I can…

1. Use SS5 to build and load the BGAPI bootloader and use the bare metal ‘Blinky’ to flash the blue LED on the Dev Kit. No issues.

2. Use Simplicity Commander to load and run the official Adafruit CP 8.2.10 release (.bin file). No issues.

3. Use Simplicity Commander to load and run the version of CP that I built (.bin file). No issues. This verifies both the official CP release, and the version I built, have no problem on the Dev Kit.

• With the custom board I can…

1. Use SS5 to build and load the BGAPI bootloader and use the bare metal ‘Blinky’ to flash the blue LED on the custom board. No issues. This verifies the tools and build process work, as well as the SiLabs part is working as expected.

2. However, when I use Simplicity Commander to load and run the official Adafruit CP 8.2.10 release (.bin file) Thonny does not work. This is strange as the previous step shows the development tools and HW appear to work as expected.

3. Similarly for the version of CP that I built, also doesn’t work. Again as the code runs on the EFR32 Dev Kit I would expect something on Thonny. The version of CP that I built is really just the exact same code as the official release anyway. And it works on the EFR32MG24 Dev Kit.

• This is how I tried to debug further…

1. Connected a serial port to the Tx VCOM pin on the EFR32 Dev Kit and I can see the data/messages between Thonny and the EFR32 Dev Kit on the terminal emulator. All looks reasonable and as expected. So now I know what to expect on the Tx VCOM signal from the custom board.

2. When I do the same on the custom board there is no Tx VCOM data/messages between Thonny and the custom board. That appears to be the reason why Thonny cannot connect, there is no VCOM interface/connection. So the question is why is that?

3. I have checked all the HW against the EFR32 Dev Kit. All the power rails look good and are at the correct voltage. The EFR32 DC switcher is at 1.8V as expected. Given that the custom board has no issues loading and running the bootloader and the blinky example I am confident the custom hardware is working.

4. So, is there anything in the EFR32MG24 CP official build that expects external HW (SPI, I2C, other) that would prevent any communication from CP to connect with Thonny? I know I need to adjust the custom board CP to adjust for the different HW interfaces, but I would expect something from the VCOM interface and Thonny to connect.

5. My custom hardware follows the EFR32MG24 Dev Kit very closely for device interfaces. The I2C and SPI Flash interfaces are the same pins, as is the RGB LED. What is different is the actual  I2C and SPI sensor devices.

6. What would prevent any Tx on the VCOM interface when all the power rails etc look correct? I verified the VCOM Tx and Rx signals are correct on the Mini Simplicity connector and they are not shorted to ground.

I would welcome any further comment or guidance on why CP won’t connect with Thonny for my custom board, when as far as I can tell there is no issue with the EFR32MG24 silicon on my board.

Thanks so much for the on-going support.

JT

tannewt, 2024-02-26:
Are you seeing the LEDs blink like CircuitPython is running? They should blink during start up. A debugger will make it much easier to get it going. My process doesn't use Segger Ozone but GDB: https://learn.adafruit.com/debugging-the-samd21-with-gdb (Applies pretty well to all cortex-m cpus.)

MHVUT, 2024-02-27:
Thanks for the message, I’ll take a look at the GBD debugging notes.

My custom board has a single RGB LED just like the SiLabs EFR32MG24 Dev Kit. That RGB LED doesn’t blink during CP start up on either the Dev Kit or the custom board, even with the official 8.2.10 release. CP works fine on the SiLabs Dev kit.

While I was trying to get the CP build environment working over the past month I also tried out non-CP code, mostly using the examples in SiLabs Simplicity Studio. One issue that I am currently working is a serial port loopback that works fine on the SiLabs Dev Kit, but sort of works on the custom board. This is all non-CP. What I suspect may be happening is an issue with the crystal accuracy (perhaps!) that when its just simple LED flashing it looks correct, but for more timing critical tasks, like serial ports at 115200, not so much.

Anyway, that is my current thinking. It may be CP is working on my custom board but because of the serial port timing issues Thonny can’t connect. If I can get the non-CP serial port loopback working then maybe CP will work.

tannewt, 2024-02-27:
> That RGB LED doesn’t blink during CP start up on either the Dev Kit or the custom board, even with the official 8.2.10 release.

Switch to 9.x. I *think* I added it there. 9.0.0 is almost stable and has many fixes.

MHVUT, 2024-02-27:
I loaded 9.0.0-beta.2 on both the SiLabs EFR32MG24 Dev Kit and my custom board, and the RGB LED does indeed flash periodically on both. Which is great, it means my custom board is running CP. I was also able to successfully rebuild 9.0.0-beta.2 as well and verified it works the same on my custom board. Again, all good.

But Thonny won’t connect. Connecting a serial terminal to the VCOM Tx/Rx signals there are text strings transmitted out from CP at boot, but all the characters are scrambled. Some of it may not be ASCII human readable anyway, but there is at least something coming out. Now I need to get the non-CP serial port loopback working which hopefully will resolve the Thonny connect issue.

Anyway, thanks for the great support. Moving up to 9.0.0 was a good call. At least I know my custom board is running CP.

MHVUT, 2024-03-01:
Hi,

Is there any documentation for what CP outputs on the serial port at boot up?

I have the SiLabs EFR32MG24 Dev Kit, and my custom board, both running the exact same 9.0.0-beta.2 build, but the captured output is different from each board. Trying to understand why I can’t connect to Thonny with my custom board, and why the transmitted serial boot sequence is different, with the same FW.

Thanks,

JT

￼

MHVUT, 2024-03-01:
[CP Boot Sequence.pdf](https://github.com/adafruit/circuitpython/files/14466948/CP.Boot.Sequence.pdf)

MHVUT, 2024-03-04:
I still can’t get my SiLabs EFR32MG24 custom board to connect to Thonny and I have been researching this problem further. With CP 9.0.0-beta.2 the RGB LED flashes periodically so I know CP is running w/o issue. The problem appears to be with the VCOM serial port configuration.

To debug further I took the very simple ‘Platform - I/O Stream EUSART Bare-Metal’ example from the SiLabs SS5 tools, built the code and flashed to the SiLabs xG24 Dev Kit board with no problem. Since this is just a simple UART loopback with very little configuration the exact same code should work on the custom board. It doesn’t. All this application does is echo serial port characters back to the terminal.

So I created another application just for the custom board and built the same ‘Platform - I/O Stream EUSART Bare-Metal’ example again, flashed to the custom board and it worked. I compared the SiLabs .slcp configuration files to understand what the differences are.

For the custom board application the ‘Low Frequency Crystal Oscillator (LFXO)’ and the ‘High Frequency Crystal Oscillator (HFXO)’ components are not installed in the build by default. But the build works. These two installed FW components appear to be the only difference between the two auto generated .slcp files.

So I added the LFXO component to the .slcp file for the ‘Platform - I/O Stream EUSART Bare-Metal’ on the custom board. No issue, still works. However, adding the HFXO component for the custom board causes the problem I have been trying to debug for sometime. Why does the auto generated application configuration for the Dev Kit include the HFXO component, but the application configuration for the custom board does not?

Question for the SiLabs folks…https://github.com/silabs-ChatNguyen what does the HFXO component do or add to the application that would cause the custom board serial interface to fail? And how and/or where is the default configuration setup when the HFXO component is not installed?

I ask because I believe this is the reason I cannot connect to Thonny with the custom board. There is clearly some difference between installing and not installing the HFXO component, and that will need to be adjusted to build CP for a custom SiLabs based board. 

For reference when the HFXO is auto generated it has the defaults as ‘Mode: Crystal Oscillator’, ‘Frequency: 39000000’, ‘HFXO precision in ppm: 50’, and ‘CTUNE: 40’. The custom board has the recommended 39MHz cystal, TXC 8Y39072005, per the SiLabs AN0016.2 app note.

Where specifically in the SiLabs CP port is the HFXO configuration managed? 

Thank you for the on-going support. Hopefully I can get past these issues this week.

JT

tannewt, 2024-03-04:
> Connecting a serial terminal to the VCOM Tx/Rx signals there are text strings transmitted out from CP at boot, but all the characters are scrambled.

This sounds like the UART is going the wrong speed.

> Is there any documentation for what CP outputs on the serial port at boot up?

When using UART for the serial console, the first text output is ["Serial console setup\r\n"](https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/serial.c#L148).

MHVUT, 2024-03-06:
Hi https://github.com/silabs-ChatNguyen do you have any input for the SiLabs CP port specifically for the HFXO. It appears this FW component has an impact on the serial port baud rate likely due to the osc. not being at the correct frequency.

Thank you.

MHVUT, 2024-03-08:
Hi @silabs-ChatNguyen I have been working on this problem all week. Are you able to help?

From my post a few days ago re the Platform - I/O Stream EUSART Bare-Metal issue, I believe I now know exactly which SiLabs function is causing the problem with my custom board.

Using the SS5 tools, adding the 'clocks' component to the .slcp for Services | Device Initialization | Peripherals | Clocks causes the Platform - I/O Stream EUSART Bare-Metal application UART to fail. Removing the clocks component from the .slcp (with the GUI), makes it work. The EFR32MG24 Dev Kit works either way.

This is the specific function in 'sl_device_init_clocks.c'...

#include "sl_device_init_clocks.h"

#include  @"em_cmu.h"

sl_status_t sl_device_init_clocks(void)
{
  CMU_CLOCK_SELECT_SET(SYSCLK, HFXO);
#if defined(_CMU_EM01GRPACLKCTRL_MASK)
  CMU_CLOCK_SELECT_SET(EM01GRPACLK, HFXO);
#endif
#if defined(_CMU_EM01GRPBCLKCTRL_MASK)
  CMU_CLOCK_SELECT_SET(EM01GRPBCLK, HFXO);
#endif
#if defined(_CMU_EM01GRPCCLKCTRL_MASK)
  CMU_CLOCK_SELECT_SET(EM01GRPCCLK, HFXO);
#endif
  CMU_CLOCK_SELECT_SET(EM23GRPACLK, LFXO);
  CMU_CLOCK_SELECT_SET(EM4GRPACLK, LFXO);
#if defined(RTCC_PRESENT)
  CMU_CLOCK_SELECT_SET(RTCC, LFXO);
#endif
#if defined(SYSRTC_PRESENT)
  CMU_CLOCK_SELECT_SET(SYSRTC, LFXO);
#endif
  CMU_CLOCK_SELECT_SET(WDOG0, LFXO);
#if WDOG_COUNT > 1
  CMU_CLOCK_SELECT_SET(WDOG1, LFXO);
#endif
  return SL_STATUS_OK;
}
```

Commenting out all but the last return statement in that function makes the IOStream application work on the custom board. So I believe this is the problem. But I can't follow what that function is actually doing and why it makes the application serial port fail. It would be helpful to understand what is happening.

Which brings me back to the circuitpython issue. Does the SiLabs CP Port have a similar clocks init function that could be causing the same problem with CP? I have looked through the SiLabs CP port code and wasn't able to find a similar function.

I believe there is a problem with the CP SiLabs port for a custom board. The code runs but there appears to be an issue with the serial port baud rate setting that prevents a custom board connecting to Thonny. Data is transmitted out the VCOM port but it is jumbled, just as if the baud rate is set incorrectly.

I have tried all the typical serial port baud rates from 9600 and up. That didn't help. I even tried adjust the 39MHz clock setting in the HFXO component to try to 'fool' the code but make the VCOM port work. That wasn't successful either.

I would appreciate a response that either someone at SiLabs is able to take a look at this problem or not. That will determine if I am going to continue on with CP.

Thank You,

JT

tannewt, 2024-03-08:
Logic analyzers may be able to detect the baudrate of the UART. That may give you insight on what the clock issue is.

I usually use PWM to verify that my clocks are setup correctly but I'm not sure if the clocks between UART and PWM are shared.

MHVUT, 2024-03-08:
Thank you, great suggestion. Interesting findings. Using the exact same .bin FW image from CP 9.0.0-beta.2 for the SiLabs EFR32MG24 Dev Kit and my custom board the baud rate is different. It would be nice if it was a factor of 2, or power of, that would give some clues of what to look at, but no. The baud rate on the custom board is 83000 (see pictures) but is as expected (115200) on the Dev Kit for the exact same .bin file. 

There is something in the Dev Kit setup/config for the build that is not the same or not captured for a custom board SiLabs CP port. I can't explain how the same .bin would have different baud rates. Same EFR32MG24 silicon and using a SiLabs recommended 39MHz crystal, no external loading capacitors (as recommended), placed right next to the EFR part. Power rails all look good. This one has me confused for more than a week 8-)).

[image]

[image]

dhalbert, 2025-07-29:
@MHVUT Hi - it's been well over a year. Did you ever resolve this?

MHVUT, 2025-07-29:
> dhalbert

Hi. No, never understood what was going on, didn’t make sense why the baud rate would be different. I gave up on the SiLabs solution.