adafruit/circuitpython #8899: jpegio example in docs raises TypeError

FoamyGuy, 2024-02-10:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-beta.0-23-g8af780c7fa-dirty on 2024-02-10; Adafruit Feather ESP32-S3 TFT with ESP32S3

(built from main with `CIRCUITPY_JPEGIO = 1`)
```

### Code/REPL

```python
from jpegio import JpegDecoder
from displayio import Bitmap

decoder = JpegDecoder()
width, height = decoder.open("/sd/example.jpg")
bitmap = Bitmap(width, height)
decoder.decode(bitmap)
# .. do something with bitmap
```

### Behavior

This code is in the docs for jpegio here: https://docs.circuitpython.org/en/latest/shared-bindings/jpegio/index.html#jpegio.JpegDecoder

When it runs it raises this TypeError due to the number of arguments used when initializing the Bitmap object.

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "take_photo.py", line 97, in <module>
TypeError: function takes 3 positional arguments but 2 were given
```
which points back to this line: 
```
bitmap = Bitmap(width, height)
```
The Bitmap wants to have a value count passed to it. 

### Additional information

I'm also unsure about if / where or how a Palette object comes into the mix with the Bitmap created from jpegio.decode().

jepler, 2024-02-11:
The 3rd arg should be 65535. There is no pallette, the image is RGB565_SWAPPED

tannewt, 2024-02-12:
Fixed by #8905

closed 2024-02-12 by tannewt